zigpy/zha-device-handlers #2045: [Device Support Request] Smart Lock TY0A01 by _TYST12_snr3p274

eduardobittencourt, 2022-12-30:
**Describe the solution you'd like**
I'm trying to use my smart lock from Intelbras in Home Assistant but no entities are found. I tried to create a quirk for this scenario but the weird cluster `0xef00` does not exists on the actual described clusters in zigpy.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000a",
        "0xef00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST12_snr3p274",
  "model": "TY0A01",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.8",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Sao_Paulo",
    "os_name": "Linux",
    "os_version": "5.15.80",
    "supervisor": "2022.12.1",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "smartthinq_sensors": {
      "version": "0.27.1",
      "requirements": [
        "pycountry>=20.7.3",
        "xmltodict>=0.12.0",
        "cchardet>=2.1.7"
      ]
    },
    "hacs": {
      "version": "1.29.0",
      "requirements": [
[... 333 lines cut ...]
# {
#   "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
#   "endpoints": {
#     "1": {
#       "profile_id": 260,
#       "device_type": "0x0051",
#       "in_clusters": [
#         "0x0000",
#         "0x0004",
#         "0x0005",
#         "0x000a",
#         "0xef00"
#       ],
#       "out_clusters": [
#         "0x0019"
#       ]
#     }
#   },
#   "manufacturer": "_TYST12_snr3p274",
#   "model": "TY0A01",
#   "class": "zigpy.device.Device"
# }
```

</details>

TheJulianJES, 2023-01-04:
You have ``"0xef00"`` in quotes in your custom quirk. It should just be ``0xef00``.
However, even if the quirk applies, it won't do anything really.

[0xef00](https://github.com/zigpy/zha-device-handlers/blob/7788036230986cebe8e65281d7f58c7566172c8a/zhaquirks/tuya/__init__.py#L30) is the custom Tuya cluster, so more support would need to be added for the lock to work.

eduardobittencourt, 2023-01-04:
Hey @TheJulianJES thanks for the reply!

Do you think we can figure out what is going on here? Your observation about my quirk was correct, now the quirk is applied when I pair the device. My ideia here is to just try some clusters and see what happens, since I'm not so handy in HA and ZHA.

If I can provide more information, please let me know :)

TheJulianJES, 2023-01-04:
I'm not really familiar with Tuya. You can take a look at all current implementations for other Tuya devices here: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ (also look at ``__init__.py`` which contains most custom Tuya clusters)

But as far as I can see, there's no "Tuya lock cluster" which you could just plug in as a replacement.
It also looks like Zigbee2MQTT doesn't even have support for any Tuya locks.

eduardobittencourt, 2023-01-06:
I don't know if it helps, but I got some information from Developer Tuya IoT Platform.

<details>
<summary>https://iot.tuya.com/micro-app/cloud/api/v1/project/pid/datapoint/descripiton</summary>

```json
{
  "login": true,
  "result": [
    {
      "schema": {
        "accessType": "ro",
        "code": "unlock_fingerprint",
        "valueDesc": "{\"min\":0,\"max\":999,\"scale\":0,\"step\":1}",
        "valueType": "Integer",
        "langDict": {
          "unit": ""
        },
        "name": "Fingerprint Unlock",
        "remark": "unlock_fingerprint"
      },
      "functionSet": {
        "accessType": "ro",
        "code": "unlock_fingerprint",
        "valueDesc": "{\"min\":0,\"max\":999,\"scale\":0,\"step\":1}",
        "valueType": "Integer",
        "langDict": {
          "unit": ""
        },
        "name": "Fingerprint Unlock",
        "remark": "",
        "mainDP": "unlock_fingerprint"
      }
    },
    {
      "schema": {
        "accessType": "ro",
        "code": "unlock_password",
        "valueDesc": "{\"min\":0,\"max\":999,\"scale\":0,\"step\":1}",
        "valueType": "Integer",
        "langDict": {
          "unit": ""
        },
        "name": "Password",
        "remark": "unlock_password"
      },
      "functionSet": {
        "accessType": "ro",
        "code": "unlock_password",
        "valueDesc": "{\"min\":0,\"max\":999,\"scale\":0,\"step\":1}",
        "valueType": "Integer",
        "langDict": {
          "unit": ""
        },
        "name": "Password",
        "remark": "",
        "mainDP": "unlock_password"
      }
    },
    {
      "schema": {
        "accessType": "ro",
        "code": "unlock_temporary",
        "valueDesc": "{\"min\":0,\"max\":999,\"scale\":0,\"step\":1}",
        "valueType": "Integer",
        "langDict": {
          "unit": ""
        },
        "name": "Temporary Password",
        "remark": "unlock_temporary"
[... 933 lines cut ...]
      {
        "code": "EVENT_TYPE_TYLINK_PROPERTY_PUBLISH",
        "name": "EVENT_TYPE_TYLINK_PROPERTY_PUBLISH",
        "id": 40,
        "desc": "设备属性下发"
      },
      {
        "code": "EVENT_TYPE_TYLINK_ACTION_REPORT",
        "name": "EVENT_TYPE_TYLINK_ACTION_REPORT",
        "id": 41,
        "desc": "设备动作上报"
      },
      {
        "code": "EVENT_TYPE_TYLINK_ACTION_PUBLISH",
        "name": "EVENT_TYPE_TYLINK_ACTION_PUBLISH",
        "id": 42,
        "desc": "设备动作下发"
      }
    ],
    "t": 1673029922938,
    "success": true,
    "status": "ok"
  }
```
</details>

closed 2023-07-12 by github-actions[bot]

zeduperes, 2023-10-14:
@eduardobittencourt i have this same door lock, did you find a solution?

Oliv3rDog, 2023-11-21:
Hi there, @eduardobittencourt did you get any working solution to integrate this lock to HA?

pheliperocha, 2023-11-22:
Hey @eduardobittencourt did you find any solution?

eduardobittencourt, 2023-11-23:
Hi guys. Unfortunately I have no updates on this case. I tried to contact Tuya and Intelbras (the company who sells the locks in Brazil) looking for further documentation, but got no response.

pheliperocha, 2023-11-24:
@eduardobittencourt I see,

My lock is from a brazilian company as well, called fortezza, yet it actually uses the Tuya TY0A01. I came across this [Tuya documentation](https://developer.tuya.com/en/docs/iot/zigbee-doorlock-dp?id=K9fembhbeab0p), and I'm currently trying to decipher the commands outlined there . If I can't figure it out, I might resort to [analyzing Zigbee traffic](https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) as a last option.

Oliv3rDog, 2023-11-24:
I have just received an Aliexpress door lock called Tropernic and it is alwesome, beautifull and cheap as well.
It shows TY0A01  por _TYST12_bfejtdig  when trying ZHA, but the lock do not understant as a connection, and gives FAIL repport. 
Did you find any Lock custom quirk to manke a reverse try?
Can we find clusters names or types from Tuya Hub Smart Life connection?

adam-zlatniczki, 2024-08-09:
I've received my lock a few days ago and going to start working on a Tuya specific lock cluster that can be used in quirks soon. While my lock comes from a different manufacturer (_TYST12_bfejtdig), when the custom cluster is done, it should be easy to create new quirks for different Tuya locks based on that. The lock clusters have quite a long list of commands and attributes so I expect it to take quite some time. I plan to create a sort-of roadmap with "releases", starting with the most important minimum feature set (like lock-unlock), then extend it with additional functionality. As far as I see it, some dashboard will be necessary as well, because locks have features which are very inconvenient to do manually via cluster commands (like handling onboarded PIN codes and such).

andrelam, 2024-12-22:
> I've received my lock a few days ago and going to start working on a Tuya specific lock cluster that can be used in quirks soon. While my lock comes from a different manufacturer (_TYST12_bfejtdig), [...]

Hey, @adam-zlatniczki , any progress on this?

allansari, 2025-01-04:
I have the same issue with (TY0A01
by _TYST12_bfejtdig)

I'm wondering why they are selling uncompatible zigbee devices with HA !!!

niker, 2025-01-04:
I have a device with signature _TYST12_54rrtbbg, Tuya App exclusive as well, not supported by anything via zigbee, Tuya integration doesn't show anything either. I was able to make it work perfectly with HA.

You need to bind Tuya app to Google Voice Assistant on your phone and control the lock from Home Assistant with virtual voice commands. Add Google Assistant integration to HA and send chat (text) commands to it.

You can switch unlock/lock by reversing the lock direction temporarily, this can be done via two tap-to-run scenes in Tuya app, create a scene that sets direction to lock and another that sets direction to unlock and trigger the scene via HA Tuya integration. 

Once I figured it out, works like a charm in HA and the delay for commands is about 2-3s. Only downside is, the lock doesn't work during an internet outage. 

Attaching my unlock/lock script, you need to figure out the Google Assistant and Tuya integration yourself, read the docs in HA.

```
alias: Main Entrance Lock Control
sequence:
  - if:
      - condition: template
        value_template: "{{(lock | default(false)) == false}}"
    then:
      - action: scene.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: scene.lock_unlocking
      - action: google_assistant_sdk.send_text_command
        data:
          command: Lock door
      - delay:
          hours: 0
          minutes: 0
          seconds: 6
          milliseconds: 0
    else:
      - action: scene.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: scene.lock_locking
      - action: google_assistant_sdk.send_text_command
        data:
          command: Lock door
      - delay:
          hours: 0
          minutes: 0
          seconds: 6
          milliseconds: 0
description: ""
icon: mdi:lock-open-alert-outline
mode: single
fields:
  lock:
    selector:
      boolean: {}
    default: false
    required: true
    name: Lock
```

kingp0dd, 2025-01-04:
I think that means that it is still routed through tuya cloud. If tuya
cloud will be used anyway, i think you can do the same with just adding the
tuya integration in HA

niker, 2025-01-04:
> I think that means that it is still routed through tuya cloud. If tuya cloud will be used anyway, i think you can do the same with just adding the tuya integration in HA

Not on my specific lock, there are no entities when added in Tuya integration. No zigbee and no Tuya, even Tuya App automations can't control the lock, only direction. The only way to open my lock is tapping in Tuya App device screen. The Google Assistant is the only workaround I found that can fully integrate to HA.