zigpy/zha-device-handlers #2047: Thermostate Danfoss Ally Quirk - Integrate new model eTRV0103

michib-xx, 2022-12-30:
https://github.com/zigpy/zha-device-handlers/blob/23a166902e50d5c32019e0d4d32a968715041bb6/zhaquirks/danfoss/thermostat.py#L114

Hello,

I just installed my new Danfoss Ally, which is has this signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4678, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000a",
        "0x0020",
        "0x0201",
        "0x0204",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0000",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Danfoss",
  "model": "eTRV0103",
  "class": "danfoss.thermostat.DanfossThermostat"
}
```

Obviously, this is a different model than the default quirk is made for. Hence, the quirk is not applied.
I assume/hope that attributes and all are the same. 

Could you integrate the new model in the default quirk?

Thanks a lot.

Micha

javicalle, 2022-12-30:
But your device signature is saying that the quirk is loading.
Maybe do you have enabled your local quirk?

michib-xx, 2022-12-30:
Yes, you are right. I locally added the model to the list and it loaded. Actually, I just recognized this after I wrote this issue.
Nevertheless, the model should be integrated into the default file.

michib-xx, 2022-12-30:
Moreover, I don't know, if the quirk is actually working for the new model. It appears to me that values are not correctly read or written, using the defined attributes.

TheJulianJES, 2023-01-16:
Are there any improvements when just adding the new model to the quirk?

michib-xx, 2023-01-17:
I have not tried your updated version but only one which I changed myself.

I expected to use attributes like `external_measured_room_sensor` or  `control_algorithm_scale_factor`. E.g., the first one seems to work. The second one does not. I can write a new value and also read it back, but the control performance of the TRV does not change. As I understood, `control_algorithm_scale_factor` is some measure for a time constant of a filter, with which the set point is filtered. My "problem" is that the temperatur overshots significantly, obviously due to a too high integral gain of the PI controller. The filter would lead to a reference that the system can follow and hence reduce wind up of the integral part.

PR #2108 (2023-01-18): Adding hardware updated Ally
Fixes: #2047