zigpy/zha-device-handlers #2050: [BUG]

gluca, 2022-12-31:
**Describe the bug**
After pairing TS130F _TZ3000_e3vhyirx courtain switch it stops working after few seconds.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair the curtain switch
2. after paired move the slides of the cpver entity created
3. it works (but with wrong percentage) a couple of times and then stops workoing
4. See error

**Expected behavior**
it should operate the motor in the proper way with percentage feedback aligned with cover openong status

**Screenshots**

<details>

<summary>

</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_e3vhyirx",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130GP"
}

```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2022.12.8",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
[... 392 lines cut ...]
2022-12-31 13:53:53.531 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xE3B5), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=6, data=Serialized[b'\x08\xcc\n\x00\x00\x10\x00\x01@!\x00\x00\x02@!\x00\x00\x01\x800\x01\x02\x800\x00\x00P0\x01\x00\x80\x10\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-79)
2022-12-31 13:53:53.532 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0x0006] Received ZCL frame: b'\x08\xcc\n\x00\x00\x10\x00\x01@!\x00\x00\x02@!\x00\x00\x01\x800\x01\x02\x800\x00\x00P0\x01\x00\x80\x10\x00'
2022-12-31 13:53:53.534 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=204, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-31 13:53:53.537 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0x0006] Decoded ZCL frame: TuyaZBOnOffAttributeCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>)), Attribute(attrid=0x4001, value=TypeValue(type=uint16_t, value=0)), Attribute(attrid=0x4002, value=TypeValue(type=uint16_t, value=0)), Attribute(attrid=0x8001, value=TypeValue(type=enum8, value=<enum8.undefined_0x01: 1>)), Attribute(attrid=0x8002, value=TypeValue(type=enum8, value=<enum8.undefined_0x00: 0>)), Attribute(attrid=0x5000, value=TypeValue(type=enum8, value=<enum8.undefined_0x01: 1>)), Attribute(attrid=0x8000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-12-31 13:53:53.540 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0x0006] Received command 0x0A (TSN 204): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>)), Attribute(attrid=0x4001, value=TypeValue(type=uint16_t, value=0)), Attribute(attrid=0x4002, value=TypeValue(type=uint16_t, value=0)), Attribute(attrid=0x8001, value=TypeValue(type=enum8, value=<enum8.undefined_0x01: 1>)), Attribute(attrid=0x8002, value=TypeValue(type=enum8, value=<enum8.undefined_0x00: 0>)), Attribute(attrid=0x5000, value=TypeValue(type=enum8, value=<enum8.undefined_0x01: 1>)), Attribute(attrid=0x8000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2022-12-31 13:53:53.542 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0x0006] Attribute report received: on_off=<Bool.false: 0>, on_time=0, off_wait_time=0, backlight_mode=<enum8.undefined_0x01: 1>, power_on_state=<enum8.undefined_0x00: 0>, 0x5000=<enum8.undefined_0x01: 1>, child_lock=<Bool.false: 0>
2022-12-31 13:53:53.544 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0x0006] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=204, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-31 13:53:53.545 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0x0006] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2022-12-31 13:53:53.572 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD9D7](TRADFRI bulb GU10 W 400lm): Device seen - marking the device available and resetting counter
2022-12-31 13:53:53.572 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD9D7](TRADFRI bulb GU10 W 400lm): Update device availability -  device available: True - new availability: True - changed: False
2022-12-31 13:53:53.576 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xE3B5), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=57345, data=Serialized[b'\x08\xcd\n\x10\xd00\x000\xd00\x00\x00\xd0 \x00\x01\xd0\x1b\x00\x00\x00\x00\x02\xd0#\x00\x00\x00\x00\x03\xd0#\x00\x00\x00\x00\x04\xd0#\x00\x00\x00\x00\x05\xd0#\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-79)
2022-12-31 13:53:53.577 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0xe001] Received ZCL frame: b'\x08\xcd\n\x10\xd00\x000\xd00\x00\x00\xd0 \x00\x01\xd0\x1b\x00\x00\x00\x00\x02\xd0#\x00\x00\x00\x00\x03\xd0#\x00\x00\x00\x00\x04\xd0#\x00\x00\x00\x00\x05\xd0#\x00\x00\x00\x00'
2022-12-31 13:53:53.579 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0xe001] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=205, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-31 13:53:53.583 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0xe001] Decoded ZCL frame: TuyaZBExternalSwitchTypeCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0xD010, value=TypeValue(type=enum8, value=<enum8.undefined_0x00: 0>)), Attribute(attrid=0xD030, value=TypeValue(type=enum8, value=<enum8.undefined_0x00: 0>)), Attribute(attrid=0xD000, value=TypeValue(type=uint8_t, value=0)), Attribute(attrid=0xD001, value=TypeValue(type=bitmap32, value=<bitmap32.0: 0>)), Attribute(attrid=0xD002, value=TypeValue(type=uint32_t, value=0)), Attribute(attrid=0xD003, value=TypeValue(type=uint32_t, value=0)), Attribute(attrid=0xD004, value=TypeValue(type=uint32_t, value=0)), Attribute(attrid=0xD005, value=TypeValue(type=uint32_t, value=0))])
2022-12-31 13:53:53.585 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0xe001] Received command 0x0A (TSN 205): Report_Attributes(attribute_reports=[Attribute(attrid=0xD010, value=TypeValue(type=enum8, value=<enum8.undefined_0x00: 0>)), Attribute(attrid=0xD030, value=TypeValue(type=enum8, value=<enum8.undefined_0x00: 0>)), Attribute(attrid=0xD000, value=TypeValue(type=uint8_t, value=0)), Attribute(attrid=0xD001, value=TypeValue(type=b [... cut ...]
2022-12-31 13:53:53.590 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0xe001] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=205, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2022-12-31 13:53:53.591 DEBUG (MainThread) [zigpy.zcl] [0xE3B5:1:0xe001] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2023-01-03:
Is your device battery powered or mains?

> 4\. See error

I can't see any error logs in your info.

gluca, 2023-01-03:
Mains

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>

Sorry for shortness and typos

javicalle, 2023-01-03:
If it is mains powered I have no idea what can it be.
We need to have some error or evidence to know where the problem can be.

Maybe a communications problem with the device?

gluca, 2023-01-03:
I've probably solved.
I removed the device from zha, restarted home assistant and re added the
device.
Now it seems to work.

TheJulianJES, 2023-01-04:
Is the curtain switch still working? (so can we close the issue?)

closed 2023-07-10 by github-actions[bot]