zigpy/zha-device-handlers #2051: [BUG] Centralite 3325-S Not showing occupancy

aldalen, 2023-01-01:
**Describe the bug**
The CentraLite 3325-S motion sensor quirk is not exposing motion/occupancy properly. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add CentraLite 3325-S to ZHA via SkyConnect

**Expected behavior**
Occupancy is shown. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4174, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 49887,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0b05",
        "0xfc46"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "CentraLite",
  "model": "3325-S",
  "class": "zhaquirks.centralite.motion.CentraLiteMotionSensor"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.8",
    "dev": false,
    "hassio": true,
[... 454 lines cut ...]
[0x1137:1:0x0001]: Performing cluster binding
[0x1137] Extending timeout for 0xce request
[0x1137:1:0x0402]: Performing cluster binding
[0x1137] Extending timeout for 0xd0 request
[0x1137:1:0x0000]: Configuring cluster attribute reporting
[0x1137:1:0x0000]: finished channel configuration
[0x1137:1:0x0020]: Performing cluster binding
[0x1137] Extending timeout for 0xd2 request
[0x1137:1:0x0019]: finished channel configuration
[0x1137:2:0x0003]: Configuring cluster attribute reporting
[0x1137:2:0x0003]: finished channel configuration
[0x1137:2:0x0000]: Configuring cluster attribute reporting
[0x1137:2:0x0000]: finished channel configuration
Error handling '_save_attribute' event with (00:0d:6f:00:0b:c8:bb:b6, 1, 0, 4, 'CentraLite') params: FOREIGN KEY constraint failed
[0x1137:2:0x0003]: 'async_configure' stage succeeded
[0x1137:2:0x0000]: 'async_configure' stage succeeded
Error handling '_save_attribute' event with (00:0d:6f:00:0b:c8:bb:b6, 1, 0, 5, '3325-S') params: FOREIGN KEY constraint failed
```

</details>

**Additional context**
The big giveaway seems to be `unknown 263 device_type of 0xc2df profile id` .  I noticed the 3305-S quirk has the 3325-S model listed, but uses a different MFG specific key.  The 3325-S seems to be using the the base CentraLite "motion" quirk. 

Seems like some changes were made a while ago via https://github.com/zigpy/zha-device-handlers/issues/19 and https://github.com/zigpy/zha-device-handlers/issues/20.

javicalle, 2023-01-03:
Devices in `zhaquirks.centralite.motion.CentraLiteMotionSensor` quirk are also defined on the `zhaquirks.centralite.motion.CentraLite3305S`, but they have diferent device signature. This can mean that they are different devices or that they have a different firmware.

Devices with the `CentraLiteMotionSensor` would have a `IasZone` cluster, that could be a motion sensor, but doesn't have any `OccupancySensing` cluster, so no occupancy for that devices.

If you are guessing if it is possible to add it, the answer is no, it is not possible to add the cluster if the device does not send the information.

It is true that the device signature is confussing because declares a occupancy device type, but not the cluster. Maybe is reporting some info trought the `MANUFACTURER_SPECIFIC_CLUSTER_ID` but no one has implemented that functionality, so hard to know.

aldalen, 2023-01-03:
I was using this before with hubitat and it got motion/occupancy information just fine (using their "generic" driver too), so I doubt its an issue with the device.  So that leads me to believe this problem is the in the quirk.  

I don't understand your last comment there... is there a way to modify the quirk to accept altered info?

closed 2023-07-10 by github-actions[bot]