zigpy/zha-device-handlers #2053: [Device Support Request]

Ireczeq, 2023-01-01:
Immax NEO Smart Siren Zigbee 3.0
https://www.immax.eu/immax-neo-smart-siren-zigbee-3-0-p11154/

Is discovered fine by ZHA, but unfortunately no entities show up. It seems a quirk is needed...

**Describe the solution you'd like**
Entities for armed/disarmed/armed-home, battery, alarm and others if possible to show up in HA

javicalle, 2023-01-03:
If you want someone to help you, you should attach as much information as possible about the device, for example the one indicated in the `[Device Support Request]` template:
* device signature
* diagnostic information

That way we can guide you to try to get a solution to your problem.

NdK73, 2023-01-08:
Probably it's very similar to #2035 : adding its ID to the correct file will probably fix it.

Ireczeq, 2023-01-09:
<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0500",
        "0x0502",
        "0xeeff"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_b6eaxdlh",
  "model": "TS0219",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.8",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Warsaw",
    "os_name": "Linux",
    "os_version": "5.15.80",
    "supervisor": "2022.12.1",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.28.4",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    },
    "nodered": {
      "version": "1.1.2",
      "requirements": []
    }
[... 324 lines cut ...]
          "0xeeff": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

PerJarlemark, 2023-01-26:
Also having one of these, it gets detected in 2023.01.02 but no entities shown

Someone made some more descriptive information about different models of it on [this page ](https://smarthomescene.com/reviews/neo-tuya-zigbee-siren-90db-with-temperature-and-humidity-sensor/)

And t[his is the product description](https://www.aliexpress.com/item/1005004158506092.html) from seller.

closed 2023-08-01 by github-actions[bot]