zigpy/zha-device-handlers #2056: [BUG] zhaquirks/develco/smoke_alarm.py

jackydec, 2023-01-02:
**Describe the bug**
Develco smoke alarm power consumption not showing.

**To Resolve**
change MODELS_INFO: [(FRIENT, "SMSZB-120")],  (on line 44) 

to

MODELS_INFO: [(DEVELCO,"SMSZB-120"), (FRIENT,"SMSZB-120")],

javicalle, 2023-01-03:
If you have tested the implementation and is working for you, do you mind create the PR to add it to the mainstream?

You can edit the file directly from the web browser:
* https://github.com/zigpy/zha-device-handlers/edit/dev/zhaquirks/develco/smoke_alarm.py

Save your changes and follow the indications to create a PR it's a pretty straightforward process.

You would need to add the `DEVELCO` import and the right format would be:
```python
        MODELS_INFO: [(DEVELCO, "SMSZB-120"), (FRIENT, "SMSZB-120")],
```

TheJulianJES, 2023-01-04:
Thanks for creating the PRs:
- https://github.com/zigpy/zha-device-handlers/pull/2062
- https://github.com/zigpy/zha-device-handlers/pull/2063

They are both okay now, so you can close either one

javicalle, 2023-01-04:
#2062 has been closed after I closed #2063.
I have reopened #2063

I'm sorry for the inconvenience.

javicalle, 2023-01-04:
#2063 has been merged.
Closing now.

closed 2023-01-04 by javicalle