zigpy/zha-device-handlers #2059: [Device Support Request] _TZE200_v6ossqfy mmwave sensor

jimmyeao, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
_TZE200_v6ossqfy mmwave sensor is detected as a plug this is a MMwave sensor [https://zigbee.blakadder.com/Tuya_PS-HPS.html](https://zigbee.blakadder.com/Tuya_PS-HPS.html)
<details>

<summary>Device signature</summary>

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_v6ossqfy",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.9",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.80",
    "supervisor": "2022.12.1",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "matter_experimental": {
[... 378 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

jimmyeao, 2023-01-03:
Unfortunately this does not work, ZHA is ignoring my custom quirks and configuring the device using Quirk: zhaquirks.tuya.ts0601_motion.MmwRadarMotion

jimmyeao, 2023-01-03:
Ok, managed to get it added, it is only exposing Motion now, was expecting it to expose other settings and sensors..

jimmyeao, 2023-01-03:
new Pastebin of debug logs
https://pastebin.com/beN00bCZ

javicalle, 2023-01-03:
If you are using it, the `MmwRadarMotionGPP` quirk should expose 4 entities:
* OccupancySensing
* AnalogInput
* Illuminance

I don't remember the AnalogInput value meaning.
Not sure where comes from that LED switch.

Your device logs is reporting (in addition to the occupancy and analog entities) the next DP:
* dp=101, value=23
* dp=102, value=0
* dp=103, value=True (Boolean=1)

No one have mapped the DPs to their function.

> * duration_of_attendance: How long was the room occupied in minutes
> * duration_of_absence: How long was the room clear in minutes

Not sure if that are the meanings of the attributes. I believe that these attributes are the one of the `u_to_o_delay` and `o_to_u_delay` attributes (occupied <--> unoccupied):
* https://github.com/zigpy/zigpy/blob/40e470454c7c7ecc167aa83b9aae72e5b0f9f28b/zigpy/zcl/clusters/measurement.py#L172

Maybe these are the DP 101 and 102 (in seconds?) Would match with your observations?
You can try to update its values from the device view, 'Admin zigbee device', select `MmwRadarManufCluster` cluster, select the attribute (ie: `dp_101`)

First of all, read and write the current value. Then try to update with another value and check the behavior to understand the function.

jimmyeao, 2023-01-04:
> 
> * dp=101, value=23
> * dp=102, value=0
> * dp=103, value=True (Boolean=1)
> 
> No one have mapped the DPs to their function.
> 

through testing I can see that
dp_101 = occupied time
dp_102 = unoccupied time
I have been unable to identify which cluster controls the led state though, I tried setting dp_103 to Bool.false, but get a service error when I try and write the value - I suspect this may the LED setting though

jimmyeao, 2023-01-04:
> 
> * dp=101, value=23
> * dp=102, value=0
> * dp=103, value=True (Boolean=1)
> 
> No one have mapped the DPs to their function.
> 

through testing I can see that
dp_101 = occupied time
dp_102 = unoccupied time
I have been unable to identify which cluster controls the led state though, I tried setting dp_103 to Bool.false, but get a service error when I try and write the value

javicalle, 2023-01-04:
Just to be sure, are the attributes mapped as?:
* dp_101 -> pir_o_to_u_delay
* dp_102 -> pir_u_to_o_delay

(Not sure here if they are `pir` or `infrared`, 24GHz doesn't seem to match any of both)

That would be my proposal for your device:
```python
"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    GreenPowerProxy,
    Groups,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import DPToAttributeMapping, TuyaLocalCluster, TuyaNewManufCluster
from zhaquirks.tuya.mcu import TuyaMCUCluster, TuyaOnOff

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class MmwRadarManufCluster(TuyaMCUCluster):
    """Neo manufacturer cluster."""

    # # Possible DPs and values
    # presence_state: presence
    # target distance: 1.61m
    # illuminance: 250lux
    # sensitivity: 9
    # minimum_detection_distance: 0.00m
    # maximum_detection_distance: 4.05m
    # dp_detection_delay: 0.1
    # dp_fading_time: 5.0
    # ¿illuminance?: 255lux
    # presence_brightness: no control
    # no_one_brightness: no control
    # current_brightness: off

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF03: ("dp_3", t.uint32_t, True),
[... 135 lines cut ...]
                    TuyaAnalogInput,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    TuyaAnalogInput,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

There would be a new switch for the led status. If not working, enable the debug logs and attach the relevant info.

Not sure about the `AnalogInput`.

jimmyeao, 2023-01-04:
The above quirk has the device detected with a switch, however the switch will not stay toggled, and does not affect the LED state.
I don't think this bit is correct either:
```
dp_101 -> pir_o_to_u_delay
dp_102 -> pir_u_to_o_delay
```
the values are minutes (integer values, seem to max out at 60) I think the one you mention are related to sensitivity
[image]

jimmyeao, 2023-01-04:
logs are [here](https://pastebin.com/3Lq1np5j)

jimmyeao, 2023-01-04:
Should this section have definitions for dp_101 - 103?
```
 attributes. Update(
        {
            # ramdom attribute IDs
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF03: ("dp_3", t.uint32_t, True),
            0xEF04: ("dp_4", t.uint32_t, True),
            0xEF06: ("dp_6", t.enum8, True),
            0xEF69: ("dp_105", t.enum8, True),
            0xEF6A: ("dp_106", t.enum8, True),
            0xEF6B: ("dp_107", t.enum8, True),
            0xEF6C: ("dp_108", t.uint32_t, True),
        }
    )
```

jimmyeao, 2023-01-04:
If I edit your quirk to add in DP_101 and DP_102, I can query those and see the correct integer values, not sure how I add to the quirk to expose those values back to HA though
```
"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    GreenPowerProxy,
    Groups,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import DPToAttributeMapping, TuyaLocalCluster, TuyaNewManufCluster
from zhaquirks.tuya.mcu import TuyaMCUCluster, TuyaOnOff

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class MmwRadarManufCluster(TuyaMCUCluster):
    """Neo manufacturer cluster."""

    # # Possible DPs and values
    # presence_state: presence
    # target distance: 1.61m
    # illuminance: 250lux
    # sensitivity: 9
    # minimum_detection_distance: 0.00m
    # maximum_detection_distance: 4.05m
    # dp_detection_delay: 0.1
    # dp_fading_time: 5.0
    # ¿illuminance?: 255lux
    # presence_brightness: no control
    # no_one_brightness: no control
    # current_brightness: off

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF03: ("dp_3", t.uint32_t, True),
            0xEF04: ("dp_4", t.uint32_t, True),
            0xEF06: ("dp_6", t.enum8, True),
            0xEF65: ("dp_101", t.uint32_t, True),
            0xEF66: ("dp_102", t.uint32_t, True),
            0xEF69: ("dp_105", t.enum8, True),
            0xEF6A: ("dp_106", t.enum8, True),
[... 126 lines cut ...]
                    Scenes.cluster_id,
                    MmwRadarManufCluster,
                    TuyaOccupancySensing,
                    TuyaOnOff,
                    TuyaAnalogInput,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    TuyaAnalogInput,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

javicalle, 2023-01-04:
> Should this section have definitions for dp_101 - 103?

No if we want to expose as new entities.
These are attributes of the `MmwRadarManufCluster`.
DP=103 is mapped to the switch here:
```python
        103: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
```

I have edit the code to fix the switch problem.

Looking for a way to expose the dp_102 and dp_103.

javicalle, 2023-01-04:
No direct way to get the values in HA.
Analog input is only mapped for some devices:
* https://github.com/home-assistant/core/blob/22dbbd4b71cd4847aaeac52240a0c46c4822d803/homeassistant/components/zha/sensor.py#L186-L201

Probably the way would by to create custom sensors just like these:
* https://github.com/home-assistant/core/blob/22dbbd4b71cd4847aaeac52240a0c46c4822d803/homeassistant/components/zha/sensor.py#L795-L809

But that will require to create the quirk and then modify HA.

javicalle, 2023-01-04:
After looking at Z2M my final proposal would be:

<details open>
<summary>24Ghz mmWave radar</summary>

```python
"""Human Presence Sensor 24GHz."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import OccupancySensing

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster, TuyaOnOff

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class MmwRadarManufCluster(TuyaMCUCluster):
    """Human Presence Sensor 24GHz."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF65: ("duration_of_attendance", t.uint32_t, True),
            0xEF66: ("duration_of_absence", t.uint32_t, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "occupancy",
        ),
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "duration_of_attendance",
        ),
        102: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "duration_of_absence",
        ),
        103: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        103: "_dp_2_attr_update",
    }

class MmwRadarMotion_var03(CustomDevice):
    """Millimeter wave occupancy sensor."""

    signature = {
        #  endpoint=1, profile=260, device_type=81, device_version=1,
        #  input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
        MODELS_INFO: [
            ("_TZE200_0u3bj3rc", "TS0601"),
            ("_TZE200_mx6u6l4y", "TS0601"),
            ("_TZE200_v6ossqfy", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MmwRadarManufCluster,
                    TuyaOccupancySensing,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

</details>

If the quirk is working for you, then we can try to create the entities in HA.

jimmyeao, 2023-01-04:
That throws an error, just looking now
```
  File "/config/custom_zha_quirks/ts0601_mmwave.py", line 32, in <module>
    class MmwRadarManufCluster(TuyaMCUCluster):
  File "/config/custom_zha_quirks/ts0601_mmwave.py", line 57, in MmwRadarManufCluster
    103: DPToAttributeMapping(
TypeError: DPToAttributeMapping.__init__() got an unexpected keyword argument 'dp_type'
```

javicalle, 2023-01-04:
I have updated the import statement for the `DPToAttributeMapping` class.

jimmyeao, 2023-01-04:
Appreciate all your help, however it still errors

```
Error setting up entry Sonoff Zigbee 3.0 USB Dongle Plus - Sonoff Zigbee 3.0 USB Dongle Plus for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 372, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_mmwave.py", line 31, in <module>
    class MmwRadarManufCluster(TuyaMCUCluster):
  File "/config/custom_zha_quirks/ts0601_mmwave.py", line 44, in MmwRadarManufCluster
    1: DPToAttributeMapping(
TypeError: DPToAttributeMapping.__init__() missing 1 required positional argument: 'dp_type'
```

javicalle, 2023-01-04:
I have updated the rest of `dp_to_attribute` definitions.
Sorry for the troubles.

jimmyeao, 2023-01-04:
> ```python
> """Human Presence Sensor 24GHz."""
> 
> from typing import Dict
> 
> from zigpy.profiles import zha
> from zigpy.quirks import CustomDevice
> import zigpy.types as t
> from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
> from zigpy.zcl.clusters.measurement import OccupancySensing
> 
> from zhaquirks.const import (
>     DEVICE_TYPE,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PROFILE_ID,
> )
> from zhaquirks.tuya import TuyaDPType, TuyaLocalCluster, TuyaNewManufCluster
> from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster, TuyaOnOff
> 
> 
> class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
>     """Tuya local OccupancySensing cluster."""
> 
> 
> class MmwRadarManufCluster(TuyaMCUCluster):
>     """Human Presence Sensor 24GHz."""
> 
>     attributes = TuyaMCUCluster.attributes.copy()
>     attributes.update(
>         {
>             # ramdom attribute IDs
>             0xEF65: ("duration_of_attendance", t.uint32_t, True),
>             0xEF66: ("duration_of_absence", t.uint32_t, True),
>         }
>     )
> 
>     dp_to_attribute: Dict[int, DPToAttributeMapping] = {
>         1: DPToAttributeMapping(
>             TuyaOccupancySensing.ep_attribute,
>             "occupancy",
>             dp_type=TuyaDPType.ENUM,
>         ),
>         101: DPToAttributeMapping(
>             TuyaMCUCluster.ep_attribute,
>             "duration_of_attendance",
>             dp_type=TuyaDPType.VALUE,
>         ),
>         102: DPToAttributeMapping(
>             TuyaMCUCluster.ep_attribute,
>             "duration_of_absence",
>             dp_type=TuyaDPType.VALUE,
>         ),
>         103: DPToAttributeMapping(
>             TuyaOnOff.ep_attribute,
>             "on_off",
>             dp_type=TuyaDPType.BOOL,
>         ),
>     }
> 
>     data_point_handlers = {
>         1: "_dp_2_attr_update",
>         101: "_dp_2_attr_update",
>         102: "_dp_2_attr_update",
>         103: "_dp_2_attr_update",
>     }
> 
> 
> class MmwRadarMotion_var03(CustomDevice):
>     """Millimeter wave occupancy sensor."""
> 
>     signature = {
>         #  endpoint=1, profile=260, device_type=81, device_version=1,
>         #  input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
>         MODELS_INFO: [
>             ("_TZE200_0u3bj3rc", "TS0601"),
>             ("_TZE200_mx6u6l4y", "TS0601"),
>             ("_TZE200_v6ossqfy", "TS0601"),
>         ],
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaNewManufCluster.cluster_id,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             242: {
>                 # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
>                 # input_clusters=[]
>                 # output_clusters=[33]
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         },
>     }
> 
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     MmwRadarManufCluster,
>                     TuyaOccupancySensing,
>                     TuyaOnOff,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             },
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         }
>     }
> ```

No need to apologise :)
the file is working now, but still the same issue with the toggle button

jimmyeao, 2023-01-04:
<details>

```
2023-01-04 16:35:51.715 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-01-04 16:35:51.731 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=24)
2023-01-04 16:35:51.748 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x9D67, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=195, SecurityUse=<Bool.false: 0>, TimeStamp=10829651, TSN=0, Data=b'\x09\x19\x02\x00\x55\x67\x01\x00\x01\x00', MacSrcAddr=0x9D67, MsgResultRadius=29)
2023-01-04 16:35:51.748 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9D67), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t\x19\x02\x00Ug\x01\x00\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=195, rssi=None)
2023-01-04 16:35:51.748 DEBUG (MainThread) [zigpy.zcl] [0x9D67:1:0xef00] Received ZCL frame: b'\t\x19\x02\x00Ug\x01\x00\x01\x00'
2023-01-04 16:35:51.749 DEBUG (MainThread) [zigpy.zcl] [0x9D67:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=25, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-01-04 16:35:51.749 DEBUG (MainThread) [zigpy.zcl] [0x9D67:1:0xef00] Decoded ZCL frame: MmwRadarManufCluster:set_data_response(data=TuyaCommand(status=0, tsn=85, datapoints=[TuyaDatapointData(dp=103, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))]))
2023-01-04 16:35:51.749 DEBUG (MainThread) [zigpy.zcl] [0x9D67:1:0xef00] Received command 0x02 (TSN 25): set_data_response(data=TuyaCommand(status=0, tsn=85, datapoints=[TuyaDatapointData(dp=103, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))]))
2023-01-04 16:35:51.750 DEBUG (MainThread) [zigpy.zcl] [0x9D67:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=25, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-01-04 16:35:51.750 DEBUG (MainThread) [zigpy.zcl] [0x9D67:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2023-01-04 16:35:51.750 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9D67), dst_ep=1, source_route=None, extended_timeout=False, tsn=25, profile_id=260, cluster_id=61184, data=Serialized[b'\x18\x19\x0b\x02\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-01-04 16:35:51.750 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x9D67), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=25, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x19\x0B\x02\x00')
2023-01-04 16:35:51.751 DEBUG (Thread-10) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fbe08d1a740>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1672850151.748809, 'ieee': a4:c1:38:da:1f:4b:bb:28, 'min_last_seen_delta': 30.0})
2023-01-04 16:35:51.751 DEBUG (Thread-10) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fbe08d1a740>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1672850151.748809, 'ieee': a4:c1:38:da:1f:4b:bb:28, 'min_last_seen_delta': 30.0}) completed
2023-01-04 16:35:51.751 DEBUG (Thread-10) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fbe08d1a740>)
2023-01-04 16:35:51.752 DEBUG (Thread-10) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fbe08d1a740>) completed
2023-01-04 16:35:51.752 DEBUG (Thread-10) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fbe08d1a740>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (a4:c1:38:da:1f:4b:bb:28, 1, 6, 0, <Bool.false: 0>))
2023-01-04 16:35:51.752 DEBUG (Thread-10) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fbe08d1a740>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                [... cut ...]
2023-01-04 16:35:57.872 DEBUG (Thread-10) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fbe08d1a740>)
2023-01-04 16:35:57.872 DEBUG (Thread-10) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fbe08d1a740>) completed
2023-01-04 16:35:57.880 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-01-04 16:35:57.893 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=1)
```

</details>

javicalle, 2023-01-04:
Arrrrrgh!
Already fixed the `TuyaDPType` import.

jimmyeao, 2023-01-04:
Well, the good news is there is no error anymore that I can see,  however the button still doesn't do anything :(

<details>

```
2023-01-04 17:03:17.771 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x84BB, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=99, SecurityUse=<Bool.false: 0>, TimeStamp=2917995, TSN=0, Data=b'\x09\x19\x02\x00\x7F\x09\x02\x00\x04\x00\x00\x00\xDD', MacSrcAddr=0x84BB, MsgResultRadius=29)
2023-01-04 17:03:17.771 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x84BB), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t\x19\x02\x00\x7f\t\x02\x00\x04\x00\x00\x00\xdd'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=99, rssi=None)
2023-01-04 17:03:17.772 DEBUG (MainThread) [zigpy.zcl] [0x84BB:1:0xef00] Received ZCL frame: b'\t\x19\x02\x00\x7f\t\x02\x00\x04\x00\x00\x00\xdd'
2023-01-04 17:03:17.772 DEBUG (MainThread) [zigpy.zcl] [0x84BB:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=25, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-01-04 17:03:17.772 DEBUG (MainThread) [zigpy.zcl] [0x84BB:1:0xef00] Decoded ZCL frame: TuyaMmwRadarCluster:set_data_response(data=TuyaCommand(status=0, tsn=127, datapoints=[TuyaDatapointData(dp=9, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xdd\x00\x00\x00', *payload=221))]))
2023-01-04 17:03:17.772 DEBUG (MainThread) [zigpy.zcl] [0x84BB:1:0xef00] Received command 0x02 (TSN 25): set_data_response(data=TuyaCommand(status=0, tsn=127, datapoints=[TuyaDatapointData(dp=9, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\xdd\x00\x00\x00', *payload=221))]))
2023-01-04 17:03:17.773 DEBUG (MainThread) [zigpy.zcl] [0x84BB:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=25, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-01-04 17:03:17.773 DEBUG (MainThread) [zigpy.zcl] [0x84BB:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2023-01-04 17:03:17.773 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x84BB), dst_ep=1, source_route=None, extended_timeout=False, tsn=25, profile_id=260, cluster_id=61184, data=Serialized[b'\x18\x19\x0b\x02\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-01-04 17:03:17.773 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x84BB), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=25, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x19\x0B\x02\x00')
2023-01-04 17:03:17.773 DEBUG (Thread-11) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fbdab4a4840>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1672851797.771991, 'ieee': dc:8e:95:ff:fe:59:b8:95, 'min_last_seen_delta': 30.0})
2023-01-04 17:03:17.774 DEBUG (Thread-11) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fbdab4a4840>, 'UPDATE devices_v11\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_last_seen_delta', {'ts': 1672851797.771991, 'ieee': dc:8e:95:ff:fe:59:b8:95, 'min_last_seen_delta': 30.0}) completed
2023-01-04 17:03:17.775 DEBUG (Thread-11) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fbdab4a4840>)
2023-01-04 17:03:17.775 DEBUG (Thread-11) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fbdab4a4840>) completed
2023-01-04 17:03:17.775 DEBUG (Thread-11) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fbdab4a4840>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (dc:8e:95:ff:fe:59:b8:95, 1, 12, 85, 221))
2023-01-04 17:03:17.775 DEBUG (Thread-11) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fbdab4a4840>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.va [... cut ...]
2023-01-04 17:03:25.005 DEBUG (Thread-11) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fbdab4a4840>)
2023-01-04 17:03:25.005 DEBUG (Thread-11) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fbdab4a4840>) completed
2023-01-04 17:03:25.019 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-01-04 17:03:25.039 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=94)
```

</details>

jimmyeao, 2023-01-04:
to be 100% clear, I toggle the button, and after a few seconds it slides back again with no results.

javicalle, 2023-01-04:
Another try:
I have replaced the `TuyaOnOff` references for `TuyaOnOffNM` (3 in total).
Could you try with the new version?

jimmyeao, 2023-01-04:
> Another try: I have replaced the `TuyaOnOff` references for `TuyaOnOffNM` (3 in total). Could you try with the new version?

Genius, thats the switch working :) now to just add those values to HA!

javicalle, 2023-01-04:
Are you able to edit the HA classes? As custom components or editing directly the container?

I'll come back later to see how can be done.
Do you confirm that these DP values are time in minutes, right?

jimmyeao, 2023-01-04:
> Are you able to edit the HA classes? As custom components or editing directly the container?
> 
> I'll come back later to see how can be done. Do you confirm that these DP values are time in minutes, right?

Yes they are minutes - not sure how to edit the classes?

Sakul85, 2023-01-13:
I also got this sensor and would be very happy about an integration into ZHA as well. 

Unfortunately my knowledge is very limited but I could support with testing and providing logs :-)

javicalle, 2023-01-15:
> Any luck with exposes those two values @javicalle ?

I haven't had time to check that (or any other) part.

TheJulianJES, 2023-01-23:
Is the [quirk from this comment](https://github.com/zigpy/zha-device-handlers/issues/2059#issuecomment-1370892759) still the latest (and ready to be merged into zha-quirks)?

alexvaltchev, 2023-01-23:
I am new here and I have the same sensor. I use ZHA, how do I manage to make it work, or there is still no work at this time? I am new here and new to HA and dont know programing. How do I apply a fix and make it work? thanks

TheJulianJES, 2023-01-23:
Paste the content from [this comment](https://github.com/zigpy/zha-device-handlers/issues/2059#issuecomment-1370892759) into a file called `ts0601_sensor_custom.py` for example.

Then, create a folder called `custom_zha_quirks` in your `/config` directory for Home Assistant and put the file in that folder. Register that folder in your `configuration.yaml` and restart HA.

See https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 for how to register that folder.
Also, follow this issue as this will (hopefully) be merged into HA in the future and you'd want to remove the custom quirk at that point.

jimmyeao, 2023-01-23:
That quirk doesnt expose all the sensors though - there is no "time occupied" for, for example. Only presence.

alexvaltchev, 2023-01-23:
> Paste the content from [this comment](https://github.com/zigpy/zha-device-handlers/issues/2059#issuecomment-1370892759) into a file called `ts0601_sensor_custom.py` for example.
> 
> Then, create a folder called `custom_zha_quirks` in your `/config` directory for Home Assistant and put the file in that folder. Register that folder in your `configuration.yaml` and restart HA.
> 
> See [#693 (comment)](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274) for how to register that folder. Also, follow this issue as this will (hopefully) be merged into HA in the future and you'd want to remove the custom quirk at that point.

I created a directory called "custom_zha_quirks" inside the "config" folder, and created a file called "ts0601_sensor_custom.py" and pasted all the code from the comment. The only thing left is to register the folder inside the configuartion.yaml file. What is the code I need to put there to register it. I cant find it online. Sorry I am newbie...

jimmyeao, 2023-01-23:
> I created a directory called "custom_zha_quirks" inside the "config" folder, and created a file called "ts0601_sensor_custom.py" and pasted all the code from the comment. The only thing left is to [...]

```
zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks/ 
```

alexvaltchev, 2023-01-23:
Thank you. I restarted home assistant and now shows this: https://prnt.sc/uKTN2BueFis0
Is this what is supposed to happen? It shows as a switch (on) possibly when there is a person in the room and (off) when can't detect anything?

What is the other sensor that you are talking about that are not exposed? Time value (how long is On or OFF) I guess? And how far to be able to detect (as in the original Smart Life app?) Thank you all!

jimmyeao, 2023-01-23:
That looks correct, the switch is turn the LED off
This device also exposes how long a room has been occupied or unoccupied for (0-60minutes) but the quirk doesn't capture that yet

alexvaltchev, 2023-01-23:
OK OK, I get it. It is great workaround for the time being. I hope a full implementation will be done in the next few months. How do I know when to remove the quirk from the configuration.yaml ?   I just need to keep checking here, I guess?

What else can be implemented to this sensor? I guess how far the sensor to reach like (distance: 5 meters for example) I don't remember what else was in the original Smart Life app, but this does the job for the time being. I am very appreciative of everyone here involved in this post. Thank you all!!!

javicalle, 2023-01-23:
The quirk is managing the values but they aren't exposed in HA. You can manage it as cluster attributes.

jimmyeao, 2023-01-23:
> The quirk is managing the values but they aren't exposed in HA. You can manage it as cluster attributes.

I haven't had to do that before, can you give me some pointers please?

alexvaltchev, 2023-01-23:
What does "not exposed to HA" means? HA can't see when the sensor report occupancy "detected" vs "not detected" and I cannot automate this? OR I am getting it wrong? I don't know what is cluster attributes. I think someone mentioned it in my original post. Will go to read now, but if simple explanation is possible, I would greatly appreciate it.

MattWestb, 2023-01-23:
From the device card right from _reconfigure_ `...` and `manage zigbee device` select cluster and command / attribute and read / write. If sleeper wake it up then sending to the device.

"not exposed to HA"  is wrong its exposed but ZHA have not getting any GUI entries for it.

jimmyeao, 2023-01-23:
> From the device card right from _reconfigure_ `...` and `manage zigbee device` select cluster and command / attribute and read / write. If sleeper wake it up then sending to the device.
> 
> "not exposed to HA" is wrong its exposed but ZHA have not getting any GUI entries for it.

Ok, it would be nicer if they were exposed as sensors as Z2M does...

alexvaltchev, 2023-01-23:
I also wonder when I get into the room, it changes from "clear to detected" in about 1-2 seconds. But when I walked out of the room, it takes about 60-65 seconds. Why is this? Can it be done to be faster or immediate? I don't know if I am mistaken, but I think in  Home Smart app, both were right on the spot, no delay. I can check now and report back.

jimmyeao, 2023-01-23:
> I also wonder when I get into the room, it changes from "clear to detected" in about 1-2 seconds. But when I walked out of the room, it takes about 60-65 seconds. Why is this? Can it be done to be [...]

MMWave sensors are not very good at reporting "clear" quickly - i have other sensors that have a configurable time, but this one does not appear to do so

alexvaltchev, 2023-01-23:
Which one would that be? I have only seen this one, and Aqara FP1s. What would you recommend? Thanks.

jimmyeao, 2023-01-23:
> Which one would that be? I have only seen this one, and Aqara FP1s. What would you recommend? Thanks.

This One [https://www.aliexpress.com/item/1005004704966668.html?spm=a2g0o.order_list.order_list_main.5.8ae818028TXdBU](url)

alexvaltchev, 2023-01-23:
The USA Wall mounted one (square) or the ceiling version (round)?

jimmyeao, 2023-01-23:
I got the wall mounted one, 5v

alexvaltchev, 2023-01-23:
Thanks. Does it work out of the box with ZHA? No need for additional quirks and device handles are extra steps?

javicalle, 2023-01-23:
The attributes can be 'exposed' in HA creating a new ZHA sensor like this:
```python
@MULTI_MATCH(
    channel_names="tuya_manufacturer",
    manufacturers={
        "_TZE200_0u3bj3rc",
        "_TZE200_mx6u6l4y",
        "_TZE200_v6ossqfy",
    },
)
class AttendanceTime(Sensor, id_suffix="duration_of_attendance"):
    """Sensor that displays the attendance time."""

    SENSOR_ATTR = "duration_of_attendance"
    _attr_device_class: SensorDeviceClass = SensorDeviceClass.DURATION
    _attr_icon = "mdi:timer"
    _attr_name: str = "Attendance time"
    _attr_native_unit_of_measurement = UnitOfTime.MINUTES
```

It's just the same approach that is already implemented here:
https://github.com/home-assistant/core/blob/7d1dec8d315702020c5bb23bf0505bac7d584257/homeassistant/components/zha/sensor.py#L878-L891

Edit to add the code in the `zha/sensor.py` file or configure as a `custom_component` in HA.
The `duration_of_absence` can be done the same way.

PS: probably sensor names can get proper names.

alexvaltchev, 2023-01-24:
@javicalle  this what is this new code above? Should I replace those lines in my original ts0601_sensor_custom.py ? I am getting confused. Can you please explain in simple English what does this do, and how to implement it. I do not see ZHA directory with a sensor.py file. Sorry for being lame, I am just very new and need step by step guide. Thank you.

javicalle, 2023-01-24:
> what is this new code above?

Jimmy was asking me how to expose the cluster attributes in HA, and that code shows how can be done.

> Should I replace those lines in my original ts0601_sensor_custom.py ?

No, that code is for update the ZHA integration.

> Can you please explain in simple English what does this do, and how to implement it. I do not see ZHA directory with a sensor.py file. Sorry for being lame, I am just very new and need step by step guide.

Most of the code is untested and prone to bugs, so it needs to be tested by someone who knows what they're doing so we can focus on the code.
If you're not familiar with HA terminology and code, I think it's best to wait for the final version to be available on HA.

alexvaltchev, 2023-01-24:
Yes, I have very basic knowledge, and agree would be better to wait until is implemented into HA. How long does it take for something like that to get official support, as I see you guys have been working already for a while. Thank you.

alexvaltchev, 2023-01-25:
Guys I got a problem. The sensor was working it was taking about 1-2 second to change from "clear" to "detected" and about 65 seconds to from "detected" to "clear" and in Home Assistant. I then paired it with Smart Home app to check how is there and to see the other options available, and it got stuck on "detected" and now it never changes" I removed it from Smart Life and re-added, but does not work. I removed it and added into Home Assistant and still does not work. All it shoes is "detected" and even after 1 hour in the closet alone, still shows that. I tried to press fast the reset button 10 times, or to keep it on for 10s or other methods and nothing helps. How can I reboot the sensor to start working again ? Please help

javicalle, 2023-02-05:
Sorry for the late response.
Could you attach the diagnosis info from the device?

skycryer, 2023-03-02:
Does that still work for you with aha in Home Assistant 2023.3.0? Seems to have a problem with the TuyaDPType stuff what I read in a different post.

besiktas97, 2023-03-07:
> Does that still work for you with zha in Home Assistant 2023.3.0? Seems to have a problem with the TuyaDPType stuff what I read in a different post.

I've got the same thing and isnt working for me. Even after I have deleted the lines for it.

javicalle, 2023-03-08:
Check this:
https://github.com/zigpy/zha-device-handlers/issues/2250#issuecomment-1453978417

besiktas97, 2023-03-08:
> Check this:
> 
> https://github.com/zigpy/zha-device-handlers/issues/2250#issuecomment-1453978417
> 
> 
> 
> 

I did exactly the same like you are commenting, but didnt work. I will try this again and let you know. Thanks.

[13 more replies]

closed 2023-10-29 by jimmyeao