zigpy/zha-device-handlers #2061: [Device Support Request] Earu EAYCBM-Z

pkejval, 2023-01-03:
Device: Earu EAYCBM-Z 32A
bought at: https://www.aliexpress.com/i/1005004894129906.html

Device didn't expose any attribute nor switch in ZHA integration. Tried some custom quirks for din rail power meters TS0601 but none is working. :(

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_davzgqq0",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.9",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "5.15.80",
    "supervisor": "2022.12.1",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "nodered": {
      "version": "1.1.2",
      "requirements": []
    },
    "weatherbit": {
      "version": "1.0.17",
      "requirements": [
        "pyweatherbitdata==1.0.15"
      ]
[... 482 lines cut ...]
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

TheJulianJES, 2023-01-06:
Might not work, but I guess it's worth a try (put this in a file and try it as a [custom quirk](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274)):

<details>
<summary>Possible quirk code</summary>

```python

"""Quirk for _TZE204_davzgqq0 TS0601."""

from zigpy.profiles import zha 
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    Scenes,
    Time,
 
)
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    MoesSwitchManufCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
)

class Tze204Davzgqq0Ts0601(CustomDevice):
    """_Tze204_Davzgqq0 TS0601 custom device implementation."""

    signature = {
        MODELS_INFO: [("_TZE204_davzgqq0", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ] 
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ] 
            }
        }
    }
```

</details>

pkejval, 2023-01-06:
It's unfortunately not working. Just exposes switch which is still in on state in HA regardless actual state of device and clicking it does nothing.

[image]

TheJulianJES, 2023-01-06:
Hmm. Try replacing both occurrences of ``TuyaOnOff`` with ``TuyaOnOffNM`` and see if that makes a difference.
(One is in the imports and the other one is in replacement/ENDPOINTS/1/INPUT_CLUSTERS)

pkejval, 2023-01-06:
Still the same.

[image]

```
"""Quirk for _TZE204_davzgqq0 TS0601."""

from zigpy.profiles import zha 
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    Scenes,
    Time,
 
)
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    MoesSwitchManufCluster,
    TuyaOnOffNM,
    TuyaOnOffManufCluster,
)

class Tze204Davzgqq0Ts0601(CustomDevice):
    """_Tze204_Davzgqq0 TS0601 custom device implementation."""

    signature = {
        MODELS_INFO: [("_TZE204_davzgqq0", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ] 
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ] 
            }
        }
    }
```

MattWestb, 2023-01-06:
Try using the EnchantedDevice device class:
https://github.com/zigpy/zha-device-handlers/blob/22968954cacaff4fc2b00e02ca1bd5622f637393/zhaquirks/tuya/mcu/__init__.py#L694

pkejval, 2023-01-06:
I hope I used that EnchantedDevice class well - code below. But it's still not working.

<details>
<summary>Quirk code</summary>

```yaml
"""Quirk for _TZE204_davzgqq0 TS0601."""
import asyncio

from zigpy.profiles import zha 
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    Scenes,
    Time,
 
)
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    MoesSwitchManufCluster,
    TuyaOnOffNM,
    TuyaOnOffManufCluster,
)

class EnchantedDevice(CustomDevice):
    """Class for enchanted Tuya devices which needs to be unlocked by casting a 'spell'."""

    def __init__(self, *args, **kwargs):
        """Initialize with task."""
        super().__init__(*args, **kwargs)
        self._init_device_task = asyncio.create_task(self.spell())

    async def spell(self) -> None:
        """Initialize device so that all endpoints become available."""
        attr_to_read = [4, 0, 1, 5, 7, 0xFFFE]
        basic_cluster = self.endpoints[1].in_clusters[0]
        await basic_cluster.read_attributes(attr_to_read)

class Tze204Davzgqq0Ts0601(EnchantedDevice):
    """_Tze204_Davzgqq0 TS0601 custom device implementation."""

    signature = {
        MODELS_INFO: [("_TZE204_davzgqq0", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ] 
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ] 
            }
        }
    }
```
</details>

MattWestb, 2023-01-06:
You can importing the  `EnchantedDevice` class and you dont need putting the code in your quirk.
Meny new devices need it for working OK and also changing device cluster and endpoints functions.

If you have good debug logging set you can see if the system is sending the attribute reads to the device (is only working then device is new joining).

I dont  have the knowledge of coding then im not one code warier ;-((

pkejval, 2023-01-07:
I have logging set up like this. Will be helpful for anyone of you when I upload home-assistant.log here?

<details>
<summary>configuration.yaml</summary>

```yaml
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```

pkejval, 2023-04-20:
Zigbee2MQTT has similar request: https://github.com/Koenkk/zigbee2mqtt/issues/16118

They found out datapoints but I don't know how to translate them into ZHA quirk code. Can anyone help me with that?

```
tuyaDatapoints: [
          [1, 'energy', tuya.valueConverter.divideBy100],
          [6, null, tuya.valueConverter.phaseVariant2],
          [16, 'state', tuya.valueConverter.onOff],
          // {"type":"Buffer","data":[4,1,0,30]} - leakage enabled
          // {"type":"Buffer","data":[4,0,0,30]} - leakage disabled
          [17, 'leakage_threshold', valueConverter.thresholdVariant1({4: 4}, {'ON': 1, 'OFF': 0})],
          //[17, 'leakage_threshold', valueConverter.threshold2],
          // {"type":"Buffer","data":[1,1,0,30,3,0,0,250,4,0,0,150]} - overcurr enabled 30A | overvolt enabled 250V | undervolt enabled 150V
          // {"type":"Buffer","data":[1,0,0,21,3,0,0,250,4,0,0,150]} - overcurr disabled 21A | overvolt disabled 250V | undervolt disabled 150V
          [18, 'thresholds', valueConverter.thresholdVariant2(['overcurrent', 'overvoltage', 'undervoltage'], {1: 1, 3: 3, 4: 4}, {'ON': 1, 'OFF': 0})],
          //[18, 'overcurrent_threshold', valueConverter.thresholdVariant2([], {1: 1}, {'ON': 1, 'OFF': 0})],
          //[18, 'overvoltage_threshold', valueConverter.thresholdVariant2([], {3: 3}, {'ON': 1, 'OFF': 0})],
          //[18, 'undervoltage_threshold', valueConverter.thresholdVariant2([], {4: 4}, {'ON': 1, 'OFF': 0})],
          [19, 'breaker_id', tuya.valueConverter.raw],
      ],
```

TolianIPB, 2023-05-21:
Add support pleaze.

alex-kms, 2023-05-26:
Are there those who do not use the device in ZHA, but can integrate it into ZHA ?

XpycTee, 2023-06-04:
Hi, I made a working custom quick using attributes from Hiking Power Meter

data about voltage, current and power is transmitted in the attribute 0x0006.

---v---------mA---------W---
[08 98] 00 [1A A2] 00 [05 DC]

v/10=RMS voltage
mA = RMS current
W = Active power

The status of the switch in the attribute 0x0110
Total Energy Delivered kWh in attr 0x0201

[ts0601_din_power.py.txt](https://github.com/zigpy/zha-device-handlers/files/11645776/ts0601_din_power.py.txt)

lukyrys, 2023-08-23:
Is there any way to set last state after power outage?

pkejval, 2023-08-24:
@XpycTee Thank you very much! Your quirk is working great.

@lukyrys I'm thinking about it right now. Maybe some HA automation?

lukyrys, 2023-08-24:
@pkejval I don't think it would be good because if these devices are connected to the start of power line and there is a power outage, what will turn it on. I think the version with wifi and that application can turn it on in the device settings. i dont know if this is available if is zigbee controlled by tuya maybe yes

MattWestb, 2023-08-24:
Zigbee standard is having attribute that can being set for it but in "normal tuya zigbee devices" is not  zigbee standard but we is adding it  to then on the cluster.
tuya MCU devices is completely different then its not using Zigbee standard at all only piping commands thru Ziogbee so all must being implanted with  DPs for it  shall working OK.

agnus777, 2024-01-05:
Hi All,

Has anyone managed to enable the OnOff function to be able to set the status upon reboot?

pkejval, 2024-01-09:
@agnus777 I managed to make ON/OFF working with this quirk in `/config/zha_quirks/ts0601_din_power.py`
Then turn it ON in Automations if is OFF.

```
"""Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch

TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214
TUYA_DIN_SWITCH_ATTR = 0x0101

SWITCH_EVENT = "switch_event"

"""Hiking Power Meter Attributes"""
HIKING_DIN_SWITCH_ATTR = 0x0110
HIKING_TOTAL_ENERGY_DELIVERED_ATTR = 0x0201
HIKING_TOTAL_ENERGY_RECEIVED_ATTR = 0x0266
HIKING_VOLTAGE_CURRENT_ATTR = 0x0006
HIKING_POWER_ATTR = 0x0267
HIKING_FREQUENCY_ATTR = 0x0269
HIKING_POWER_FACTOR_ATTR = 0x026F
HIKING_TOTAL_REACTIVE_ATTR = 0x026D
HIKING_REACTIVE_POWER_ATTR = 0x026E

"""Earu Power Meter Attributes"""
EARU_TOTAL_ENERGY_DELIVERED_ATTR = 0x0201
EARU_VOLTAGE_CURRENT_POWER_ATTR = 0x0006
EARU_DIN_SWITCH_ATTR = 0x0110

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint32_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_deliver_reported(value / 100)
        elif attrid == TUYA_CURRENT_ATTR:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.power_reported(value / 10)
        elif attrid == TUYA_VOLTAGE_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(value / 10)
        elif attrid == TUYA_DIN_SWITCH_ATTR:
            self.endpoint.device.switch_bus.listener_event(
                SWITCH_EVENT, self.endpoint.endpoint_id, value
[... 273 lines cut ...]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    EaruManufClusterDinPower,
                    TuyaElectricalMeasurement,
                    TuyaPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            16: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

```

lukyrys, 2024-01-10:
@pkejval does it solve auto power on after power failure ? i mean same config which is on my water valve 
[image]

pkejval, 2024-01-10:
No it doesn't because the device can't do it as you noticed in the official
tuya app. Only way out is Tuya or Home Assistant automation. I have
automation running every minute to check state:

- id: "1692860675855"
  alias: Automatizace - keep breakers on
  description: ""
  trigger:
    - platform: template
      value_template:
        '{{ (not is_state("switch.jistic_1", "on")) or (not is_state(
"switch.jistic_2",
        "on")) or (not is_state("switch.jistic_3", "on")) }}'
      alias: Not all are ON
    - platform: time_pattern
      minutes: /1
  condition:
    - condition: template
      value_template:
        '{{ (not is_state("switch.jistic_1", "on")) or (not is_state(
"switch.jistic_2",
        "on")) or (not is_state("switch.jistic_3", "on")) }}'
  action:
    - service: switch.turn_on
      data: {}
      target:
        entity_id:
          - switch.jistic_1
          - switch.jistic_2
          - switch.jistic_3
  mode: single

agnus777, 2024-01-10:
@pkejval I will try this automation! thanks a lot for the nice work done!.

perhaps this function is not integrated due to some regulations on the operation of RCD breakers... I don't know.
in any case I wrote to the seller on aliexpress, which is the official Earu store, if there is a way to set this or not

I will post update if any

closed 2024-07-15 by github-actions[bot]