zigpy/zha-device-handlers #2064: [Device Support Request] Innr RF 263 / RF 264

Steephsel, 2023-01-04:
Hi all, I've been having troubles with my Innr RF 263 and Innr RF 264. Some are and some are not working correctly, probably because they are not officially supported (yet? ;)) Beauce I am cheap and didn't want to pay a huge amount on original Hue filament lights I bought Innr, 18 pcs of them. Would save me a lot of money and trouble if they would be supported!

**Is your feature request related to a problem? Please describe.**
Partially yes, as described above some are actually working and others are not and going unavailable at random. 
As shown on the screenshot below, those that haven't been bugging out (for now) show energy sensors, those failing don't.
[image]

**Describe the solution you'd like**
Add support for them or perhaps point me to a custom Quirk.

<details>

<summary>Device signature RF 263</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4454, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0b05",
        "0x1000",
        "0xfc82"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "innr",
  "model": "RF 263",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Device signature RF 264</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4454, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
[... 1514 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

dmulcahey, 2023-01-04:
These work OOTB. Please add more details for the issues you are experiencing. If it’s just that some are going unavailable you more likely than not have mesh issues.

Steephsel, 2023-01-04:
> These work OOTB. Please add more details for the issues you are experiencing. If it’s just that some are going unavailable you more likely than not have mesh issues.

I checked on https://zigbee.blakadder.com/bulbs.html, they shouldnt be supported in ZHA. 
[image]

I never had mesh issues with Z2M/Raspbee before going to ZHA/SkyConnect+extension cable. The bulbs themselves are powered continuesly and plenty of other devices around them. Its only some bulbs that go unavailable, at random it seems. I have 3 lamp fixtures with 6, 6 and 4 bulbs each.

tube0013, 2023-01-04:
The list you link is based on user submissions. ZHA works spec compliant devices so it may not be explicitly listed if no one has submitted it there

[image]

Steephsel, 2023-01-04:
> The list you link is based on user submissions. ZHA works spec compliant devices so it may not be explicitly listed if no one has submitted it there
> 
> [image]

So in that case it would rather be a bug if they keep going unavailable? I'm having trouble with multiple devices (see other topics..) maybe it's just ZHA and/or SkyConnect not being able to handle them somehow? Either way I'd still like to use my lamp! Because I don't even have my fixture connected to the dumb power switch - making it always on - when the lights go stuck I have to get a ladder and unscrew the bulb. 😭

TheJulianJES, 2023-01-04:
ZHA supports all compliant devices out-of-the-box (while other integrations like Z2M need explicit support for every device). So for Z2M there's a concrete list on which devices work. ZHA doesn't have this, as many "just work" (if they're not Tuya-based) and so users don't report them as "working". Innr, IKEA, Hue bulbs, and many others don't need any quirks at all to work.

> So in that case it would rather be a bug if they keep going unavailable?

Yes. It's some network issue then. When you migrated from Z2M to ZHA, did you restore the backup in ZHA, or did you create a clean network with the SkyConnect?
Also, on the ZHA config page, what channel does it show? There might be some interference issues with your WiFi for example (2.4 GHz only). See this graphic: https://support.metageek.com/hc/article_attachments/115016160948/ZigBee_Channels.png

TheJulianJES, 2023-01-04:
Also, are you using the included USB extension cable for the SkyConnect? (It helps to prevent interference (from USB 3 ports, APs, ...)

Steephsel, 2023-01-05:
> ZHA supports all compliant devices out-of-the-box (while other integrations like Z2M need explicit support for every device). So for Z2M there's a concrete list on which devices work. ZHA doesn't [...]

ZHA is on Channel 15, I checked the Wifi channels with a wifi scanner and amended accordingly because I was familiar with this issue. Will let u know when I get home from work again 🤣  

> Also, are you using the included USB extension cable for the SkyConnect? (It helps to prevent interference (from USB 3 ports, APs, SSDs, ...)

Yes I am using the cable because I already knew this could be an issue

Steephsel, 2023-01-08:
So just a quick update, I removed and readded my most buggy lights from one of my fixtures and during the process i noticed some IEEE addresses starting with different sets, which also resulted in different options being added to that bulb. 

[image]

[image]

Curious to see if they go unavailable now after readding and if those are the one with lesser options!

TheJulianJES, 2023-01-08:
Try reconfiguring all bulbs, restart HA, and see if more options show up.
(Otherwise, I'd recommend re-pairing the ones with missing options again and then reconfigure them just to be safe.)

It looks the like the interview didn't really complete fully for some reason(?)

closed 2023-07-16 by github-actions[bot]