zigpy/zha-device-handlers #2072: [Device Support Request] Radiator Valve Tuya/CloudEven TS0601 Type _TZE200_0hg58wyk [Device Support Request]

HAuser1234, 2023-01-06:
**I have baught a Cloud Even TRV and discovered that there is no support yet in ZHA**
I would be very glad if you can implement this TRV in a Quirk!

I have attached:
- device signature
- Diagnostic information
- Device discovery/binding logs

Link to the TRV 
https://www.ebay.de/itm/155337234865?hash=item242ad219b1:g:HxAAAOSwlhZjsVQg&amdata=enc%3AAQAHAAAAoISDpicDt6I%2Fe791mFplU2NF1exk2zrxAshFQvAlbZ73dvXwmzCsD7m5Vged3vjHI4dCOyQo6przUfAkAmWmZGrFvWKL3Ki013kZIqUSnolu8fDvDMWmi4P%2B3pmkd4utooM6pM1DPFgkijhCSClt8grjc2blAWUQo3%2B6b3VUr2W2uzvkrBG460g2Ee8xk%2FOpZZMWi5RR4i5HR2%2FO11zBRRs%3D%7Ctkp%3ABk9SR7Cts4SvYQ#rpdCntId

If you want any further information I am happy to provide that

<details>

<summary>Device signature</summary>

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_0hg58wyk",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
</details>

<details>

<summary>Diagnostic information</summary>
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.1.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.10.0-13-amd64",
    "supervisor": "2022.12.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.21",
    "chassis": "desktop",
    "run_as_root": true
  },
  "custom_components": {
    "indego": {
      "version": "3.8.1",
      "requirements": [
        "pyIndego==2.0.30"
      ]
    },
[... 377 lines cut ...]
[0x6207](TS0601): power source: Battery or Unknown
[0x6207](TS0601): completed initialization
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6207), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=10, data=Serialized[b'\x00\x16\x00\x07\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=87, rssi=None)
[0x6207:1:0x000a] Received ZCL frame: b'\x00\x16\x00\x07\x00'
[0x6207:1:0x000a] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=22, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0x6207:1:0x000a] Decoded ZCL frame: Time:Read_Attributes(attribute_ids=[7])
[0x6207:1:0x000a] Received command 0x00 (TSN 22): Read_Attributes(attribute_ids=[7])
[0x6207:1:0x000a] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=22, command_id=<GeneralCommand.Read_Attributes_rsp: 1>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x6207:1:0x000a] Sending reply: Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0007, status=<Status.SUCCESS: 0>, value=TypeValue(type=LocalTime, value=726344484))])
[0x1E94](DS01): last_seen is 11511839.15874052 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0x1E94](DS01): Update device availability -  device available: False - new availability: False - changed: False
[0xB09E](MS01): last_seen is 1457661.506999731 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0xB09E](MS01): Update device availability -  device available: False - new availability: False - changed: False
[0xC15D](DS01): Device seen - marking the device available and resetting counter
[0xC15D](DS01): Update device availability -  device available: True - new availability: True - changed: False
[0xE73B](TH01): Device seen - marking the device available and resetting counter
[0xE73B](TH01): Update device availability -  device available: True - new availability: True - changed: False
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6207), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\x19\x17$\x00\t'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=87, rssi=None)
[0x6207:1:0xef00] Received ZCL frame: b'\x19\x17$\x00\t'
[0x6207:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=23, command_id=36, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x6207:1:0xef00] Unknown cluster command 36 b'\x00\t'
[0x6207:1:0xef00] Received command 0x24 (TSN 23): b'\x00\t'
[0x6207:1:0xef00] No explicit handler for cluster command 0x24: b'\x00\t'

</details>

MattWestb, 2023-01-06:
That is the only tuya TRV family that no one have getting working and its 4 identical devices that need one new quirk.
https://github.com/zigpy/zigpy/discussions/653#

HAuser1234, 2023-01-06:
is there any chance that this quirk can be created remotely? (By using the diagnostics info)

MattWestb, 2023-01-06:
Its depends of the devs some can doing  it very good.

@jacekk015 do you have time for taking one look on one old new tuya TRV ??
Its first gen TRV but not getting one quirk so some user have it cold in there homes :-))

MattWestb, 2023-01-06:
## By the way welcome to GitHub and ZHA !!!

HAuser1234, 2023-01-06:
thank you!
I would be very glad if someone could help. Maybe this could also be benefical to other users since this is a whole unsupportet TRV family.

MattWestb, 2023-01-06:
All first gen tuya TRVs was getting little limited functions but we have getting extended quirks for most of them but this was not getting any basic or extended then no devs and no users was having the interest but i hope its at least we can getting basic support for it.

HAuser1234, 2023-01-06:
ok, i would anyways only need to set the temperature and maybe turn it on or off.
The rest would be done in Home Assistant anyways

MattWestb, 2023-01-06:
Thats i have with my Siterwell TRV with the basic quirk and its working OK but is always nice getting all functions of the device then its possible.

The kitchen TRV is holding the temperature in one °C so its working very well :-)))

HAuser1234, 2023-01-06:
yes additional functions would be great but I would be more than happy just to set the temperature through zigbee :)

jacekk015, 2023-01-08:
Will look on this. It seems that support in Z2M is available, but as a different device.
Blakadder added it as E-TOP
https://zigbee.blakadder.com/ETOP_HT-10.html

MattWestb, 2023-01-08:
From tuya TRV matrix:

<html>
<body>
<p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">E-top is making different software IDs for all customers.</p>

OEM / Brand | Model | Manufacturer Name | Model ID | ZHA Quirk | Request
-- | -- | -- | -- | -- | --
E-top | HT-10R-Z | _TZE200_2dpplnsn | TS0601 | ⭕ Moes I ? |  
E-top | HT-10R-Z | _TYST11_2dpplnsn | dpplnsn | ⭕ Moes II ? |  
Immax | 07703L | _TZE200_wlosfena | TS0601 | ⭕ Moes I ? | 691
Immax | 07703L | _TYST11_wlosfena | losfena | ⭕ Moes II ? | 691
Cloud Even | ? | _TZE200_0hg58wyk | TS0601 | ⭕ Moes I ? |  

<p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><g-emoji class="g-emoji" alias="o" fallback-src="https://github.githubassets.com/images/icons/emoji/unicode/2b55.png" style="box-sizing: border-box; display: inline-block; min-width: 1ch; font-family: &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;, &quot;Segoe UI Symbol&quot;; font-size: 1em; font-weight: var(--base-text-weight-normal, 400); line-height: 1; vertical-align: -0.075em; font-style: normal !important;">⭕</g-emoji><span> </span>= Not implanted: look in the last section "Adding support for not supported device"</p>
</body>

HAuser1234, 2023-01-08:
Thank you very much for looking into this :)

HAuser1234, 2023-01-12:
Hello, are there any updates on this topic? otherwise I will probably return the TRVs to the seller soon. The deadline for returning is over soon.
thank you in advance

jacekk015, 2023-01-12:
I do it in a free time, so there's no strict deadline. You decide.

HAuser1234, 2023-01-13:
ok I think I will just wait a little bit longer. If you have spare time until then I would be happy if you can look into this if not it is of course also no problem.
I just wanted to know if there is any chance that this can be realised. If there were none I would have send it back now. 
So thank you again for the help an I am looking forward to a potential solution :)

jacekk015, 2023-01-18:
First version of the quirk.
Bugs possible, so enable DEBUG logging and post the logs in case of errors.
BTW post the logs of pairing process and 10 minutes after, using quirk below.
[ts0601_trv_etop.py.zip](https://github.com/zigpy/zha-device-handlers/files/10451792/ts0601_trv_etop.py.zip)

HAuser1234, 2023-01-19:
Hi first thank you very much !!

I can successfully now set the Temperature and turn it off or on.
The battery state isn't successfully recognised.
[image]

Also I cant set the modes schedule/away/manual through the user interface however here it can be changed:
[image]

jacekk015, 2023-01-19:
@HAuser1234 Again.
I need pairing **logs from HA** - not pairing window.
Enable DEBUG level first and restart HA.
Remove device from HA first.
I need logs of pairing process and still wait for 10 minutes.
After you pair TRV device still sends data in background - that's why we need to wait.
DON'T CUT ANYTHING from the logs.
You can download logs from HA settings - it's somewhere there - just click download as file.
Post it here as a file.

As for other errors just post here part of the logs of time you see the error. Best +/-5 minutes - so I can see what happened before and after.

Same for other functions you want to have.
Take a break[for logs clearness]
Switch something on TRV
Post here part of the logs with that action +5 minutes.
Zigbee is slow - when you switch something on TRV then HA gets it after even 10-15 seconds. It can take even longer if the device is far away from coordinator and data need to jump over 2 or 3 routers first.
We need that extra log time for that purpose - 5 minutes is safe enough.

HAuser1234, 2023-01-19:
Thats the pairing logfile:
[home-assistant_2023-01-19T07-23-53.570Z.log](https://github.com/zigpy/zha-device-handlers/files/10456382/home-assistant_2023-01-19T07-23-53.570Z.log)
I have captured it but wast sure if this was it

jacekk015, 2023-01-19:
You can post here all problematic parts of the log.

Here you get an error that command could not be sent because of coordinator can't find the route to the TRV -> NWK_NO_ROUTE
Check your topology and 
```
2023-01-19 08:17:03.967 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xAB30](TS0601): failed to set attribute: value: False attribute: 1026 cluster_id: 61184 endpoint_id: 1 Request failed after 5 attempts: <Status.NWK_NO_ROUTE: 205>
```

You link quality LQI is about 85, probably in 255 max scale. That could be reason of late delivery.
Check you routers/repeaters around. Maybe some nearby router need to be re-enabled from power.
```
2023-01-19 08:17:27.377 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xAB30), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\ta\x02\x00Y\x10\x02\x00\x04\x00\x00\x00\xb4'], tx_options=<TransmitOptions.NONE: 0>, radius=27, non_member_radius=0, lqi=84, rssi=None)
```

jacekk015, 2023-01-19:
Open window detection never will work properly.
TRV is to close to the hot heater.
My TRVs reported temp properly when I've put them in the freezer.
Many users tried this.
Nevertheless I've seen that under 0x0108 attribute:
```
2023-01-19 08:08:58.165 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0108 (command 0x0002)
```
Can make sensor of it. But usability probably won't be good.

HAuser1234, 2023-01-19:
ok thank you I have checked most devices on my network have interestingly such a low LQI even those near the main point

according to the manual it is a on/off thing, not a sensor.

jacekk015, 2023-01-19:
Steel rack case is an isolator.
Zigbee re-routs all the time, but it's mainly when you add a mains powered device, which acts as a router.
Sometimes is best to wait 2-3 hours. You can click after time a topology view and see a connections between devices.
Bulb or Zigbee socket is a router.
Cheapest solution is to add repeater - IKEA have one cheap device - have one in my network.
https://www.ikea.com/pt/en/p/tradfri-signal-repeater-10400408/

Every mains powered device, which acts as a router you can check children LQI under device card in HA.
3 dots button -> Manage Zigbee Device -> Neighbor devices
Check LQI to the next router or Coordinator

W can make it a switch. Later evening, off my local time.
There will be no name on the switch - not possible to name it technically.

HAuser1234, 2023-01-19:
ok thank you for the information. I will look further into this! I have a wifi 6 network with beamforming antennas, which would mean that the wifi intensity is really uneven within the house as well as multiple wifi channels are used at the same time when I understand that correctly. A probably pretty hard environment for zigbee.
Oh that would be great if that would work!

MattWestb, 2023-01-19:
@jacekk015 Numbers and functions for the preset and i patching one current file for ZHA.

jacekk015, 2023-01-19:
@MattWestb 
_TZE200_0hg58wyk

```
value == 0:
operation_preset = self.Preset.Manual

value == 1:
operation_preset = self.Preset.Away

value == 2:
operation_preset = self.Preset.Schedule
```

MattWestb, 2023-01-19:
[climateE-top.zip](https://github.com/zigpy/zha-device-handlers/files/10461424/climateE-top.zip)
Added all 5 E-top devices i have found.
Copy into the HA container `/usr/src/homeassistant/homeassistant/components/zha/climate.py` (make one backup of the original if somthing is going wrong).
Normal is the HA config folder at `/config/`.
So if putting the file in your HA config folder you can using `cp /config/climate.py /usr/src/homeassistant/homeassistant/components/zha/climate.py` inside the container for doing that and then restarting HA

jacekk015, 2023-01-19:
@HAuser1234 
Battery state - that TRV doesn't report battery level in percent.
It alarms only if power is low, so probably you will need to wait for reaction when battery will be empty.
None of your logs contain 0x0523 attribute report, so the TRV just doesn't report if all is OK
Replace the quirk and restart HA - you should see a switch -> Window detection function.
[ts0601_trv_etop.py.zip](https://github.com/zigpy/zha-device-handlers/files/10461543/ts0601_trv_etop.py.zip)

HAuser1234, 2023-01-20:
There seems to be a bug in there the thermostat is not recognised anymore
[home-assistant_2023-01-20T09-18-19.764Z.log](https://github.com/zigpy/zha-device-handlers/files/10464869/home-assistant_2023-01-20T09-18-19.764Z.log)
(all logs after restart and a new binding of one thermostat)
no entitys are found in this thermostat

jacekk015, 2023-01-20:
Yes. Small bug was there.
[ts0601_trv_etop.zip](https://github.com/zigpy/zha-device-handlers/files/10465821/ts0601_trv_etop.zip)

MattWestb, 2023-01-23:
The HVAC action is one left over from not having patched climate.py and you can deleting it.
Until some one can writing tests for all tuya TRVs is the maintainer not accepting merging of this quirks. I have trying getting the updated climate.py for other tuya TRVs but no luck getting them merged so need applying patching every HA update and hoping ZHA is not braking it and must redoing the patch.

HAuser1234, 2023-01-23:
Ah ok thank you

jacekk015, 2023-01-23:
Unfortunately your TRV doesn't send any state of TRV or valve.
unbekannt is because that is unsupported.
Z2M and your logs show one attribute 0x0403 which changes from [0] to [1]
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L2329

Maybe that's the state, but strange if Z2M didn't check that.
You, as an owner of the TRV, can only check if that value switches if the TRV goes from heating to idle and back.

```
2023-01-19 08:08:58.056 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:09:13.566 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:09:24.529 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:09:30.641 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [1] for attribute 0x0403 (command 0x0002)
2023-01-19 08:09:51.662 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:09:57.757 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [1] for attribute 0x0403 (command 0x0002)
2023-01-19 08:10:43.655 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:10:48.881 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [1] for attribute 0x0403 (command 0x0002)
2023-01-19 08:11:37.403 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:11:37.561 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:11:37.830 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:11:47.612 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:11:58.561 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:12:09.561 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:12:20.589 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:12:31.586 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:12:42.584 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:13:04.562 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:13:15.565 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:13:26.574 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:13:37.561 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:13:48.589 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
2023-01-19 08:13:58.566 DEBUG (MainThread) [zhaquirks.tuya] [0xab30:1:0xef00] Received value [0] for attribute 0x0403 (command 0x0002)
```

closed 2023-07-29 by github-actions[bot]