zigpy/zha-device-handlers #2075: [BUG] After latest upgrades PIR triggers 2 motion events 1-minute apart instead of a single event

NdK73, 2023-01-08:
**Describe the bug**
After one of the 2023 HA updates my PIR sensor triggers 2 motion events 1 minute apart. Previously it reported "motion" state for 1 minute then returned to normal.

**To Reproduce**
Steps to reproduce the behavior:
1. Add PIR sensor via ZHA and keep device page open
2. trigger motion detector
3. note on+off events followed after 1 minute by another couple of on+off events

**Expected behavior**
I'd expect to see motion events only when something moves in front off the sensor, not when motion is over.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "TUYATEC-h542kfah",
  "model": "RH3040",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.1.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Rome",
    "os_name": "Linux",
    "os_version": "5.10.0-20-amd64",
    "supervisor": "2022.12.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "tasmota_irhvac": {
      "version": "2021.12.0",
[... 341 lines cut ...]
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
In december it worked as expected: on for 1 minute when detecting motion, then off.

TheJulianJES, 2023-01-08:
There weren't any changes to motion or sensor related stuff.
1. Is the sensor low on batteries maybe?
2. Could it detect some other moving heat source?
3. Do you have another motion sensor that's the same model? (How does that one behave?)
4. Are the intervals of the other motion events always the same?
5. Can you attach a screenshot of the HA logbook showing the events?
6. Did you try re-pairing the sensor?

MattWestb, 2023-01-09:
Im sorry TheJulianJES  but i think its not one "tuya problem":
```
  "manufacturer": "TUYATEC-h542kfah",
  "model": "RH3040",
  "class": "zigpy.device.Device"
```

TheJulianJES, 2023-01-09:
Wasn't sure. Some other issues in the deCONZ repo mentioned Tuya a lot (for TUYATEC stuff), so I thought it's an earlier Zigbee manufacturer ID used by Tuya.

NdK73, 2023-02-03:
@TheJulianJES Sorry it took so long. I ordered another (identical) PIR and batteries.
The sensor reported 100% battery (and it was actually quite new).

The "old" battery in the new sensor is still reported as 100%, and makes it work like the old sensor used to: 1 minute on then off.

When I placed new battery in old sensor it couldn't detect movements at all. Once re-paired it resumed "bad" behaviour: on-off-1 minute-on-off :(

NdK73, 2023-05-20:
After latest upgrades it resumed correct behaviour. Mah!
Tks.

closed 2023-05-20 by NdK73