zigpy/zha-device-handlers #2076: [BUG] OSRAM Switch-LIGHTIFY sends broadcast and turns on/off all devices

floriandotorg, 2023-01-08:
**Describe the bug**
The OSRAM Switch-LIGHTIFY sends a broadcast messages to all ZigBee devices to turn them on/off when pressing the arrow down or up button.

**To Reproduce**
1. Connect OSRAM Switch-LIGHTIFY to ZHA 
(https://zigbee.blakadder.com/OSRAM_AC0251x00NJ.html)
3. Not setting up any automations
4. Press up or down arrow button
5. All ZigBee switches/lights etc. turn on or off

**Expected behavior**
Nothing happens when one of the buttons is clicked because no automations are set up.

**Additional context**
The problem is further described here:
https://github.com/zigpy/zha-device-handlers/issues/324

It appears that the switch is factory configured to send ZigBee broadcast messages. When using it with the OSRAM app, it can be reconfigured to not do so.

zigbee2mqtt has apparently fixed this by reconfiguring the switch on initial connect:
https://github.com/Koenkk/zigbee2mqtt/issues/962

It would be spectacular if someone could implant that fix into ZHA cause the switch is unusable and outright dangerous in some scenarios like this.

Thank you very much for the great work ❤️
Florian

TheJulianJES, 2023-01-08:
Like mentioned in the other issue, please fill-out the issue template.

Diagnostic information and the device signature are still missing. Diagnostic information can be downloaded by clicking the "Download Diagnostics" button on the device config page of the OSRAM switch (three small dots).
There, you'll also find the device signature under "Manage Zigbee device" -> "Device signature" tab

floriandotorg, 2023-01-09:
Hey Julian, thanks for your reply! Which issue template? I filled out the template for bugs.

Anyway, please find attached the device signature:

```JSON
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4364, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0810",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0020",
        "0x1000",
        "0xfd00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0810",
      "in_clusters": [
        "0x0000",
        "0x1000",
        "0xfd00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0810",
      "in_clusters": [
        "0x0000",
        "0x1000",
        "0xfd00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "OSRAM",
  "model": "Lightify Switch Mini",
  "class": "zhaquirks.osram.switchmini.OsramSwitchMini"
}
[... 481 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0008": {
            "endpoint_attribute": "level",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0300": {
            "endpoint_attribute": "light_color",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

If you need any other information, please let me know :-)

MattWestb, 2023-01-09:
Your Zigbee controller is one light controlling device that sending light commands to devices / groups that is bond to it.
If you have not putting your lights in one new group is it very likely that they is suing the `Default Lightlink Group` with Group Id: `0x0000`.
And your controller (and most lights) is using that as default if you not have changing the binding of it.
You can making one new group and putting all you light in and deleting them after after adding them to group `0x0000`.
Or you can deleting the binding of your light controller and binding it to one new group or the coordinator (its depends what the device can do group or device binding). If not have bond the controller its not sending any command to the coordinator so you need finding how to do it right.

TheJulianJES, 2023-01-09:
> Which issue template? I filled out the template for bugs.

All good now. (The issue template asks for diagnostic information and the signature)

From what MattWestb wrote, try the following:

Go to the device page -> three dots -> Manage Zigbee device (same place where you grabbed the signature)
Select the "Bindings" tab at the top. Then you have two things to try:
1. Select your Zigbee coordinator in the first dropdown menu and click unbind. Directly after having pressed unbind, press some buttons on the remote to make sure it wakes up. (You can repeat this a couple of times to be sure the remote received the command) (then see if the issue is resolved)
2. Select the name of your 0x0000 group. It should be "Default LightLink Group" or "No name group 0x0000".
Then, select all checkboxes and click "unbind group". Press some buttons on the remote directly after that (you can also repeat this multiple times). (see if this resolves the issue)

floriandotorg, 2023-01-09:
Hey @MattWestb and @TheJulianJES, thank you both so much for your support, I appreciated it!

I tried both:

1. The unbind button just shows the loading spinner forever. I tried ~20 times. I moved close to the Zigbee stick, I pressed buttons before and/or after, but same, it just loads forever. The problem is still there.
[image]
2. The group (in fact no groups) are in the dropdown:
[image]
I also looked under Configure > Groups, but same here no groups found, also no default group:
[image]

MattWestb, 2023-01-09:
If i remember right is spooky with the binding of this remotes and its looks you cant getting it unbound.

Then trying the Zigbee groups way.
First make one new group for your devices (then is normally having 0x0000 as factory and is not sett in ZHA so you dont see it) and adding then to the new group.
From the` Manage Zigbee Device` (as you was doing the binding) select the group cluster (0X0004) and then command remove (0x0003) for the device and "issue zigbee command".
You can also trying remove all but then you need adding  the new group or you is not getting any status from the device in ZHA.

floriandotorg, 2023-01-09:
Hey @MattWestb, thank you so much for your explanation!

So, I've added a group and added all devices:
[image]

Next, I tried to remove one of the switches from the default group:
[image]

But I got an error: `Failed to issue cluster command with status: 0x0000`
[image]

I then used remove_all and that worked. Then I tried to use add:
[image]

But it also failed: `Failed to issue cluster command with status: 0x0002`

Now I indeed get no status from the switch in ZHA anymore 😬
(also attempted to remove the switch from the Default group and add it again in ZHA, but no change)

Oh, and also the switch is still switching this switch ..

Any way to solve this? Guess this is what you call a cluster bug!

And again thanks for the help, I really appreciate it.

MattWestb, 2023-01-09:
I dont have OSRAM devices then they is not my taste then it come to function (or luck / crappy one).

Zigbee lights is normally controlled by groups and you can adding and deleting them by cluster commands.

Zigbee light controllers (your remote) shall being bonded to one group and sending command to it.

Many controllers have strange behaviors like must deleting binding before adding then dont have enough space for adding the coordinator (Philips HUE).
Also tuya is adding groups to remotes as lights (not binding) = very wrong.

I cant helping you fixing it then i dont knowing the device bu you can trying different combinations and you is very likely fixing it with little try and error.

Also search in community forum then i remember it was some strange with  the OSRAMs remotes but i cant remember where i have reading it.
If cant fixing the switch then you must "move" all lights to one new group and deleting the group `0x0000` in them so they not listening to it.

Z2M is not having any binding in there device wiki or it so very likely it cant being bonded to one custom group.

floriandotorg, 2023-01-31:
Thanks for all the help @MattWestb. I had to give up in the end, learning: no OSRAM ;)

closed 2023-01-31 by floriandotorg

MattWestb, 2023-01-31:
My feeling is that they have used old hardware with week radio and many hard and software bugs so i have kept away from them and is not having so many strange problem also trying getting real Zigbee 3 devices that can using the network futures like end device can keeping contact with there parents better.

Hope your future Zigbee adventure is working well !!

aaronegger, 2024-04-22:
@1doubleDD1 Sadly this has not fixed it for me.. I removed every group, not only the default group, but i even before zha shows the dialog with "Interview" the switch turns on/off all devices. All i can do is to send the switches back to amazon..

1doubleDD1, 2024-04-22:
@aaronegger  Did you try also unbinding from the "Manage Zigbee device" menu? I did it several times in a row; unbinding, deleting group, deleting the device from ZHA, then resetting the device; then back from the start.

I have a couple of Osram switches; yes, they are quite fiddly to initially set up, but afterwards I've had absolutely no issues with them whatsoever. Unfortunately might have to play with them again as I'm rethinking migration to ZHA, deConz was so much more stabile.

aaronegger, 2024-04-25:
@1doubleDD1 I did that. It finally worked. But for testing purposes i tried the function to bind it directly to a device.
Not the group, but a device, as i suspected it will bind to any device not only groups. Binding the switch to a single device worked, but unbinding did not work. Therefore i think the whole product is faulty and bad designed and therefore not reliable usable.
I will try one more time to get them working reliably and if i don't find any solution i will send them back. Very sad as they had the perfect size i needed. I need a narrow button to attach them to a hose of an vaccum cleaner (with 3d printed holders) and there is nothing except loratap that offers such devices.
The Osram device was perfect, als it has a dustproof design.

1doubleDD1, 2024-04-25:
@aaronegger Not sure that I completely follow your last message, but why do you need it to be binded to a device? Once it's behaving properly, meaning it's not switching off all devices, I use it via events in automations, and it works perfectly. Each of their buttons typically has 3 events per "pressing position" (?): short click, long press, and release after long press. You can check produced events in the Events tab of Developer options.

Example of my trigger in one automation:

```typescript
platform: event
event_type: zha_event
event_data:
  unique_id: 00:0d:6f:00:0d:9f:62:ec:1:0x0006
```

aaronegger, 2024-04-28:
@1doubleDD1 No i don't need i bound to a specific device. I just wanted to test if i would be able to unbind the switch, if it for example binds itself.

I think i will never bind any device to any device, as i want to be able to control what happens on an event and not let the device decide what button will do.

I only have some trust issues. I don't trust the switch to work properly in the future.

1doubleDD1, 2024-04-28:
@aaronegger I can only speak from my experience. I have 3 Osram switches, for about 6+years now. They were initially used in Osram system, then HA+deConz, and now HA+ZHA. Once properly paired and unbined, I've never had any issues with them. They are even still on their original batteries, not that it means much. I'm perfectly happy with them, their size and functionality.