zigpy/zha-device-handlers #2078: [BUG]  Error setting up entry Sonoff Zigbee 3.0 Hub for zha - ZHA failing to start

woody166, 2023-01-09:
**Describe the bug**
After recent upgrade to HA core 2023.1.2, ZHA is fialing to start with custom quirk in place for TS0601 TRV with following error:

Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_trv_beca.py:596
First occurred: 15:44:43 (1 occurrences)
Last logged: 15:44:43

Error setting up entry Sonoff Zigbee 3.0 Hub for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 382, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_trv_beca.py", line 553, in <module>
    class BecaWindowDetection(LocalDataCluster, OnOff):
  File "/config/custom_zha_quirks/ts0601_trv_beca.py", line 596, in BecaWindowDetection
    command_id: Union[foundation.Command, int, t.uint8_t],
AttributeError: module 'zigpy.zcl.foundation' has no attribute 'Command'

**To Reproduce**
Steps to reproduce the behavior:
1. rename custom quirks file
2. reload ZHA, starts running but not all entities/controls avaiable for TRV as were previously
3. rename custom quirk back to original name, ZHA failes to load
4. See error above

**Expected behavior**
ZHA to ooad and "all" entities/controls be avilable to access.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_b6wax7g0",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv.MoesHY368_Type1new"
}```

</details>

<details>
<!-- Diagnostic information can be acquired by clicking on the "Download Diagnostics" button in the device settings view -->
<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

MattWestb, 2023-01-09:
See https://github.com/zigpy/zha-device-handlers/issues/1061#issuecomment-1375212509 on how to fix custom quirks.

closed 2023-01-09 by puddly

woody166, 2023-01-09:
Awesome Guys, many thanks for the quick response and providing the solution