zigpy/zha-device-handlers #2081: [Device Support Request] Aqara Single Switch Module T1 (With Neutral) on ZHA with Home Assistant

stoffies00711, 2023-01-10:
**FIRSTLY.  Please be patient with me.  Just a home user with no programming skills and very new to GitHub.
 I use a Sonoff USB Zigbee dongle on Home Assistant Supervised.
Various Zigbee devices (Sonoff and Aqara)
Results with this specific  Aqara varies depending on software.**

ZHA
- I use ZHA because it's simple and works with most of my Zigbee devices. 
- The Aqara device is listed as compatible bit it only displays a dodgy temperature value with no switch or energy values.
- All other Sonoff and Aqara sensors work perfectly.  

HOMEKIT
The native Homekit integration (when using the Aqara Hub instead of the Sonoff USB).
 - It displays all info on the Aqara Cloud App
 - Integration in Home Assistant shows Switch ONLY

Zigbee2MQTT
- Tried this last night and it displayed all values I needed. Looks great but..
- Worked perfect with this Aqara device but then lost some entities on my other Zigbee devices such as battery levels.

I switched back from Zigbee2MQTT to ZHA because I lost more info than I gained overall. 

**I would like to know if anyone succeeded in integrating this specific device and all its values in ZHA on Home Assistant without being forced to migrate to other addons just for one device. **

<details>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0009",
        "0x000a",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019",
        "0xffff"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "31": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n0agl1",
  "class": "zigpy.device.Device"
}

```

</details>

<details>
<!-- Diagnostic information can be acquired by clicking on the "Download Diagnostics" button in the device settings view -->
<summary>Diagnostic information</summary>

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
![Screenshot_6](https://user-images.githubusercontent.com/87973552/211482490-1b89515d-8fdf-482f-9fa9-2c895267023c.jpg)

SkywiperSolutions, 2023-01-11:
I'm in the same boat - only that I use a ConbeeII dongle.
Except for the temperature measurements, these devices were working (more or less stable) until around HA Core 202110 update/and or ConbeeII dongle firmware upgrade in October. The firmware update was a requirement to have the newest version of Aqara Plugs working.
There's another thread here: #1231. I didn't try it out because it seems to be not fully working and I have critical appliances on several of these switches.

stoffies00711, 2023-01-11:
I've tried it but I must be doing something wrong.  After pairing again the quirk doesn't show.  
I've enabled and add the quirk path, copied the quirk from the post and restarted and repaired but no joy

stoffies00711, 2023-01-11:
It now shows quirk info under all my other Aqara devices except this one.

stoffies00711, 2023-01-11:
> I'm in the same boat - only that I use a ConbeeII dongle. To me it looks like the quirk was removed from the official device handler library (somewhere around October 2022?). I tried this custom quirk: [#1231 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1231#issuecomment-1294502283) and it seems to _work_ (on/off, power, energy) - including the sync when switching the physical button.

I went through this thread and it seems that the quirk is more to resolve the rocker/momentary switch type and not being able to change it. 

Most likely will not resolve our issue.

microsoluce, 2023-02-16:
Same here with a Conbee II. I paired 6 T1 modules today in ZHA, but I only got Device's Temperature (with wrong info btw), LQI/RSSI and Identify button.

400HPMustang, 2023-04-09:
I figured I'd just jump in here since this is the most recent ticket for this. I have a few of these relays and I am using ZHA with a SkyConnect and cannot get the lumi.switch.n0agl1 to show up correctly either. I get a temperature (the wrong temperature mind you) and an identify button. The "funny" thing is that the entity is a switch but there's no switch available to use. I also tried the custom device handler and it would not work.

chrisandsally, 2023-06-10:
Same issues, anyone found a workaround?

antoniochacon, 2023-06-20:
> Same issues, anyone found a workaround?

Same here

400HPMustang, 2023-06-20:
> > Same issues, anyone found a workaround?
> 
> Same here

I’m not sure about the work around that was linked earlier but I at least have my switch working using the steps detailed here 

https://github.com/zigpy/zha-device-handlers/issues/882#issuecomment-1387687255

MarkuBu, 2023-08-02:
Is there any progress on this topic? I tried to install a T1 and connect it with ZHA and Skyconnect, but without success.

nusatiy, 2023-12-20:
Does the Aqara T1 thermostat work with an external sensor?