zigpy/zha-device-handlers #2088: [Device Support Request] Monoxide detector _TZE200_nzsh6kps TS0601

anxdroid, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
This is a Tuya Smoke detector which is not covered by the T0601_smoke quirk

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_nzsh6kps",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Core",
    "version": "2023.1.2",
    "dev": false,
    "hassio": false,
    "virtualenv": true,
    "python_version": "3.10.8",
    "docker": false,
    "arch": "armv7l",
    "timezone": "Europe/Rome",
    "os_name": "Linux",
    "os_version": "5.15.84-v7+",
    "run_as_root": false
  },
  "custom_components": {
    "xiaomi_miot": {
      "version": "0.7.4",
      "requirements": [
        "construct==2.10.56",
        "python-miio>=0.5.6",
        "micloud>=0.3"
      ]
    },
    "iaquk": {
      "version": "1.6.2",
      "requirements": []
    },
    "jq300": {
      "version": "0.10.5",
      "requirements": [
[... 270 lines cut ...]
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
[Link to the online shop. The box says "Safealarm" (I guess it's the brand ?)
](https://www.expert4house.com/en/smart-home/sensors-and-detectors/tuya-carbon-monoxide-detector-with-smart-zigbee-siren)

javicalle, 2023-01-15:
I'm pretty sure that there is a custom quirk around for that kind of sensors with alarm but I can't find it.

It seems that device reports CO values and probably some switch status for the alarm.

Please, enable the debug logs and attach a few device reports.

anxdroid, 2023-01-15:
Many thanks for your replay. To send the device reports, I don't know how to get to a deeper log level (i.e. zigbee low level traffic) for this specific device. Is It something I can do from the UI or I have to set the loglevel for zha and do some filtering by hand?

javicalle, 2023-01-15:
Here are instructions to enable the debug logs:
* https://www.home-assistant.io/integrations/zha/#debug-logging

It can be very verbose. Try to catch the operation part for your device looking at the logs while interacting with the device.

TheJulianJES, 2023-01-15:
FYI, it's not in the docs, but debug logging can now also be enabled through the UI: 
[image]

anxdroid, 2023-01-15:
Thanks you to both of you guys. The debug for the integration was already active. I'll try to follow your instructions to produce something useful

anxdroid, 2023-01-17:
Something strange is happening, I fear that zha is using the wrong quirk.
[image]
Logs are enabled according to https://www.home-assistant.io/integrations/zha/#debug-logging and I have tried to grep the log with several ids related to the device:

`2023-01-17 19:46:14.645 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0000] Attribute id: 4 value: _TZE200_aoclfnxz
2023-01-17 19:46:14.645 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0000] Attribute id: 5 value: TS0601
2023-01-17 19:46:15.597 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0000] Attribute id: 4 value: _TZE200_aoclfnxz
2023-01-17 19:46:15.597 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0000] Attribute id: 5 value: TS0601
2023-01-17 19:46:15.598 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0xef00] Attribute id: 296 value: 0
2023-01-17 19:46:15.598 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0204] Attribute id: 1 value: 0
2023-01-17 19:46:16.248 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0xef00] Attribute id: 1026 value: 1
2023-01-17 19:46:16.249 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0201] Attribute id: 37 value: 1
2023-01-17 19:46:16.249 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0xef00] Attribute id: 257 value: 1
2023-01-17 19:46:16.250 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0xef00] Attribute id: 1027 value: 1
2023-01-17 19:46:16.251 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0xef00] Attribute id: 528 value: 21
2023-01-17 19:46:16.251 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0xef00] Attribute id: 1060 value: 1
2023-01-17 19:46:19.411 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0xef00] Attribute id: 536 value: 210
2023-01-17 19:46:21.906 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0201] Attribute id: 27 value: 2
2023-01-17 19:46:21.906 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0201] Attribute id: 0 value: 2100
2023-01-17 19:46:21.907 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0201] Attribute id: 18 value: 2100
2023-01-17 19:46:21.907 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0201] Attribute id: 30 value: 0
2023-01-17 19:46:21.907 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0201] Attribute id: 41 value: 0
2023-01-17 19:46:21.907 DEBUG (MainThread) [zigpy.appdb] [0x3e23:1:0x0201] Attribute id: 28 value: 4
2023-01-17 19:55:17.792 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (37, 218, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x3e23 endpoint=0>, 0, <ZDOCmd.Bind_req: 0x0021>, 0, b'\xd9\x91t\xae\xfe\xffr\x02\\\x01\x00\xef\x03\xe72\x05\xff\xff.!\x00\x01', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2023-01-17 19:55:27.963 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 218, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x3e23 endpoint=0>, 0, <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>, 0, 0, 0, 0]
2023-01-17 19:55:27.967 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xda 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x3e23 endpoint=0>, status: 0xd0`

`2023-01-17 19:46:15.114 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_aoclfnxz TS0601 (5c:02:72:ff:fe:ae:74:91)
2023-01-17 19:46:15.115 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 5c:02:72:ff:fe:ae:74:91: <class 'zhaquirks.tuya.ts0601_electric_heating.MoesBHT'>
2023-01-17 19:46:38.046 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 3) params: FOREIGN KEY constraint failed
2023-01-17 19:46:38.068 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 4) params: FOREIGN KEY constraint failed
2023-01-17 19:46:38.155 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 5) params: FOREIGN KEY constraint failed
2023-01-17 19:46:38.177 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 6) params: FOREIGN KEY constraint failed
2023-01-17 19:46:38.185 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 24) params: FOREIGN KEY constraint failed
2023-01-17 19:46:38.194 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 22) params: FOREIGN KEY constraint failed
2023-01-17 19:46:38.208 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 23) params: FOREIGN KEY constraint failed
2023-01-17 19:46:38.221 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 21) params: FOREIGN KEY constraint failed
2023-01-17 19:46:38.405 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 17) params: FOREIGN KEY constraint failed
2023-01-17 19:46:40.878 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (5c:02:72:ff:fe:ae:74:91, 1, 513, 19) params: FOREIGN KEY constraint failed
[... cut ...]
2023-01-17 19:57:05.029 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x30E7](TS0601): Update device availability -  device available: True - new availability: True - changed: False
2023-01-17 19:57:30.200 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xC1D0](TS0601): Device seen - marking the device available and resetting counter
2023-01-17 19:57:30.201 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xC1D0](TS0601): Update device availability -  device available: True - new availability: True - changed: False`

can this be of any use ?

javicalle, 2023-01-17:
I can't get the info from the logs.
I would suggest to add the device to the `TuyaCOSensor` quirk.

Enable the local quirk. Here you have some instructions:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Cpy in your local quirk folder the current `ts0601_co.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_co.py

Edit the file and modify this part:
```python
        MODELS_INFO: [
            ("_TZE200_7bztmfm1", "TS0601"),
            ("_TZE200_nzsh6kps", "TS0601"),
        ],
```

Save changes, restart HA.
Enable the debug logs, interact with the device and attach here. And please put the logs between ` ``` ` marks
````
```
many logs here
```
````

anxdroid, 2023-01-18:
thank you, step by step looks like we're getting to it :-)

grep cc:86:ec:ff:fe:b0:78:16 (IEEE)
```
2023-01-18 22:46:26.335 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_nzsh6kps TS0601 (cc:86:ec:ff:fe:b0:78:16)
2023-01-18 22:46:26.335 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for cc:86:ec:ff:fe:b0:78:16: <class 'ts0601_co.TuyaCOSensor'>
2023-01-18 22:46:52.981 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (cc:86:ec:ff:fe:b0:78:16, 1, 1037, 0) params: FOREIGN KEY constraint failed
2023-01-18 22:49:22.887 DEBUG (MainThread) [homeassistant.components.zha.api] Reconfiguring node with ieee_address: cc:86:ec:ff:fe:b0:78:16
2023-01-18 22:50:28.269 INFO (MainThread) [homeassistant.components.zha.api] Removing node cc:86:ec:ff:fe:b0:78:16
2023-01-18 22:50:28.270 INFO (MainThread) [zigpy.application] Removing device 0x30e7 (cc:86:ec:ff:fe:b0:78:16)
2023-01-18 22:50:44.276 INFO (MainThread) [zigpy.application] New device 0xeefb (cc:86:ec:ff:fe:b0:78:16) joined the network
2023-01-18 22:50:44.280 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0xEEFB ieee=cc:86:ec:ff:fe:b0:78:16 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x83\xfb\xee\x16x\xb0\xfe\xff\xec\x86\xcc\x8e'
2023-01-18 22:50:44.281 DEBUG (MainThread) [zigpy.zdo] [0xeefb:zdo] ZDO request ZDOCmd.Device_annce: [0xEEFB, cc:86:ec:ff:fe:b0:78:16, 142]
2023-01-18 22:50:44.585 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0xEEFB ieee=cc:86:ec:ff:fe:b0:78:16 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x90\x00\xfb\xee\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'
2023-01-18 22:50:44.790 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0xEEFB ieee=cc:86:ec:ff:fe:b0:78:16 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x92\x00\xfb\xee\x01\x01'
2023-01-18 22:50:44.969 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0xEEFB ieee=cc:86:ec:ff:fe:b0:78:16 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x94\x00\xfb\xee\x14\x01\x04\x01Q\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xef\x02\x19\x00\n\x00'
2023-01-18 22:50:45.199 INFO (MainThread) [zigpy.device] [0xeefb] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_nzsh6kps' nwk=0xEEFB ieee=cc:86:ec:ff:fe:b0:78:1 is_initialized=True>
2023-01-18 22:50:45.200 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE200_nzsh6kps' nwk=0xEEFB ieee=cc:86:ec:ff:fe:b0:78:16 is_initialized=True>
2023-01-18 22:50:45.204 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_nzsh6kps TS0601 (cc:86:ec:ff:fe:b0:78:16)
2023-01-18 22:50:45.205 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for cc:86:ec:ff:fe:b0:78:16: <class 'ts0601_co.TuyaCOSensor'>
2023-01-18 22:50:45.226 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0xEEFB:cc:86:ec:ff:fe:b0:78:16 entering async_device_initialized - is_new_join: True
2023-01-18 22:50:45.227 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0xEEFB:cc:86:ec:ff:fe:b0:78:16 has joined the ZHA zigbee network
2023-01-18 22:50:45.245 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (cc:86:ec:ff:fe:b0:78:16, 1, 0, 4, '_TZE200_nzsh6kps') params: FOREIGN KEY constraint failed
2023-01-18 22:50:45.284 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (cc:86:ec:ff:fe:b0:78:16, 1, 0, 5, 'TS0601') params: FOREIGN KEY constraint failed
2023-01-18 22:50:45.424 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (cc:86:ec:ff:fe:b0:78:16, 1, 1037, 0) params: FOREIGN KEY constraint failed
```
grep 0xeefb (NWK)
```
2023-01-18 22:50:44.273 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[35, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0xfffd>, 0, <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, 0, 0, 19, b'\x83\xfb\xee\x16x\xb0\xfe\xff\xec\x86\xcc\x8e', 0, 175, 207, 40, 31, 97, 3, -68]
2023-01-18 22:50:44.273 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0xfffd>, 0, <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, 0, 0, 19, b'\x83\xfb\xee\x16x\xb0\xfe\xff\xec\x86\xcc\x8e', 0, 175, 207, 40, 31, 97, 3, -68]
2023-01-18 22:50:44.276 INFO (MainThread) [zigpy.application] New device 0xeefb (cc:86:ec:ff:fe:b0:78:16) joined the network
2023-01-18 22:50:44.277 DEBUG (MainThread) [zigpy.device] [0xeefb] Scheduling initialization
2023-01-18 22:50:44.281 DEBUG (MainThread) [zigpy.zdo] [0xeefb:zdo] ZDO request ZDOCmd.Device_annce: [0xEEFB, cc:86:ec:ff:fe:b0:78:16, 142]
2023-01-18 22:50:44.284 INFO (MainThread) [zigpy.device] [0xeefb] Requesting 'Node Descriptor'
2023-01-18 22:50:44.285 DEBUG (MainThread) [zigpy.device] [0xeefb] Extending timeout for 0x90 request
[... cut ...]
2023-01-18 22:50:45.200 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE200_nzsh6kps' nwk=0xEEFB ieee=cc:86:ec:ff:fe:b0:78:16 is_initialized=True>
2023-01-18 22:50:45.204 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_nzsh6kps TS0601 (cc:86:ec:ff:fe:b0:78:16)
2023-01-18 22:50:45.245 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (cc:86:ec:ff:fe:b0:78:16, 1, 0, 4, '_TZE200_nzsh6kps') params: FOREIGN KEY constraint failed
```

anxdroid, 2023-01-27:
Is the log useful ? Should I provide some more info ?

javicalle, 2023-01-28:
Sorry, I lost the track of that issue.

I can't see any problem in your logs. I see that you have repaired the device a couple of times.

From the pictures I would expect at least a battery report but IU can't see any in your logs.

Please attach the logs from switching the self-test button.
You can also power off the device (mains or battery) and attach the debug logs when power on the device. At this moment usually the device reports its full status.

anxdroid, 2023-02-01:
Logs after pressing the self test button (I did it twice)
```
2023-02-01 08:48:02.957 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, 1, 260, 61184, b'\t \x01\x00\x04\t\x04\x00\x01\x01', 0, 175, 255, 143, 123, 103, 3, -55]
2023-02-01 08:48:02.958 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, 1, 260, 61184, b'\t \x01\x00\x04\t\x04\x00\x01\x01', 0, 175, 255, 143, 123, 103, 3, -55]
2023-02-01 08:48:02.986 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09200100040904000101'
2023-02-01 08:48:02.988 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 172, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, 260, 61184, 1, b'\x18 \x0b\x01\x86', <DeconzTransmitOptions.USE_APS_ACKS|USE_NWK_KEY_SECURITY: 6>, 0)
2023-02-01 08:48:03.060 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, 172, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, 1, <TXStatus.SUCCESS: 0>, 0, 0, 0, 0]
2023-02-01 08:48:03.062 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xac 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, status: 0x00
2023-02-01 08:48:03.081 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, 1, 260, 61184, b'\t!\x01\x00\x05\x0b\x04\x00\x01\x00', 0, 175, 255, 143, 123, 103, 3, -55]
2023-02-01 08:48:03.082 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, 1, 260, 61184, b'\t!\x01\x00\x05\x0b\x04\x00\x01\x00', 0, 175, 255, 143, 123, 103, 3, -55]
2023-02-01 08:48:03.107 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09210100050b04000100'
2023-02-01 08:48:03.108 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 173, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, 260, 61184, 1, b'\x18!\x0b\x01\x86', <DeconzTransmitOptions.USE_APS_ACKS|USE_NWK_KEY_SECURITY: 6>, 0)
2023-02-01 08:48:03.163 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 173, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, 1, <TXStatus.SUCCESS: 0>, 0, 0, 0, 0]
2023-02-01 08:48:03.164 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xad 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, status: 0x00
2023-02-01 08:48:03.420 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, 1, 260, 61184, b'\t"\x01\x00\x05\x08\x01\x00\x01\x00', 0, 175, 255, 143, 123, 103, 3, -55]
2023-02-01 08:48:03.421 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, 1, 260, 61184, b'\t"\x01\x00\x05\x08\x01\x00\x01\x00', 0, 175, 255, 143, 123, 103, 3, -55]
2023-02-01 08:48:03.450 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0xeefb>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09220100050801000100'
2023-02-01 08:48:03.451 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 174, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, 260, 61184, 1, b'\x18"\x0b\x01\x86', <DeconzTransmitOptions.USE_APS_ACKS|USE_NWK_KEY_SECURITY: 6>, 0)
2023-02-01 08:48:03.518 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 174, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, 1, <TXStatus.SUCCESS: 0>, 0, 0, 0, 0]
2023-02-01 08:48:03.519 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xae 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, status: 0x00

[... cut ...]
2023-02-01 08:50:33.530 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 241, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, 1, <TXStatus.SUCCESS: 0>, 0, 0, 0, 0]
2023-02-01 08:50:33.533 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xf1 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0xeefb endpoint=1>, status: 0x00
```

hope that grepping the logs with 0xeefb is giving something useful

javicalle, 2023-02-02:
Let's see what is in the logs.

(If I have decoded it right) the first block reporting is:
* TuyaCO2ManufCluster:get_data(data=TuyaCommand(status=0, tsn=4, datapoints=[TuyaDatapointData(dp=9, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x01', *payload=<enum8.undefined_0x01: 1>))]))
* TuyaCO2ManufCluster:get_data(data=TuyaCommand(status=0, tsn=5, datapoints=[TuyaDatapointData(dp=11, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))
* TuyaCO2ManufCluster:get_data(data=TuyaCommand(status=0, tsn=5, datapoints=[TuyaDatapointData(dp=8, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))]))

The second block are the reports:
* TuyaCO2ManufCluster:get_data(data=TuyaCommand(status=0, tsn=5, datapoints=[TuyaDatapointData(dp=9, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x01', *payload=<enum8.undefined_0x01: 1>))]))  (3 times)
* TuyaCO2ManufCluster:get_data(data=TuyaCommand(status=0, tsn=6, datapoints=[TuyaDatapointData(dp=11, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))]))  (2 times)
* TuyaCO2ManufCluster:get_data(data=TuyaCommand(status=0, tsn=7, datapoints=[TuyaDatapointData(dp=8, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))]))  (once)

So, the device reports (at least) to DPs 8, 9 and 11.

After rebooting device reports:
* TuyaCO2ManufCluster:set_time_request(data=[0, 3])  (<-- time synch)
* TuyaCO2ManufCluster:Read_Attributes(attribute_ids=[7])
* TuyaCO2ManufCluster:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0007, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint32_t, value=728556618))])

I have to decode all the messages one by one so not an easy way to go forward.
If someone can create and test a quirk with the reported DPs, maybe we could know what are for every one.

closed 2024-02-04 by github-actions[bot]