zigpy/zha-device-handlers #2090: TS0011 and TS0013

amiraizat124, 2023-01-13:
**Is your feature request related to a problem? Please describe.**
I cannot assign the proper quirk while paring the TS0011 and TS0013

I already tried using local quirk and without a local quirk.
I attached info for TS0011.

<details>

<summary>Device signature</summary>

```yaml
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_5ftkaulg",
  "model": "TS0011",
  "class": "zigpy.device.Device"
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks/
```

</details>

<details>
<summary>Additional logs</summary>

```
[0x2aba] Discovered basic device information for <Device model='TS0011' manuf='_TZ3000_5ftkaulg' nwk=0x2ABA ieee=a4:c1:38:af:e8:e0:74:67 is_initialized=True>
Device is initialized <Device model='TS0011' manuf='_TZ3000_5ftkaulg' nwk=0x2ABA ieee=a4:c1:38:af:e8:e0:74:67 is_initialized=True>
Checking quirks for _TZ3000_5ftkaulg TS0011 (a4:c1:38:af:e8:e0:74:67)
Considering <class 'ts001x.Tuya_Single_No_N'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'ts001x.TuyaSingleNoNeutralSwitch'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'ts001x.Tuya_Single_No_N'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'ts001x.TuyaSingleNoNeutralSwitch'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'ts001x.Tuya_Single_No_N'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'ts001x.TuyaSingleNoNeutralSwitch'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.tuya.ts001x.Tuya_Single_No_N'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.tuya.ts001x.TuyaSingleNoNeutralSwitch'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2'>
Fail because endpoint list mismatch: {1, 2, 242} {1}
Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
'button' component -> 'ZHAIdentifyButton' using ['identify']
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
'select' component -> 'ZHAStartupOnOffSelectEntity' using ['on_off']
device - 0x2ABA:a4:c1:38:af:e8:e0:74:67 entering async_device_initialized - is_new_join: True
device - 0x2ABA:a4:c1:38:af:e8:e0:74:67 has joined the ZHA zigbee network
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2023-01-13:
Is both devices with or without neutral ?
Also post the TS0013 signature then its looks only have some cluster changed.

amiraizat124, 2023-01-13:
Signature for TS0013
```
 "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
```

Both without neutral

MattWestb, 2023-01-13:
I was patching the latest quirk with new classes for your devices.
Unzip and put in the local quirk folder and restart HA.
[ts001x.zip](https://github.com/zigpy/zha-device-handlers/files/10410271/ts001x.zip)
Look if the system is loading the quirk and reporting back.

amiraizat124, 2023-01-13:
[image]
[image]

works great for TS0011 and TS0013. Thank

MattWestb, 2023-01-13:
## GREAT !!
I need you doing some more test then i was adding your device is the end and was thinking it was one new family of device but i was finding that the 2 gang was already in the quirk and i need redoing the complete file for getting it looking better.

I posting one new quirk later :-))

MattWestb, 2023-01-13:
Updated quirk with moving all without tuya cluster in separate parts and comment for making the file better.

It shall working the same as the first only changing names.
[ts001x.zip](https://github.com/zigpy/zha-device-handlers/files/10410917/ts001x.zip)
Pleas test and if its working i can doing on PR for getting it in next release of HA.

amiraizat124, 2023-01-13:
Yeah, still working great 

 TS0011
  ```
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_5ftkaulg",
  "model": "TS0011",
  "class": "ts001x.TuyaSingleNoNeutralSwitch_2"
  ```
  
  
  TS0013
  ```
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_09gto2zn",
  "model": "TS0013",
  "class": "ts001x.Tuya_TripleNoNeutralSwitch_2"
  ```

MattWestb, 2023-01-13:
Was finding some bad formatting in the code then making the PR :-((

Can you testing the final version 2 ?
[ts001x.zip](https://github.com/zigpy/zha-device-handlers/files/10411113/ts001x.zip)
PR is made and then its merged it closing this issue.

Dont forgetting deleting the local quirk then it in the updated HA for not getting problem in the future then its being updated.

amiraizat124, 2023-01-13:
Yes..Still working good

MattWestb, 2023-01-13:
Great then its more then likely its coming in next HA release then its being merged :-)))
Thanks for testing !!!

amiraizat124, 2023-01-16:
TS0012 (No Neutral) failed to control individual switches. I trigger 1 switch, but both turned on.

Device Signature: Out cluster only has "0x000a" no "0x0019"
```
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_18ejxno0",
  "model": "TS0012",
  "class": "ts001x.TuyaDoubleNoNeutralSwitch_2"

```

I modified the replacement for TuyaDoubleNoNeutralSwitch_2 output cluster from Ota.cluster_id to Time.cluster_id.
Now it works fine for my TS0012.
```
replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
```

Don't know if this the correct way or not.

MattWestb, 2023-01-16:
Its needs one new device class then its one new family without time cluster like the TS004X switches is having.

I looking implanting it in the PR for the first ones.

MattWestb, 2023-01-16:
@amiraizat124 Is the first posted signature of the new TS0012 the same as the device without the updated quirk ? 
`"class": "ts001x.TuyaDoubleNoNeutralSwitch_2"` is overriding all settings and is its not the same as the "original" i is putting in the wrong info in the new device class signature.

amiraizat124, 2023-01-16:
My TS0012 (Moes) is from a different manufacturer from my TS0011 (Smatrul) and TS0013 (Smatrul).

amiraizat124, 2023-01-16:
If I remove the custom quirk, the quirk change to other quirk for the TS0012
```
"node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_18ejxno0",
  "model": "TS0012",
  "class": "zhaquirks.tuya.ts001x.TuyaDoubleNoNeutralSwitch_2"
```

MattWestb, 2023-01-16:
Its not helping then one quirk is overriding the signature and can changing the node description.

If you not have mounting it in the wall is the easiest deleting the device and reparing the device and have debug logging for ZHA and you is getting it in the HA log in the paring process.

I have doing one new device class and family:
```
# Switch_3 family = Normal NoNeutralSwitch (without tuya cluster) and wothout time cluster.
class TuyaDoubleNoNeutralSwitch_3(EnchantedDevice, TuyaSwitch):
    """Tuya 2 gang no neutral light switch (v3)."""
```
Can you testing if its working OK ?
[ts001x.zip](https://github.com/zigpy/zha-device-handlers/files/10423793/ts001x.zip)

amiraizat124, 2023-01-16:
> If you not have mounting it in the wall is the easiest deleting the device and reparing the device and have debug logging for ZHA and you is getting it in the HA log in the paring process.

I'll update u back, currently I don't have physical access to the TS0012.

closed 2023-01-16 by dmulcahey

RED4911, 2023-05-06:
After the update, my ts0012 turns on all the lights if you press any key. How to fix it?

mr-mmc, 2024-02-16:
> After the update, my ts0012(TS0012_TZ3000_e98krvvk) turns on all the lights if you press any key. How to fix it?

i have same problem :(  (3gang)