zigpy/zha-device-handlers #2096: [BUG] _TZ1800_ladpngdx does not work correctly

moritzbeck01, 2023-01-15:
**Describe the bug**
As mentioned in https://github.com/zigpy/zha-device-handlers/issues/1168:
It seems like pressing the normal button reports true, but releasing it reports nothing. In contrast, pressing the tamper button changes the same entity to false, releasing it does nothing aswell.

**To Reproduce**
1. Doorbell entity is `false`
2. Press the doorbell button
3. Doorbell entity changes to `true`
4. Release the doorbell button
5. Doorbell entity does *not* change to `false`
6. Press the tamper button
7. Doorbell entity changes to `false`
8. Release the tamper button

**Expected behavior**
I press the doorbell button, the doorbell entity reports `true`. I release it, it reports `false`.
There is no tamper entity (why?), but I would expect the same behavior with the tamper button and a potential tamper entity.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ1800_ladpngdx",
  "model": "TS0211",
  "class": "zhaquirks.tuya.ts0211.TuyaDoorbell0211"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.1.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.80",
    "supervisor": "2022.12.1",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
[... 361 lines cut ...]
  user_id: null
```

Event 2:

```
event_type: zha_event
data:
  device_ieee: 84:2e:14:ff:fe:f6:d3:9b
  unique_id: 84:2e:14:ff:fe:f6:d3:9b:1:0x0500
  device_id: f461037a8c12c5b0b35b665927357440
  endpoint_id: 1
  cluster_id: 1280
  command: single
  args: []
  params: {}
origin: LOCAL
time_fired: "2023-01-04T20:07:19.666115+00:00"
context:
  id: 01GNZ6ACVJQJQXJSJCQ1QDDZHV
  parent_id: null
  user_id: null
```

</details>

javicalle, 2023-01-15:
It seems that device triggers 3 events for every button press:
* https://github.com/zigpy/zha-device-handlers/pull/1400#issue-1153382712

I'm not sure why it behaves that way, but I'll suggest this change to the quirk:

<details open>
<summary>ts0211.py</summary>

```python
"""Tuya Doorbell."""

from typing import Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Identify, Ota, PowerConfiguration
from zigpy.zcl.clusters.homeautomation import Diagnostic
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import CustomCluster
from zhaquirks.const import (
    COMMAND_SINGLE,
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZHA_SEND_EVENT,
)

class IasZoneDoorbellCluster(CustomCluster, IasZone):
    """Custom IasZone cluster for the doorbell."""

    cluster_id = IasZone.cluster_id

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: Tuple[IasZone.ZoneStatus],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        """Handle cluster request."""
        # args looks like [<ZoneStatus.Alarm_1: 1>, <bitmap8.0: 0>, 0, 0]
        # we are only interested in 3 bits of the ZoneStatus item

        # with thanks to
        # https://github.com/Koenkk/zigbee-herdsman-converters/blob/26553bfa9d747c1a8b6322090429c04cf612c3c1/converters/fromZigbee.js#L2193-L2206

        arg = args[0]
        if arg & IasZone.ZoneStatus.Alarm_1:
            self.listener_event(ZHA_SEND_EVENT, COMMAND_SINGLE, [])

        # the doorbell also sets (arg & IasZone.ZoneStatus.Tamper) and
        # (arg & IasZone.ZoneStatus.Battery) but those should eventually
        # be handled by a more generic IasZone handling

        super().handle_cluster_request(hdr, args, dst_addressing=dst_addressing)

class TuyaDoorbell0211(CustomDevice):
    """Tuya doorbell."""

    # {
    #   "node_descriptor": "NodeDescriptor(
    #      logical_type=<LogicalType.EndDevice: 2>,
    #      complex_descriptor_available=0,
    #      user_descriptor_available=0,
    #      reserved=0,
    #      aps_flags=0,
    #      frequency_band=<FrequencyBand.Freq2400MHz: 8>,
    #      mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>,
    #      manufacturer_code=4619,
    #      maximum_buffer_size=82,
    #      maximum_incoming_transfer_size=82,
    #      server_mask=11264,
    #      maximum_outgoing_transfer_size=82,
    #      descriptor_capability_field=<DescriptorCapability.NONE: 0>,
    #      *allocate_address=True,
    #      *is_alternate_pan_coordinator=False,
    #      *is_coordinator=False,
    #      *is_end_device=True,
    #      *is_full_function_device=False,
    #      *is_mains_powered=False,
    #      *is_receiver_on_when_idle=False,
    #      *is_router=False,
    #      *is_security_capable=False)",
    #   "endpoints": {
    #     "1": {
    #       "profile_id": 260,
    #       "device_type": "0x0402",
    #       "in_clusters": [
    #         "0x0000",
    #         "0x0001",
    #         "0x0003",
    #         "0x0500",
    #         "0x0b05"
    #       ],
    #       "out_clusters": [
    #         "0x0019"
    #       ]
    #     }
    #   },
    #   "manufacturer": "_TZ1800_ladpngdx",
    #   "model": "TS0211",
    #   "class": "zhaquirks.tuya.ts0211.TuyaDoorbell0211"
    # }

    signature = {
        MODEL: "TS0211",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IasZone.cluster_id,
                    Diagnostic.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    IasZoneDoorbellCluster,
                    Diagnostic.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
```

</details>

moritzbeck01, 2023-02-04:
@javicalle Can you open a PR for this?

javicalle, 2023-02-04:
Not until there is confirmation that the quirk works as it should.
Were you able to validate the quirk on your device?

moritzbeck01, 2023-02-05:
> Not until there is confirmation that the quirk works as it should. Were you able to validate the quirk on your device?

After repairing the device with the new quirk enabled, the behaviour is as follows:
- when I press the bell button, an event fires and the entity changes to `false`
- when I release the bell button, an event fires
- when I press the tamper button, the entity changes to `true`
- when I release the tamper button, nothing happens

Just for clarity, this is what I expected to happen:
- as with all the other devices, there is just one entity
- the entity changes status according to the bell button
- the tamper button does nothing

javicalle, 2023-02-08:
Please attach the debug logs for button and tamper indicating the sequence of events. You can do it by enabling the debug logs from the device view.

moritzbeck01, 2023-02-14:
[error_log-1.txt](https://github.com/zigpy/zha-device-handlers/files/10734331/error_log-1.txt)

1. press bell
2. release bell
3. press bell
4. release bell
5. press tamper
6. release tamper
7. press tamper
8. release tamper
9. press bell
10. press tamper
11. release tamper
12. release bell
13. press bell
14. press tamper
15. release bell
16. release tamper

I'm not sure if every event came through, because the light didn't blink on every action, even though I actively waited a second or so between every action.

With regards to my previous comment, I have to correct something:
The entity also sets and resets on releasing the bell/tamper button. So both pressing and releasing the bell button sets the entity to true and both pressing and releasing the tamper button resets the entity to false.
Sorry for the confusion, I didn't test for this exact scenario (pressing/releasing intertwined), even though it makes much sense.

moritzbeck01, 2023-03-24:
Is there any further information I can provide?

moritzbeck01, 2023-09-20:
Not stale

moritzbeck13, 2024-03-19:
Not stale

closed 2024-09-22 by github-actions[bot]