zigpy/zha-device-handlers #2097: _TZE200_3towulqd TS0601 Tuya Motion Sensor with illuminance sensor [BUG]

Roku38, 2023-01-15:
**Describe the bug**
I've been using the quark that @javicalle suggested for the motion sensor and noticed that it works fine with a problem where if the room is dark the motion sensor occupancy trigger doesn't work. if I add some light to the room the sensor works fine. I than re-sync'd the device with tuya's smart life app and it doesn't seems to have that issue. I confirmed the behavior with several sensors.

Is there something that can be done or is it limitation of the dp instructions?

**To Reproduce**
1. Installed ZHA and add the ZHA quirk as suggest here - https://github.com/zigpy/zha-device-handlers/issues/1599#
2. sync the motion sensor with ZHA
3. make sure the room is dark
4. walk around - you'll notice that the sensor get triggered as the red light turns on
5. observe - no trigger in HA logs for the motion occupancy sensor

**Expected behavior**
To get triggered in all situations. Note: this doesn't happen if the device is using TUYA native integration and the sensor synced with smart-life

TheJulianJES, 2023-01-15:
Please also provide the device signature and diagnostic information for the device, so the issue post is complete.

javicalle, 2023-01-15:
Can be related to the `sensitivity_level`?
I mean, device can detect the movement but a low sensitivity makes that no event will be trigered.

If not related (in addition to providing the data from the device) activate the debug logs and attach them to the issue.

Roku38, 2023-01-15:
thank you @javicalle. I'll turn debugging on. Calling device info shows that the sensitivity in both cases is set to high. For now I switched back to using tuya hub.

Kaot93, 2023-02-09:
Probably have the same bug. Haven't tested it very much but I get no clearance in complete darkness either, which is pretty odd I think. 

Also my illuminance sensor doesn't deliver anything useful, while I get differing values if I read the cluster itself. 
Is there something with the conversion?

javicalle, 2023-02-10:
Without any debug logs from device with the normal and the rogue behavior anything can't be do?

Kaot93, 2023-02-10:
I think that's the stuff you want to know: 
```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "omada": {
      "version": "0.4.0-beta.1",
      "requirements": []
    },
    "adaptive_lighting": {
      "version": "1.4.1",
      "requirements": []
    },
    "dreame_vacuum": {
      "version": "v0.16.3",
      "requirements": [
        "pillow",
        "numpy",
        "pybase64",
        "requests",
        "pycryptodome",
        "python-miio>=0.5.6",
        "py-mini-racer"
      ]
    },
    "multimatic": {
      "version": "1.14.0b1",
      "requirements": [
        "pymultimatic==0.7.0b2"
      ]
    },
    "tapo": {
      "version": "1.5.1",
      "requirements": [
        "plugp100==2.5.1"
      ]
    },
    "localtuya": {
      "version": "5.0.0",
      "requirements": []
    },
    "dwd_weather": {
      "version": "1.2.24",
      "requirements": [
        "simple_dwd_weatherforecast==1.1.5",
        "markdownify==0.6.5"
      ]
    },
    "zha_toolkit": {
      "version": "v0.8.33",
      "requirements": [
        "packaging>=20.8",
        "pytz"
[... 320 lines cut ...]
            },
            "unsupported_attributes": {
              "0x0002": {
                "attribute_name": "max_measured_value"
              }
            }
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
``|

MattWestb, 2023-02-10:
The devs need debug logs from ZHA then paring and also then sending commands to and from the device.
One guide for setting the debug longing is here https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-4559550

Kaot93, 2023-02-10:
I attached the Logfile here:

Did my best filtering out the relevant time window.
[HA.log](https://github.com/zigpy/zha-device-handlers/files/10705805/HA.log)

jure123, 2023-02-10:
I think I found a fix for this:

In ts0601_pirmotion.py quirk, find this line:
`converter=lambda x: 10000 * math.log10(x) + 1`

and modify it to:
`converter=lambda x: 10000 * math.log10(x) + 1 if x > 0 else 0`

The quirk failed in full darkness when illumination was 0, because math.log10(0) threw "ValueError: math domain error", so the quirk failed to update the senzor status in ZHA. 

The whole fixed quirk file:

<details>
  <summary>ts0601_pirmotion.py</summary>

```
"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Ota, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaPowerConfigurationCluster,
)

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class SensitivityLevel(t.enum8):
    """Sensitivity level enum."""

    LOW = 0x00
    MEDIUM = 0x01
    HIGH = 0x02

class OnTimeValues(t.enum8):
    """Sensitivity level enum."""

    _10_SEC = 0x00
    _30_SEC = 0x01
    _60_SEC = 0x02
    _120_SEC = 0x03

class PirMotionManufCluster(TuyaMCUCluster):
    """Neo manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update({0xEF09: ("sensitivity_level", SensitivityLevel)})
    attributes.update({0xEF0A: ("keep_time", OnTimeValues)})

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "occupancy",
            dp_type=TuyaDPType.BOOL,
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if not x else 0,
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
        9: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "sensitivity_level",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: SensitivityLevel(x),
        ),
        10: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "keep_time",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: OnTimeValues(x),
        ),
        12: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: 10000 * math.log10(x) + 1 if x > 0 else 0,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        9: "_dp_2_attr_update",
        10: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class PirMotion(CustomDevice):
    """Tuya PIR motion sensor."""

    signature = {
        MODELS_INFO: [("_TZE200_3towulqd", "TS0601")],
        ENDPOINTS: {
            # endpoints=1 profile=260 device_type=0x0402
            # in_clusters=[0x0000, 0x0001, 0x0500],
            # out_clusters=[0x000a, 0x0019]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    PirMotionManufCluster,
                    TuyaOccupancySensing,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```
</details>

@javicalle, would you be so kind and apply this fix to your original ts0601_pirmotion.py quirk you posted here: https://github.com/zigpy/zha-device-handlers/issues/1599#issuecomment-1146852599 ? Actually this log(x) fix has been mentioned in some later comments in that post but many users haven't noticed it. Also your original quirk reports inverted occupancy status (occupied / cleared are inverted). My file above also fixes this (using: `converter=lambda x: IasZone.ZoneStatus.Alarm_1 if not x else 0,`). It would be very helpful to update your original quirk because most users install that version and probably many have hard time experiencing those issues. Thanks!

Kaot93, 2023-02-10:
That makes sense. I will try your fix

My illuminance sensor also only gives values of 1 or 2 while the "measured value" cluster readout give values up to 3000. Is there maybe something wrong with the conversion for my sensor?

kruebsam, 2023-03-13:
As of what I did find out, the structure/content of those "zhaquirks.tuya.mcu" and "zhaquirks.tuya" did change with the latest update as of March 23 resulting in the below errors. I tried to fix it, but couldn´t find out how. Any ideas?
```
Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_pirmotion.py:22
First occurred: 21:47:39 (1 occurrences)
Last logged: 21:47:39

Error setting up entry socket://192.168.2.9:8888 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 110, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_pirmotion.py", line 22, in <module>
    from zhaquirks.tuya.mcu import (
ImportError: cannot import name 'TuyaDPType' from 'zhaquirks.tuya.mcu' (/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/mcu/__init__.py)
```

javicalle, 2023-03-13:
Check here:
https://github.com/zigpy/zha-device-handlers/issues/2250#issuecomment-1453978417

matteozamu, 2023-04-11:
Does anyone have the opposite status on the motion sensor? I mean, I always have motion detect when there is no movement whereas I get no "cleared (no detected)" when there is movement around.

ethzero, 2023-04-17:
> Does anyone have the opposite status on the motion sensor? I mean, I always have motion detect when there is no movement whereas I get no "cleared (no detected)" when there is movement around.

Absolutely! I had a "TS0202 by_TZ3000_6ygjfyll", a simple PIR-only sensor working completely fine for weeks.
I then:
* Added a few "TS0601 by _TZE200_3towulqd" PIR motion + illuminance sensors, found they _didn't_ play nice with HA,
* Installed ZHA quirk [from issue 1599](https://github.com/zigpy/zha-device-handlers/issues/1599), which did then expose the TS0601 motion + illuminance values, however...
* Not only did the quirk seem to mis-report the true Occupancy status for each of the TS0601s, I found that the TS0202 was now also consistently mis-reporting the correct occupancy status.

I've now subsequently disabled the "ts0601_pirmotion.py" (by simply renaming it "ts0601_pirmotion.py.DISABLED" and removing the __pycache__ file twin) and restarting HA. This has effectively rolled-back the TS0202 to it's previous working state and obviously my trio of TS0601s no longer work, again.

Happy to provide any further debugging info if directed to where this can be found in HA.

matteozamu, 2023-05-10:
Hi everyone, any updates about this issue? Is there any information I can provide to help?

closed 2023-11-13 by github-actions[bot]