zigpy/zha-device-handlers #2098: [Legrand wireless switch product id 7 522 85]

Buci007, 2023-01-15:
**Probleme**
I used a Legrand 7 522 85 wireless On/Off switch until last week integrated in HA (Home Assistant 2023.1.0) with Sonoff Zigbee 3.0 Dongle Plus-E. The switch fired ZHA event e.g.: device_ieee: 00:04:74:00:00:c1:c3:be, endpoint_id: 1, command: "off" and I was able to use it for automatization. I had to change the battery and reconnect to the system. The switch is reconnected, but fire only "Checking event." Do not fire any events on push of the buttons. I tried to reconnect several time the reason is the same. I have a Legrand gateway as well and paired to this the switch. In the Legrand system the switch worked. I reset and connected again to HA, the result same, no event fired.
Link to the switch:
https://vikiwat.com/wireless-switch-netatmo-752285-vallena-life-cream-legrand
Link to the Legrand developer site
https://developer.legrand.com/documentation/compatible-product/
Wireless Switch On / Off, Cloud type: NLTOO, Product type ZLE18 is declared ZigBee 3.0 compatible product.

**Describe the solution you'd like**
I would like to use again the switch

<details>

<summary>Signatures</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4129, maximum_buffer_size=89, maximum_incoming_transfer_size=63, server_mask=10752, maximum_outgoing_transfer_size=63, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0019",
        "0xfc01"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Remote switch",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.1.0",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Budapest",
    "os_name": "Linux",
    "os_version": "4.4.180+",
    "run_as_root": true
  },
  "custom_components": {
    "sonoff": {
[... 523 lines cut ...]
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xFEFC), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=76, profile_id=0, cluster_id=32801, data=Serialized[b'R\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=124, rssi=-69)
[0xfefc:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xfefc:1:0x0001]: Configuring cluster attribute reporting
[0xFEFC:1:0x0001] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=92, command_id=<GeneralCommand.Configure_Reporting: 6>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0xFEFC:1:0x0001] Sending request: Configure_Reporting(config_records=[AttributeReportingConfig(direction=0, attrid=0x0020, datatype=32, min_interval=3600, max_interval=10800, reportable_change=1), AttributeReportingConfig(direction=0, attrid=0x0021, datatype=32, min_interval=3600, max_interval=10800, reportable_change=1)])
[0xfefc] Extending timeout for 0x5c request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xFEFC), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=77, profile_id=0, cluster_id=32801, data=Serialized[b'T\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=124, rssi=-69)
[0xfefc:1:0x000f]: bound 'binary_input' cluster: Status.SUCCESS
[0xfefc:1:0x000f]: Configuring cluster attribute reporting
[0xFEFC:1:0x000f] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=94, command_id=<GeneralCommand.Configure_Reporting: 6>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0xFEFC:1:0x000f] Sending request: Configure_Reporting(config_records=[AttributeReportingConfig(direction=0, attrid=0x0055, datatype=16, min_interval=30, max_interval=900, reportable_change=1)])
[0xfefc] Extending timeout for 0x5e request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xFEFC), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=78, profile_id=260, cluster_id=1, data=Serialized[b'\x18\\\x07\x8c\x00 \x00\x86\x00!\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=124, rssi=-69)
[0xFEFC:1:0x0001] Received ZCL frame: b'\x18\\\x07\x8c\x00 \x00\x86\x00!\x00'
[0xFEFC:1:0x0001] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=92, command_id=7, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0xFEFC:1:0x0001] Decoded ZCL frame: PowerConfiguration:Configure_Reporting_rsp(status_records=[ConfigureReportingResponseRecord(status=<Status.UNREPORTABLE_ATTRIBUTE: 140>, direction=<ReportingDirection.SendReports: 0>, attrid=32), ConfigureReportingResponseRecord(status=<Status.UNSUPPORTED_ATTRIBUTE: 134>, direction=<ReportingDirection.SendReports: 0>, attrid=33)])
[0xfefc:1:0x0001]: Successfully configured reporting for 'set()' on 'power' cluster
[0xfefc:1:0x0001]: Failed to configure reporting for '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: [ConfigureReportingResponseRecord(status=<Status.UNREPORTABLE_ATTRIBUTE: 140>, direction=<ReportingDirection.SendReports: 0>, attrid=32), ConfigureReportingResponseRecord(status=<Status.UNSUPPORTED_ATTRIBUTE: 134>, direction=<ReportingDirection.SendReports: 0>, attrid=33)]
[0xfefc:1:0x0001]: finished channel configuration
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2023-01-15:
I dont  have experience of Legrand remotes but if its doing checkins its very positive.
IKEA remotes is not doing it then being new paired and need being `reconfigured` from the device card for getting reporting OK to the system.
Tyre doing one reconfigure of your remote and waking it up the sending commands to it.

Buci007, 2023-01-15:
Thanky you for your response. With reconfigure and push the on and off button on the switch I got a result failed. Details: cluster and binding all are NOK. Reconfigure again with short push on EZ button. (reset and pairing button) The reconfiguration is failed, but binding and reporting - OK. BinaryInput - OK, PowerConfiguration - OK, PollControl - OK, OnOff - NOK, LevelControl - NOK. Ckecking event fired in every hours. No other event fired if On/Off button is pressed. The switch worked 1-2 months without any problem before the battery changed. The switch use the standard zigpy.device.Device quirk. Is there any change can cause this behaviour?

MattWestb, 2023-01-15:
I think (= not knowing) that ZHA is trying binding to much cluster then the device can handling and the reconfigure is failing (if looking in the log om sure you is having binding table full).

I see 2 ways trying fixing it.
1. Delete the device and waiting one minute and adding it new to HA and look if its being OK configured (without reconfigure at first).
2. Manual deleting some bindings or all in the device from device card and `Manage Zigbee Device` (the three ` . ` ) right from reconfigure) and try deleting some or all bindings and adding what you is needing (power, OnOff and Level and perhaps pull control).

From your writing we can see that the 3 first is being made OK and the 2 last is not working and we like have its opposite.

I think its one light controller device so likely is using group binding and not device binding but you must testing what is working for knowing.
If you is deleting all device and group binding you is not reporting any thing to ZHA (can being that is sending to group 0 as default then its working) and normally group 0 is the standard reporting group and the coordinator is the standard device if using device binding.

Sorry i cant saying exact what shall working or not then i have not experience of the device.

Edit: looks its using group bindings (from one other but not the same device) https://www.zigbee2mqtt.io/devices/067774.html.

javicalle, 2023-01-15:
Maybe that quirk can do the job:
* https://github.com/zigpy/zha-device-handlers/issues/1970#issuecomment-1334336305

Probably the `device_automation_triggers` needs to be tunned.

javicalle, 2023-01-16:
Your device's signature match with the already defuned `RemoteDimmer` quirk.

To check, remove any previous quirk for your device.
Copy the current `legrand/dimmer.py` file to your local quirk folder:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/legrand/dimmer.py

Go to the `RemoteDimmer` class and put your device in the `MODELS_INFO` like this:
```python
class RemoteDimmer(CustomDevice):
    """Remote dimmer ."""

    signature = {
        MODELS_INFO: [
            (f" {LEGRAND}", " Remote dimmer switch")
            (f" {LEGRAND}", " Remote switch")
        ],
```

Save changes, delete any `__pycache__` folder and restart HA.

Buci007, 2023-01-16:
Hi! I did what you suggested.

    class RemoteDimmer(CustomDevice):
        """Remote dimmer ."""

    signature = {
        MODELS_INFO: [
            (f" {LEGRAND}", " Remote dimmer switch"),
            (f" {LEGRAND}", " Remote switch")
        ],

I add the missing "," at the end of the first device line, with this update the quirk is loaded without crashing.
[home-assistant_2023-01-16T13-06-30.798Z.log](https://github.com/zigpy/zha-device-handlers/files/10426065/home-assistant_2023-01-16T13-06-30.798Z.log)

The device signature lists that the quirk is used.
[Signature.txt](https://github.com/zigpy/zha-device-handlers/files/10426069/Signature.txt)

The switch is as a binary input sensor detected, as before and only fires "Legrand Remote switch Checkin event was fired" No any response on push Off or On buttons. Thanks!

Buci007, 2023-01-16:
Maybe I was not clear, the switch doesn't work, only fire "Checking event" :-( Do you have any other suggestion?

Buci007, 2023-01-16:
I add the device diagnostic, maybe it can help
[Legrand  Remote switch diagnostic.txt](https://github.com/zigpy/zha-device-handlers/files/10426255/Legrand.Remote.switch.diagnostic.txt)

Buci007, 2023-01-16:
Thank you! You are right, but what can I do others? I binded it several times. I deleted the device and added again with "Add device". I made factory reset of the switch and tried everything was possible on my known based on the Legrand developer site. I did Reconfiguration as well. Binding and Reporting in the head I got OK. In the table LegrandPowerConfiguration, BinaryInput, PollControll i got OK. On/Off and LevelControl - NOK  binding. In the Zigbee device handling I cannot bind manually. There are listed the devices can bind, but nothing happen if I try to bind. I really appreciate if you can advice me, what can I do.

MattWestb, 2023-01-16:
Delete the device in ZHA and wait one minute (the database need committing the deletion) and then adding it in ZHA and only giving one name and aria and test if its sending events from OnOff cluster.

Buci007, 2023-01-16:
I deleted the device and waited 2 minutes. Then I restarted the system and device with name "Legrand Remote switch" is back with their Entities. After deleting it with reload the configuration of the SONOFF Zigbee 3.0 USB Dongle the switch was back again. 
I must deleted it 5 times with restart the system as well to avoide, that the switch come back.
Currently I have 28 devices and 142 entities listed in the coordinator excluded this Legrand Switch.

After the switch is deleted, I tried to add with the standard and with the custom quirks as well.
I tried to reconfigurate as well.
PollControl sometimes binded, sometimes not. The On/Off and LevelControll clusters never binded.
As you wrote there are the 2 cluster binding missing.
[Legrand  Remote switch Diagnostic with the custom quirk.txt](https://github.com/zigpy/zha-device-handlers/files/10426957/Legrand.Remote.switch.Diagnostic.txt)
I got Checking Events and Battery info. 
Is there any solution to force these bindings?

javicalle, 2023-01-17:
This is a battery device. Have you tried to keep it awake during all the pairing process?
Just push the button every second or so to prevent it from getting sleep until the pairing process ends.

Buci007, 2023-01-17:
I tried several type before, but I wanted to answer you after I made the test again. I found the info in HA community pages, that the device(s) come back automatically if they are not factory reset. I deleted the switch and did a factory reset with 20 second push on the EZ button. Waited 5 minutes. I started the Add device process. Pushed 1 second the EZ (pairing) button on the switch. Green light started blinking about 10-15 second. I pushed per second the Off button until the process stopped, about 30-40 second. On/Off cluster binded, the switch fire On / Off event :-) I guess the battery clusters didn't binded, no battery info :-( But I'm happy that it works again :-)
Where can I see the the bindings if I missed to save the Binding login? I download the diagnostic data of the device and compared the working and not working variant. They are the same there and there isn't any infromation about bindings..
Thanks for you both!!!!

MattWestb, 2023-01-17:
That is one personality of the device i have not seen before !!!
Great that you is have getting it working !
You can trying adding binding of power configure cluster and deleting some other if its not working. And now you is knowing how getting it back if it stop working :-))

If you like making more advanced binding work you can using https://github.com/mdeweerd/zha-toolkit it working very well but is little more technical but it can reading all bindings in normal devices.

javicalle, 2023-01-17:
Glad to know you get it working again!
Just to be sure, because you said that device was already working before the battery change (I didn't read this part 😓):
Is the quirk needed for this device?

It doesn't cost anything to include the new device in the quirk, but I'd like to be sure.

Buci007, 2023-01-17:
No, I didn't use the custom quirk. I was able to bind with the standard. "zigpy.device.Device"

WLR13, 2023-05-10:
Hi all,
I'm unfortunately having issues with battery powered switches from Legrand (067723) as well. The buttons (on/off) events only show up in HA assistant sometimes, but not always.  When it works and I continue to press another button within <1 sec, the button presses are usually recognized within HA. However if I wait a little longer, nothing happens. Only after a couple of more clicks / seconds, HA starts recognizing the events again.

Does this issue sound familiar to anyone?

I've 3 Legrand battery switches of which 2 show an "N" at the end of their product number (067723N - google is not showing anything this). Only the switch without the N is working as expected...

Did anyone have similar issues / tips how this could be solved?

Thanks!

EzEsAz, 2023-05-11:
Hi WLR13, 
Check the firmware of the switches.
Check the distance between switch and coordinator (big distance without router supported can cause this probleme)
Check Legrand developer forum [https://developer.legrand.com/forums/](url)
067723 is a Céliane wireless switch. There are 2 verions [067723A](https://www.legrand.fr/pro/catalogue/commande-sans-fils-celiane-with-netatmo-pour-eclairage-et-prise-connectes-blanc-sans-plaque) and [067723L](https://www.legrand.fr/pro/catalogue/commande-sans-fils-sans-pile-self-e-celiane-with-netatmo-pour-eclairage-et-prises-enjoliveur-blanc)

WLR13, 2023-05-14:
Hi EzEsAz,

Thanks, I'll do that. Just an update on the behavior. I was wrong when I said that the switch does not work sometimes. It just takes 5-10 seconds for the lights to come on... Sometimes it works instantly (as expected).

Will update here if I find out anything at Legrand forum.

closed 2023-11-17 by github-actions[bot]