zigpy/zha-device-handlers #2099: [Device Support Request] TZE200_v6ossqfy TS0601

Schlauwi, 2023-01-15:
**Is your feature request related to a problem? Please describe.**
Bought a human presence sensor to identify presence.

**Describe the solution you'd like**
I would like to detect the presence of a person. In addition, the time of presence or absence should be detected.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_v6ossqfy",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

**Additional context**
Product: [https://www.amazon.de/dp/B0BQB9WN9L?psc=1&ref=ppx_yo2ov_dt_b_product_details](url)
More information can also be found here: [https://www.zigbee2mqtt.io/devices/TS0601_human_presence_sensor.html](url)

javicalle, 2023-01-15:
Duplicate of #2059

javicalle, 2023-01-15:
There are a cutom quirk available: 
https://github.com/zigpy/zha-device-handlers/issues/2059#issuecomment-1370892759

We can follow the discussion there.

closed 2023-01-15 by javicalle

Schlauwi, 2023-01-15:
> There are a cutom quirk available: [#2059 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2059#issuecomment-1370892759)
> 
> We can follow the discussion there.

thanks a lot - quirk is working fine

javicalle, 2023-01-15:
I still need to review the HA part, but if you are familiar with it, do you mind create the PR in the `ts0601_motion.py` file?

Thanks in advanced.

Schlauwi, 2023-01-15:
I'm really sorry, but unfortunately I‘m not familiär with ist.