zigpy/zha-device-handlers #2100: [Device Support Request] TZE200_68nvbio9 TS0601

Schlauwi, 2023-01-15:
**Is your feature request related to a problem? Please describe.**
I bought a roller motor, which was also recognized as this, with the functions "open", "stop" and "close". Unfortunately, however, the battery level is missing.

**Describe the solution you'd like**
I would like to see the value of the battery level in %.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_68nvbio9",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_cover.TuyaMoesCover0601"
}
```

**Additional context**
Product: `https://de.aliexpress.com/item/1005004798347508.html?spm=a2g0o.order_list.order_list_main.4.d2ce5c5fitIdnr&gatewayAdapt=glo2deu`

javicalle, 2023-01-15:
We will need the debug logs from the device to get the battery status DP.

Here are instructions to enable the debug logs:
* https://www.home-assistant.io/integrations/zha/#debug-logging

It can be very verbose. Try to catch the device report after a command is send. There must be some `No datapoint handler for...` or similar.

closed 2023-07-21 by github-actions[bot]

wbkang, 2023-09-02:
Sorry to open this old issue, but I have the exact same device, and I am willing to follow up. I tried turning on the debug, and grepped through 2 weeks worth of logs, but I am not seeing any such message for this device. Is there anything else I can try?

javicalle, 2023-09-03:
The quirk should log messages like:
* "Received Attribute Report. Command is 0x%04x, Tuya Paylod values [Status : %s, TSN: %s, Command: 0x%04x, Function: 0x%02x, Data: %s]"
*  Received Attribute Report - Unknown Command. Self [%s], Header [%s], Tuya Paylod [%s]

Try to grep yours devices IEEE from logs and check what is telling us.

wbkang, 2023-09-07:
Sorry it turns out I only have a bit more than 24 hours worth of data

`-- Journal begins at Tue 2023-09-05 08:39:12 EDT, ends at Wed 2023-09-06 22:21:33 EDT. --`

This is the list of all unique lines that I have:

```
% journalctl -t docker | grep a4:6d:d4:ff:fe:20:1c:f2 | egrep -i "Received Attribute Report. Command is|Received Attribute Report - Unknown Command." | cut -d' ' -f 8- | sort | uniq
DEBUG (MainThread) [zhaquirks.tuya] a4:6d:d4:ff:fe:20:1c:f2 Received Attribute Report. Command is 0x0002, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0202, Function: 0x00, Data: [4, 0, 0, 0, 0]]
DEBUG (MainThread) [zhaquirks.tuya] a4:6d:d4:ff:fe:20:1c:f2 Received Attribute Report. Command is 0x0002, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0202, Function: 0x00, Data: [4, 0, 0, 0, 100]]
DEBUG (MainThread) [zhaquirks.tuya] a4:6d:d4:ff:fe:20:1c:f2 Received Attribute Report. Command is 0x0002, Tuya Paylod values[Status : 0, TSN: 1, Command: 0x0203, Function: 0x00, Data: [4, 0, 0, 0, 0]]
DEBUG (MainThread) [zhaquirks.tuya] a4:6d:d4:ff:fe:20:1c:f2 Received Attribute Report. Command is 0x0002, Tuya Paylod values[Status : 0, TSN: 1, Command: 0x0203, Function: 0x00, Data: [4, 0, 0, 0, 100]]
DEBUG (MainThread) [zhaquirks.tuya] a4:6d:d4:ff:fe:20:1c:f2 Received Attribute Report. Command is 0x0002, Tuya Paylod values[Status : 0, TSN: 1, Command: 0x020d, Function: 0x00, Data: [4, 0, 0, 0, 75]]
```

I wonder if the number is actually related https://github.com/zigpy/zha-device-handlers/issues/2562 🤷‍♂️

atrambauer, 2024-08-20:
Hi, i installed this device via ZHA and it is working great and it would love to have battery level available.
How can i help to debug and get the data needed?

atrambauer, 2024-08-20:
> Hi, i installed this device via ZHA and it is working great and it would love to have battery level available. How can i help to debug and get the data needed?
I got it working adding cutom quirk https://github.com/zigpy/zha-device-handlers/issues/3118