zigpy/zha-device-handlers #2101: [BUG] TS011F The physical button does not change the ON/OFF status in HA.

jasimancas, 2023-01-15:
The device measures well, and the functions are adequate, however, in my case, when I manually press the button to turn on the plug in HomeAssistant it does not refresh the status, it remains OFF always, however the KW and KWH measurements do vary, measuring the consumption correctly.

Is it a bug in the plug or do I have to make some adjustment?

My exact model is _TZ3000_dpo1ysak
Quirk: zhaquirks.tuya.ts011f_plug.Plug

Translated with www.DeepL.com/Translator (free version)

TheJulianJES, 2023-01-16:
IIRC the Tuya ``TS011F`` plugs (e. g. BW-SHP15) do not support attribute reporting for the on/off attribute.
These plugs would need to be polled constantly to work around this issue (polling is somewhat bad/annoying).

jasimancas, 2023-01-17:
I mean that the only option I have to know the status is to always turn it off and on from HA and never use the physical button? Does it only happen with this model?

TheJulianJES, 2023-01-17:
Only newer Tuya based plugs seem to have this issue. I have a couple of them too (showing the same issue), while others work fine.
ZHA could only implement an ugly workaround by polling these models of plugs. (Proper plugs send an attribute report (push message) that their state was changed)

jasimancas, 2023-01-17:
This only happens with third party zigbee solutions, right? In the case of using the Tuya Gateway would it return the status correctly? I mean it is not a failure of the device firmware but of the programming itself?

MattWestb, 2023-01-17:
The device firmware have not implemented status reporting for all functions and very likely is tuya doing workaround in there cloud for getting it working.
If the device was working as the Zigbee standard is have doing the reporting OK but sadly is tuya not following the standard in most of there products.

TheJulianJES, 2023-01-17:
It is a firmware issue that the plugs do not follow Zigbee specs (so they don't support attribute reporting for this).

Tuya specially tries pretty hard to only make their stuff work on their own gateways (and block third party gateways).
I'd even show a warning message before you could add a Tuya device to ZHA.
Have a look on this issue tracker, most issues are Tuya related.

I don't own a Tuya hub, so can't tell you if the status would be correct there.

It should by possible to work around the issue in ZHA itself though by constantly polling this specific plug (but that's a workaround and not a solution). But it's the only way I see to make this work for Tuya's firmware.

jasimancas, 2023-01-17:
Today I should receive a TUYA gateway that I ordered to verify that all my devices have the latest firmware version, I can test it and send the result to us, because I doubt that a sniffer can be used in the TUYA network to get data from the zigbee traffic between TUYA and the device.

MattWestb, 2023-01-17:
You can sniffing the Zigbee network also if using one tuya ZBGW you only need one hardware that can doing it like one TI CC-2531 with sniffer firmware or one EZSP with normal coordinator firmware.

Also good having one tuya ZBGW for knowing the functions tuya have implanted and we dont knowing and can sniffing how they is setting them up in there system.

jasimancas, 2023-01-17:
https://es.aliexpress.com/item/1005001794162200.html

This is the gateway I bought, model G01-PRO, I also have a CC2531 with the sniffer firmware installed.

jasimancas, 2023-02-07:
When I press ON/OFF button in OFF mode:
[image]

When I press ON/OFF button in ON mode:
[image]

The plug report to TUYA app the status.

Its possible adapt this to ZHA?

jasimancas, 2023-02-09:
Any idea with this @javicalle ?

javicalle, 2023-02-10:
Don't seems to be a 'EnchantedDevice' problem.

Can you do the same with the device in ZHA?
I'm not familiar with the Zigbee protocol but maybe we can see something by comparing. I recall some problems with the 'Direction' value in some devices but I don't know if that can be the case.

MattWestb, 2023-02-10:
Hola @jasimancas Its looks tuya ZBGW have setting up attribute reporting OK and its working for your device. Normally tuya is not setting up mush binding and reporting then there device have defaults in the firmware that working OK what i have seen in sniffs.
Can you doing one sniff the adding the plug to the tuya ZBGW and we can see what it setting up the device ?
Its exposing the tuya ZVGW network key and it can being changed by deleting the ZBGW in the app and reading it new then tuya is making one new network.

Command sent in the wrong direction is one "personality" of tuya devices that can making problems in both ends then acks is not working OK.

jasimancas, 2023-02-10:
The tests have been done with the plug connected to the TUYA gateway, and sniffing the traffic.

Do I connect it to the ZHA gateway in the test environment and sniff doing the same actions?

MattWestb, 2023-02-10:
Yes im interesting of tuya ZBGW paring and also some on/off from the device.
@javicalle is interesting of how ZHA is doing and we can comparing it if you do one sniffing of that 2.

jasimancas, 2023-02-11:
Tell me if it is enough, I deleted all my test devices, with only the TUYAZBGW on the network I added the plug, waited a few seconds, turned on, waited, turned off, waited, turned on, waited and turned off.

If more tests are needed, let me know and I will do them.

PS: No problem with the KEY, this zigbee network is always OFF, I only turn it on for testing, the GW is not in production (if that is why you were telling me to change the network key).
[TS011F.zip](https://github.com/zigpy/zha-device-handlers/files/10713804/TS011F.zip)

MattWestb, 2023-02-11:
Its doing many strange things but in the end the OnOff is reported to the coordinator.
I have not looking so exactly but the coordinator is requesting application version from the basic cluster all the time.
@jasimancas can you looking if its still requesting it after some times and also if its reporting the metiering parameters to the coordinator or if its being pulled (the coordinator is sending read attribute and the device is sending it s one replay) ?

I shall looking little deeper later then i have time digging more.

MattWestb, 2023-02-11:
After getting the network key the system is sending tuya magic.
Then the system is writing one attribute on basic cluster:
```
IEEE 802.15.4 Data, Dst: 0xde3e, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xde3e, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Basic (0x0000)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 18
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 11
    Frame Control Field: Profile-wide (0x10)
    Sequence Number: 11
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 19
        Attribute: Unknown (0xffde)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 19 (0x13)
```
Then reading some attribute on ObOff cluster:
```
IEEE 802.15.4 Data, Dst: 0xde3e, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xde3e, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 255, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 255
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 19
ZigBee Cluster Library Frame, Command: Read Attributes, Seq: 12
    Frame Control Field: Profile-wide (0x10)
    Sequence Number: 12
    Command: Read Attributes (0x00)
    Attribute: OnOff (0x0000)
    Attribute: OnTime (0x4001)
    Attribute: OffWaitTime (0x4002)
    Attribute: Unknown (0x5000)
    Attribute: Unknown (0x8000)
    Attribute: Unknown (0x8001)
    Attribute: Unknown (0x8002)
    Attribute: Unknown (0x8003)
```
and replay from the device:
```
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0xde3e
ZigBee Network Layer Data, Dst: 0x0000, Src: 0xde3e
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 94
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 12
    Frame Control Field: Profile-wide (0x18)
    Sequence Number: 12
    Command: Read Attributes Response (0x01)
    Status Record
        Attribute: OnOff (0x0000)
        Status: Success (0x00)
        Data Type: Boolean (0x10)
        On/off Control: Off (0x00)
    Status Record
        Attribute: OnTime (0x4001)
        Status: Success (0x00)
        Data Type: 16-Bit Unsigned Integer (0x21)
        On Time: 0.0 seconds
[... 104 lines cut ...]
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 96
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 14
    Frame Control Field: Profile-wide (0x18)
    Sequence Number: 14
    Command: Read Attributes Response (0x01)
    Status Record, Uint: 113778
        Attribute: Current Summation Delivered (0x0000)
        Status: Success (0x00)
        Data Type: 48-Bit Unsigned Integer (0x25)
        Uint48: 113778 (0x000000000001bc72)
```
Then the device is starting looking if it can updating the TC-Link key and its OK and doing it OK.

Then the device is reporting OnOff and Simple Metering.

Its some more unknowns commands sent and attribute reported from unknown attributes but i think its not important.
Only strange thing is the reading of application version.

Also the device is reading time from the coordinator and it getting it OK (tuya extended shuddering with strange commands). 

I have not looking on frames flags and manufacture code tagging but i think the key is reading and writing attribute then the device is in init more.

Can we trying sending some of this to the device after the tuya magic is being sent ?

jasimancas, 2023-02-11:
Tell me what you need me to do and I'll get right on it.

I can find out the parameters it reports that we don't have "inventoried" although maybe it's internal to the device?

MattWestb, 2023-02-11:
We are casting the tuya magic from the quirk then it joining and i think we need casting the  attribute reads for the OnOff cluster for getting it start reporting the status changes.
Also the Metering and Measurement cluster can needing it but i think we can starting with testing OnOff cluster.

Is the device reporting OK from Metering and Measurement cluster in tuya ZBGW  (it can being very long between the reporting) or its the ZBGW pulling the device (reading the attribute in the device for getting the information) ?

If we is needing more for getting the device reporting OK i think we is having it in your sniff only need digging deeper it it but i think the attribute thing is the key for getting it working.

jasimancas, 2023-02-11:
When I press the physical button the status change in the app is instantaneous, as well as the registration in wireshark, in the case of the measurement I am not so clear.

I understand that it could be seen seeing the order of the sniffed zigbee frames to see if it is the TYZBGW who asks for the measurement data or on the contrary it is the plug that sends it.

The logic says that it would be the device who would have to inform the ZBGW of the state that changes, another thing is that the ZBGW is asking for the state every X time, but in a change of state the device is who must indicate the change to the ZBGW so that this one refreshes the interface/control application.

javicalle, 2023-02-11:
If someone want to try it, you can simulate the `OnOff` readings in the quirk overwriting the `spell` this way:
```python
class Plug(EnchantedDevice):
    """Tuya plug with restore power state support."""

    async def spell(self) -> None:
        """Initialize device so that all endpoints become available."""
        attr_to_read = [4, 0, 1, 5, 7, 0xFFFE]
        basic_cluster = self.endpoints[1].in_clusters[0]
        await basic_cluster.read_attributes(attr_to_read)
        onoff_read_attr = [0x0000, 0x4001, 0x4002, 0x5000, 0x8000, 0x8001, 0x8002, 0x8003]
        await self.endpoints[1].on_off.read_attributes(onoff_read_attr)

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            .../...

```

The write attribute can be done the same way:
```python
        await basic_cluster.write_attributes({0xffde: 0x13})
```

jasimancas, 2023-02-12:
I just did a test and something strange happens, in the production environment if I press the physical button it does nothing, however in the test environment I have paired it again and now if it responds, it changes the interface switch when I press, I have done it several times and the behavior is as desired.

I have copied the custom_component/zha and the quirk from test to production and the behavior is the same.

Maybe I have to remove it and add it again?

MattWestb, 2023-02-12:
@jasimancas GREAT !!

Yes you need deleting the device from ZHA waiting one minute and then adding it new.
The casting of the magic is only being done then the device is doing one new joining and then the in the system is not online then the quirk is being loaded at system start.

And the good thin is look like tuya have only making one more magic but not tricking with all the flags in the frame handling that is more difficult fixing.

MattWestb, 2023-02-12:
@javicalle I think its one candidate making one `EnchantedOnOffDevice` class in your `_INIT_`  ;-)

jasimancas, 2023-02-12:
I have not applied the modification of @javicalle , I have simply deleted it and put it back and I have detected that.

jasimancas, 2023-02-12:
Yeah, deleted, waited 1 minute, added again and now it's responding correctly!!!

We can close this issue unless you want some more tests to fine tune the behavior.

MattWestb, 2023-02-12:
If its working as the normal tuya magic is the device keeping it until being retested **and** repowered without being added to one system (normal its can being retested after being in one tuya ZBGW and is keeping it then moved to one open zigbee system if not being repowered).

I like getting more testing and @javicalle can implanting it OK in the system so we is knowing its working OK.

Is the power measuring working OK = being reported OK ??
I was seen in the sniff that the 2und endpoint is sending back the requests on the first endpoint so it can making problems for the system getting the data OK.

Do you have more types of TS011F devices or only this type ?
I have the LIDL 3AC 4 USB but its working OK and dont need it (i think) so cant testing so mush.

So pleas keep it open until is in place :-)))

jasimancas, 2023-02-13:
I only have that type, my intention is to implement more to control the cost of home (washing machine, dishwasher, refrigerator, etc), but I am looking for a recessed socket.

By the way, any idea of a good Zigbee one?

MattWestb, 2023-02-13:
tuya was making good ones with good measurement chips that was sending digital data over UART to the Zigbee module with Silaabs chip.
Most new is having one cheaper measurement chips that is using frequency for data and Telink Zigbee module that is having problems with the frequency "decoding" and if not getting updated firmware they is power off under 250 V and also making many wrong measurement (peeks or zero V).
(one very long Z2M issue with power off from TS011F if you like reading more)

For the moment i dont putting money on this devices than its so mush problems with them or going with some known brand that is working but is too expensive for my.

I think this device is having some alarm attributes that is not implanted like over current alarm that can being added but need some with device that can testing it.

closed 2023-08-19 by github-actions[bot]

TheJulianJES, 2023-10-23:
Since I saw this issue referenced recently:
This issue was fixed in newer Home Assistant Core versions. You might need to re-configure or re-pair the plug once though.