zigpy/zha-device-handlers #2103: [BUG] _TZ3000_npzfdcof - Seen as a light instead of a switch

HollyFredD, 2023-01-16:
**Zigbee Model:** TS0001
**Zigbee Manufacturer:** _TZ3000_npzfdcof
**Description:** Wall switch module
**Model :** TuYa WHD02
**White label :** TuYa iHSW02, Aubess TMZ02

Pairing with ZHA work, but device is seen as a _light_

**Exposed information :**
- on/off toggle is ok but displayed as _light_ vs. _switch_ or _power_
- _power_on_state_ is ok
- _backlight mode_ should be fine, but I didn't have the chance to test it. Is it the LED control ?

MattWestb, 2023-01-16:
The device class is `ON_OFF_LIGHT` in the  device and shall not being changed in the quirk.
Also HA is blocking changing the device class for lights :-((.
But you can changing the icon if it to one light by clicking on the light and then setting and icon and selecting what you like.

HollyFredD, 2023-01-16:
OK thanks for your answers. Will use the modify device type in configuration.yaml.
Considering the device class and the HA rules, let's consider this request as closed.

closed 2023-01-16 by HollyFredD

TheJulianJES, 2023-01-16:
Yeah, it's unfortunate that Tuya tends to report most switches as lights, but if we would modify everything with a quirk, we'd break existing installations (and some people want to use them as lights).

```yaml
zha:
  device_config:
    84:71:27:ff:fe:93:17:24-1:    # format: {ieee}-{endpoint_id}
      type: "switch"              # corrected device type
```
So just adding that seems to be the best solution. (Note: It's the IEEE of the device then a "-" and then the endpoint. You can find the endpoint from your signature. If the switch has multiple entries, you need those last two lines for every endpoint)

aistis-, 2023-01-25:
Do the sensors works for this switch? Mine does not - it is always showing 0.

[image]

HollyFredD, 2023-01-25:
Mine works. Can't tell why it how, but it works