zigpy/zha-device-handlers #2104: Remote as AlarmControlPanel

mycanaletto, 2023-01-16:
The zigbee remote controls (Heiman or Woox) are seen as an AlarmControlPanel. This is a choice of the integrators that does not correspond to the use

The problem is that if you want to monitor the events you will find that it sends several in a row and **it is impossible to differentiate them.**

The temporary solution is to process only the first one in single mode and pause it. But this is not a totally reliable solution, especially if you operate several of them...

I tried to change the type :

```
zha:
  device_config:
     a4:c1:38:96:0b:cf:xx:xx-1:
       type: "remote"  # or button
```

It doesn't work

I noticed that under Z2MQTT it works fine as a remote, the behavior is normal and we can process the information correctly.
```

> arm_day_zones
> arm_all_zones
> disarm
> emergency

```

So under ZHA we should be able to choose the operating mode.

I tried to play with clusters or ZhaToolkit without success...

Information and updates welcome...

TheJulianJES, 2023-02-04:
Please fill out the issue template. Provide the device signature and diagnostic information.

mycanaletto, 2023-03-08:
Heiman HS1RCEF
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "HEIMAN",
  "model": "RC-EF-3.0",
  "class": "zigpy.device.Device"
}
```
Woox R7054
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "_TZ3000_0zrccfgx",
  "model": "TS0215A",
  "class": "zigpy.device.Device"
}
```

mycanaletto, 2023-09-04:
Not resolved...

closed 2024-03-10 by github-actions[bot]