zigpy/zha-device-handlers #2106: Using ZBDongle-E in router mode, getting info in the logs to report the issue here.

lucius-the, 2023-01-16:
I am using ZBDongle-E (plugged in my HomeAssistant PC) for a coordinator in ZHE. And using another ZBDongle-E, flashed in router mode, as a router to extend range. Because all of my end devices are battery powered, I used a spare ZBDongle-E to make a router.

I am getting this in HA logs:
```
Logger: homeassistant.components.zha.core.channels.base
Source: components/zha/core/channels/base.py:486
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 15 January 2023 at 18:12:34 (2 occurrences)
Last logged: 15 January 2023 at 18:12:34

[0x73B0:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected
```

And for this reason I am posting here.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4742, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008"
      ],
      "out_clusters": []
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "SONOFF",
  "model": "DONGLE-E_R",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

[... 615 lines cut ...]
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
This stick seems to work ok in router mode, because I am seeing that end devices are connected through it. I do see that this ZBDongle-E router is represented in HA as a being some kind of a dual light switch with colors to choose from. There are no switches and lights, of course, it's just a stick, frashed with router firmware, plugged into a USB power supply that's plugged in to a wall socket.

Reason I am posting here is because the logs are instructing me to do so. I guess there's a reason why those messages are logged, so... providing info as requested !

MattWestb, 2023-01-16:
I have making 2 versions of one _Zigbee Controller Bridge_ and have implanted one light device for getting something in the ZHA GUI but its making the same bad light level setting on the cluster. 
Its one warning that is very harmful and can being ignored.

@TheJulianJES Shall we doing one quirk for the Sonoff repeater and deleting all cluster and only having Identity cluster so it can being used in the GUI ?

TheJulianJES, 2023-01-16:
So just to confirm, ``0x73B0`` is the NWK id of the ZBDongle-E router?

MattWestb, 2023-01-16:
Its have "0x0006" as in so at least one (light) switch.

TheJulianJES, 2023-01-16:
Right. Completely missed this:
> ZBDongle-E router is represented in HA as a being some kind of a dual light switch with colors to choose from

Yup, I think we can just remove all relevant clusters via a quirk then

MattWestb, 2023-01-16:
@lucius-the Can you testing this custom quirk 
[SonoffR.zip](https://github.com/zigpy/zha-device-handlers/files/10428782/SonoffR.zip) ?
It shall deleting an needed clusters and you is not getting any warnings then the system is loading.

lucius-the, 2023-01-17:
Uhm... how do I test this quirk ? Where Do i need to put files from the ZIP ?

MattWestb, 2023-01-17:
Set up local quirk in (Z)HA and putting the unzipped file in the directory and restart HA.
Instruction https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.

Sorry i see that i have missing the second endpoint so its not working and i need fixing it.

MattWestb, 2023-01-17:
Updated quirk with  endpoint 2 [SonoffR.zip](https://github.com/zigpy/zha-device-handlers/files/10433028/SonoffR.zip).

MattWestb, 2023-01-17:
@lucius-the Can you posting one link to the router firmware you have installed ?
I was looking on the normal one and im missing some functions so i dont knowing if its the same.

lucius-the, 2023-01-17:
I used this firmware for router: https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router

I'll try to use the new quirk now, following your instructions and will report back

MattWestb, 2023-01-17:
Great i hope i have getting all parameters OK and the quirk is being loaded.
Then i renaming the device class to **itead** router then its one more model out there with model **xsp1989** https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle/efr32mg21_zigbee_router_6.7.10.gbl also as signed for origianl Sonoff ZBB.

MattWestb, 2023-01-17:
### OPS !!
That i have not seen before and i was testing the quirk on one of my test system.
ZHA is trying loading the device from the database with the old config and cant finding all parts then i have deleting them.
Can you deleting the router and waiting one minute so the database is writing the deletion and then adding it new ?

lucius-the, 2023-01-17:
Yes, but I'll have to try this in a few hours, can't do now...

lucius-the, 2023-01-17:
Ok, so I have deleted the SONOF router device and waited a few minutes. Now I've re-activated the quirks again and restarted HA. Starts without issues. Joined the SONOFF router device. Now it's not showing the light objects any more.

However, 5 of my end devices (that were connected through this router) seem to be disconnected now... There are all Aqara senzors, was difficult to get them joined. I will wait and see if they reconnect. At the moment they are not reporting on motion detect.

lucius-the, 2023-01-17:
Also got this:
```
Logger: bellows.thread
Source: runner.py:120
First occurred: 21:18:05 (1 occurrences)
Last logged: 21:18:05

Attempted to use a closed event loop
```

and this:
```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:97
First occurred: 21:18:10 (1 occurrences)
Last logged: 21:18:10

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/v8/__init__.py", line 51, in pre_permit
    await self.setPolicy(
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```
In HA logs.

MattWestb, 2023-01-17:
Aqara device is not jumping to one other router is its parent is offline they silent leaving the network without saying god by.
If they is having one paring  / reset button  press it briefly and the LED is blinking and they is jumping to one router in the near. If its not working you need repairing them without deleting them. Sorry for that :-((

The error is initiating the network and have with permit joining but i have not seen it before and i dont knowing the code. If all is working i think its OK.

lucius-the, 2023-01-17:
Ok, I'll wait and see if they reconnect, if not will try pairing again (with these devices it rarely goes fine from first) but ok ! Thank you for a quick solution !

Also, do I need to keep this in quirks or will it be merged with some next release (and then I have to remove it or something?)

MattWestb, 2023-01-17:
Great !!
I doing one PR with the quirk and changing the name so we can adding the second version **then** its coming.
Then the PR is merged and (Z)HA is updating with it in you shall deleting the local quirk for not getting problems in the future (local quirk is overriding ZHA ones) if somthing is changed in ZHA.

TheJulianJES, 2023-02-04:
Is there already a PR for this?

MattWestb, 2023-02-04:
PR made with last reported working quirk and "branding" the device class Itead.

TheJulianJES, 2023-02-04:
> I do see that this ZBDongle-E router is represented in HA as a being some kind of a dual light switch with colors to choose from. There are no switches and lights, of course, it's just a stick, flashed with router firmware, plugged into a USB power supply that's plugged in to a wall socket.

If possible, could you check if (without the quirk) the dimmable light entity dims the on-board LED?

TheJulianJES, 2023-02-15:
@lucius-the Can you check (without the quirk) if the dimmable light entity dims the on-board LED?

Z2M seems to expose that option and if we just remove it, we could break functionality for some.

closed 2023-08-22 by github-actions[bot]

ThiSaadeh, 2023-10-10:
After update 2023.10.1 I'm getting the same error on my ZBDongle-E in router mode.

2023-10-10 18:10:24.899 WARNING (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x3458:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected

The network address is also confirmed to be the ZBDongle.

def1149, 2023-10-19:
Has this issue been addressed? I'm getting the same error on my Dongle-E in Router mode.

UsbBitMinter, 2024-01-17:
Same here, I am using the Dongle-E as well as a router and frequently getting the same message in the logs. 
Message:  
[Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected