zigpy/zha-device-handlers #2111: TS110E 2 Gang no neutral dimmer - _TZ3210_4ubylghk

colombianlizard, 2023-01-17:
Hi guys, I have a couple TS110E no neutral 2 gang dimmer switches that only work with the On/Off part but the dimmer functionality is not working from HA, I have a couple more 1 gang that use the actual quirk and those work, is there any way the quirk can be adjusted to make it usable for the 2 gang model?
Here is the DS for my 2 gang modules:

`
{`
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0xef00"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_4ubylghk",
  "model": "TS110E",
  "class": "zigpy.device.Device"
}
`

javicalle, 2023-01-18:
There is a quick and dirty proposal:
```python
class DimmerSwitch2Gang(TuyaDimmerSwitch):
    """Tuya Dimmer Switch Module 2 Gang."""

    signature = {
        MODELS_INFO: [
            ("_TZ3210_4ubylghk", "TS110E"),
            ("_TZ3210_weaqkhab", "TS110E"),
        ],
        ENDPOINTS: {
            1: {
                # "profile_id": 260,
                # "device_type": "0x0101",
                # "in_clusters": ["0x0000","0x0004","0x0005","0x0006","0x0008","0x0300","0xef00"],
                # "out_clusters": ["0x000a","0x0019"]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                # "profile_id": 260,
                # "device_type": "0x0101",
                # "in_clusters": ["0x0004","0x0005","0x0006","0x0008","0x0300","0xef00"],
                # "out_clusters": []
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                # "profile_id": 41440,
                # "device_type": "0x0061",
                # "in_clusters": [],
                # "out_clusters": ["0x0021"]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    F000LevelControlCluster,
                    Color.cluster_id,
                    TuyaManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    F000LevelControlCluster,
                    Color.cluster_id,
                    TuyaManufCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

You would need to add some imports (`Color` and `TuyaManufCluster`):
```python
from zigpy.zcl.clusters.lighting import Color
```
```python
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaDimmerSwitch,
    TuyaManufCluster,
    TuyaZBExternalSwitchTypeCluster,
)
```

Save changes, restart HA and re pair the device.

For any error we will need the debug logs from the device.

colombianlizard, 2023-01-19:
Thanks for your help @javicalle, I'm dealing with issues with my zhaquirks install as I can't make the quirk file to be catch by the module, I tried editing the prior ts110e.py file but it looks like at some point I installed zha-quirks 0.0.90 using Python pip and can find the right fiel to modify, I created a custom quirk with your code and nothing happens, I will let you know as soon I get to sort out the problem....

javicalle, 2023-01-19:
I have updated the quirk from my comment.
Remove the `Basic.cluster_id` from endpoint 2.
At the `ENDPOINTS` definition and the `replacement` too.

colombianlizard, 2023-01-19:
Now the quirk is being loaded and the two devices have changed their device signature but the dimmer functionality is still not working, just on/off....

javicalle, 2023-01-19:
Enable the debug logs from the integration view. Interact with the device both from HA and from the physical device and attach the logs

colombianlizard, 2023-01-19:
Here is the log file, the device I'm refering to is Nwk: 0xbdd7
[home-assistant_2023-01-19T18-14-32.637Z.log](https://github.com/zigpy/zha-device-handlers/files/10459953/home-assistant_2023-01-19T18-14-32.637Z.log)

javicalle, 2023-01-19:
I can see a few `LevelControl` reports in your logs:
```
2023-01-19 13:11:59.676 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e6a000700aa00
2023-01-19 13:11:59.676 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2023-01-19 13:11:59.677 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2023-01-19 13:11:59.677 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>)
2023-01-19 13:11:59.677 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x176a000800010001
2023-01-19 13:11:59.679 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x176a0026001f00220200000102d7bd0104010800080018f80a00f021a40300afac28e05300bb
2023-01-19 13:11:59.680 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xbdd7>, 1, 260, 8, b'\x18\xf8\n\x00\xf0!\xa4\x03', 0, 175, 172, 40, 224, 83, 0, -69]
2023-01-19 13:11:59.680 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0xbdd7>, 1, 260, 8, b'\x18\xf8\n\x00\xf0!\xa4\x03', 0, 175, 172, 40, 224, 83, 0, -69]
2023-01-19 13:11:59.681 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBDD7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=8, data=Serialized[b'\x18\xf8\n\x00\xf0!\xa4\x03'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=172, rssi=-69)
2023-01-19 13:11:59.681 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Received ZCL frame: b'\x18\xf8\n\x00\xf0!\xa4\x03'
2023-01-19 13:11:59.682 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=248, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-01-19 13:11:59.684 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Decoded ZCL frame: F000LevelControlCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0xF000, value=TypeValue(type=uint16_t, value=932))])
2023-01-19 13:11:59.685 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Received command 0x0A (TSN 248): Report_Attributes(attribute_reports=[Attribute(attrid=0xF000, value=TypeValue(type=uint16_t, value=932))])
2023-01-19 13:11:59.686 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Attribute report received: manufacturer_current_level=932
2023-01-19 13:11:59.686 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Getting brightness 932
2023-01-19 13:11:59.687 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xBDD7:1:0x0008]: received attribute: 0 update with value: 237
2023-01-19 13:11:59.688 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0xbdd7>, ep: 1, profile: 0x0104, cluster_id: 0x0008, data: b'18f80a00f021a403'
```

If I'm reading it correctly, that is a device report with the new brightness.
Device is reporting a `932` value and `current_level` is updated with a `237` value.

This kind of sequence repeats with diferent values, so my guess is that the device reports a transition from an initial value to the final one.

This other part seems to be an HA interaction:
```
2023-01-19 13:12:30.971 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Sending Cluster Command. Cluster Command is 4, Arguments are ()
2023-01-19 13:12:30.971 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Setting brightness to 945
2023-01-19 13:12:30.971 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Setting the NO manufacturer id in command: 240
2023-01-19 13:12:30.972 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=253, command_id=240, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2023-01-19 13:12:30.973 DEBUG (MainThread) [zigpy.zcl] [0xBDD7:1:0x0008] Sending request: moveToLevelTuya(payload=TuyaLevelPayload(level=945, transtime=0))
.../...
2023-01-19 13:12:31.001 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBDD7), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=8, data=Serialized[b'\x18\xfd\x0b\xf0\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-63)
[... cut ...]
HA sends a command with TuyaLevel=945, bightness=240
The device gets the SUCCESS ack
Later the device gets a status report with manufacturer_current_level=945 and update the `current_value` to 240

So, from the logs point of view everything is working as expected.
It's true that the last log can be interpreted like the status is not being updated, but if I recall it I believe that is a common behavior for that devices and not a problem in yours.

Have you tried to remove and pair the device again?
Is anything working? HA -> device? device -> HA?

closed 2023-07-25 by github-actions[bot]