zigpy/zha-device-handlers #2120: [Device Support Request] TS130F add settings to frontend

jasimancas, 2023-01-20:
**Is your feature request related to a problem? Please describe.**
In the official TUYA application through the TUYA zigbee gateway, the device has a settings menu with several options. Is it possible to add this menu to the HomeAssistant web interface so that I don't have to modify them through cluster editing?

For example "Motor steering" or "Motor Mode", Both are perhaps two Boolean options.

**Describe the solution you'd like**
I can sniff the zigbee traffic and get the right parameters if needed.
<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_vd43bbfq",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130FTO"
}
```

</details>

javicalle, 2023-01-21:
I can give it a try, but my approach would be to create the entities in HA. 
Are you able to edit the ZHA integration? Through docker image or as a custom component.

jasimancas, 2023-01-21:
The current installation I have is HAOS, but there is no problem to create a development environment for this, for example on a PI 3, I have an unused Conbee II coordinator and so I don't have to alter my production installation.

Does it work for you?

javicalle, 2023-01-21:
Yes, it can do the job.

Probably we would need to update the `zha/switch.py` and maybe also the current quirk.
We need to know the cluster and attributes to expose. The accepted values for each attribute would be also required.

javicalle, 2023-01-21:
Ok, that is my proposal.
These are the necessary parts to expose (ie) the `motor_reversal` attribute from the `TuyaCoveringCluster`.

1. In the quirks part, update the `TuyaCoveringCluster` (`ts130f.py` file) adding (overwriting) the `ep_attribute`:
```python
class TuyaCoveringCluster(CustomCluster, WindowCovering):
    """TuyaSmartCurtainWindowCoveringCluster: Allow to setup Window covering tuya devices."""

    ep_attribute = "tuya_window_covering"

    attributes = WindowCovering.attributes.copy()
    attributes.update({0xF000: ("tuya_moving_state", t.enum8)})
    attributes.update({0xF001: ("calibration", t.enum8)})
    attributes.update({0xF002: ("motor_reversal", t.enum8)})
    attributes.update({0xF003: ("calibration_time", t.uint16_t)})

    .../...
```

2. In the ZHA part, update the `homeassistant/components/zha/select.py` adding two new classes:
```python
class MotorDirection(types.enum8):
    """Tuya motor reversal."""

    Normal = 0x00
    Inverted = 0x01

@CONFIG_DIAGNOSTIC_MATCH(channel_names="tuya_window_covering")
class TuyaCoverMode(ZCLEnumSelectEntity, id_suffix="motor_reversal"):
    """Representation of a ZHA curtain mode configuration entity."""

    _select_attr = "motor_reversal"
    _enum = MotorDirection
    _attr_name = "Motor direction"
```
That would expose the `motor_reversal` attribute in HA as a select entity.
The entity would match the `ep_attribute = "tuya_window_covering"` defined in (1)
The posible values are taken from the enum `MotorDirection`.

3. We need also update the `homeassistant/components/zha/cover.py` (because of the `ep_attribute` overwrite). Add that class just after the existing `ZhaCover`:
```python
@MULTI_MATCH(channel_names="tuya_window_covering")
class ZhaTuyaCover(ZhaCover):
    """Representation of a ZHA cover."""

    def __init__(self, unique_id, zha_device, channels, **kwargs):
        """Init this sensor."""
        super().__init__(unique_id, zha_device, channels, **kwargs)
        self._cover_channel = self.cluster_channels.get("tuya_window_covering")
        self._current_position = None

```

I believe that it is also necessary to initialize the ZHA attribute, but I haven't got that part under control.
For now, I would tell you that if the entity is disabled in HA, try updating the value from the cluster manager and/or restarting HA.

jasimancas, 2023-01-24:
I have installed the test environment, created the folder for the custom_zha_path and edited the changes you request.

If I put ep_attribute = "tuya_window_covering" directly the device does not create the cover entity, in fact it does not expose any entity (except LQI and RSSI).

Another question, to edit the ZHA files, do I have to do it through custom_components?

javicalle, 2023-01-24:
> If I put ep_attribute = "tuya_window_covering" directly the device does not create the cover entity

That was one of the possible side effects. Doesn't generate any error in the HA startup? In theory the point (3) is in charge of this part. 

You can try just removing the `ep_attribute` and modifying this part:
```python
@CONFIG_DIAGNOSTIC_MATCH(
    channel_names="tuya_manufacturer", 
    manufacturers={"_TZ3000_vd43bbfq"}, 
)
```

Check for any error during the HA startup.

jasimancas, 2023-01-24:
`ts130f.py` I leave it unmodified, as it is in the repository.
Modify select.py and cover.py?

javicalle, 2023-01-24:
Yes, leave the `ts130f.py` unmodified (or move it outside the folder).
Modify the `select` with my last comment.
You can keep the `cover` modification, because will not affect.

jasimancas, 2023-01-24:
Forget everything, my fault, it wasn't loading the ZHA custom correctly, I checked the LOG, modified manifest, rebooted and with the instructions in your first post I got this:

[image]

javicalle, 2023-01-24:
And... it's working? 🤞🏻

The `tuya_moving_state` can be implemented the same way. You will need to know the accepted values and its meaning.

jasimancas, 2023-01-24:
If I modify the value and then read the attribute, yes, it updates!

Is it possible for this to come out in a more "readable" way?

[image]

Maybe 0 and 1?

javicalle, 2023-01-24:
> If I modify the value and then read the attribute, yes, it updates!

And the cover behavior, changes in accordance with the value?

> Is it possible for this to come out in a more "readable" way?

Sure, check the `bulb_type` here:
https://github.com/zigpy/zha-device-handlers/blob/8c461640ef578e1dc5efec1c425678fc3897d8d9/zhaquirks/tuya/ts110e.py#L46-L52

But, why? If already in HA you don't have to read it from the cluster manager any more?.

jasimancas, 2023-01-24:
> And the cover behavior, changes in accordance with the value?

I have two identical devices, one I am sniffing the traffic to test changes and the other exactly the same is the one I have in the development environment and yes, it is correct.

> But, why? If already in HA you don't have to read it from the cluster manager any more?.

All right, it would not be necessary, it was simply to "clean up" the value it shows, nothing more.

By the way, sniffing the traffic I have discovered a new attribute that is not in the cluster listing "TuyaCoveringCluster" would it be a good idea to add it?

jasimancas, 2023-01-24:
Ok, these changes we have made, do we have to do PR? Or do you take care of them?

I have more TUYA devices that may be able to bring more things taking advantage of the Gateway for example, if necessary let me know and I'll get on it.

MattWestb, 2023-01-24:
@javicalle I think its better adding TS0130F for matching in ZHA if its possible ?
All this devices is window covering / blinds and i think they need the inverting function for some users.

Hola @jasimancas great work done so today 12 points is going to Spa..... !!!

By the way you can using [ZHA-Toolkit](https://github.com/mdeweerd/zha-toolkit) for scanning devices for all commands and attributes they is having. 
But some devices can behaving crazy after being scanned (some LIDL / tuya devices) and need being deleted and and new joined in ZHA for working normal.

javicalle, 2023-01-24:
> @javicalle I think its better adding TS0130F for matching in ZHA if its possible ?

The problem with that approach is that if there's any `TS0130F` that don't get the quirk (ie because the signature mismatch) can cause errors because the device don't have the 'extra' attributes.
Matching the `tuya_window_covering` would guaranty that the device has the quirk loaded.

> these changes we have made, do we have to do PR? Or do you take care of them?

Please, create the PR with the changes.
Ping me in the HA PR and I will review it.

jasimancas, 2023-01-24:
@MattWestb Thanks for the points! ;-)
This is the attribute info (ZHA_SCAN):

`"0x8000": { "attribute_id": "0x8000","attribute_name": "32768","value_type": ["0x30","enum8","Discrete"],"access":"REPORT|WRITE|READ","access_acl": 7,"attribute_value": 0},`

According to the TUYA app is:
[image]
Value 0x00 = STRONG MOTOR
Value 0x01 = WEAK MOTOR

@javicalle With this option any blind regardless of the model, as long as it is TUYA will load the extras that have been programmed, right?
I have two TS0130F with 2 different submodels/signatures.

If necessary I will open a new issue for the new attribute or an issue to expose in interface at least the values that the app allows to edit if it is ok.

javicalle, 2023-01-24:
Let see if I can explain all the approach.

`zha-device-handlers` implements quirks for _some_ Zigbee devices

Simplifying it a lot, the quirk will be applied if the device signature matches with the quirk one. This implies the model, the manufacturer and the endpoints.
When matched ZHA applies the quirk to the device.
If it does not match, the device is treated as a standard Zigbee device.

Will all the Tuya `TS130F` aply the quirk? We don't know.
Can be Tuya `TS130F` devices (now or in the future) with different endpoints than the detected ones? It can be.

So let's assume that there are Tuya `TS130F` devices that are **quirked** and there are that **are not**.

With the `ep_attribute` overwriting approach, all devices that are **quirked** will generate new entities for the 'extra' attributes. That's correct because only the quirked ones have that extra attributes defined. The standar Zigbee devices will not have the extra attributes.

Will the new entities work for all devices that expose them? We do not know.
But in this case it's something I'm willing to assume because we know the attribute will exist on those devices (although it may not work).
I know that this approach is not to the liking of everyone who prefers that if does not work not generate the entity, but I don't want to check device by device if is supported or not.

Is your question answered?

PS: that explanation is specific to the `TS130F` quirks and because all are using the `TuyaCoveringCluster` which implements the 'extra' attributes. That explanation don't have to be a regular rule for all the quirks/devices.

javicalle, 2023-01-24:
> If necessary I will open a new issue for the new attribute or an issue to expose in interface at least the values that the app allows to edit if it is ok.

It's fine to me if you put everything in just one PR (or 2 if you have to update the quirk).

> This is the attribute info (ZHA_SCAN):
> 
> `"0x8000": { "attribute_id": "0x8000","attribute_name": "32768","value_type": ["0x30","enum8","Discrete"],"access":"REPORT|WRITE|READ","access_acl": 7,"attribute_value": 0},`

To which cluster belongs that attribute? The Tuya one (`0xEF00`)?

MattWestb, 2023-01-24:
@jasimancas Great finding !!!
I think the tuya naming is little strange.
I think its the speed of the motor Low / High speed but you can testing what is doing and implanting it right :-)))

@javicalle we was implanting the tuya TS011f / TS0121 plugs with tuya power on state in ZHA with only model and they all not quirked devices is getting the same and no complains.

javicalle, 2023-01-24:
Well, there are some complains: 😉 
* https://github.com/home-assistant/core/issues/82108
* https://github.com/home-assistant/core/pull/83233#issuecomment-1339895682

MattWestb, 2023-01-24:
You have right ;-(
Good its not braking anything but its dont looks very nice but working.

Is it possible making one handle in the quirk like "tuya_power_state" that ZHA can using for applying the handle for device that have one quirk that need some extra tagged function in ZHA ?

Im thinking on the mess with tuya TRVs that must every being updated in ZHA. If it was possible making one handle like "tuya_TRV1", "tuya_TRV2", "tuya_TRV3", and so on in the quirk and ZHA is using the right fix for the device famely.
Alternative making one unique attribute on one cluster that the quirk is writing then its loading and ZHA can "detecting" the device then its loading the device from the database by the attribute and using the right extras for the device type.

jasimancas, 2023-01-25:
> Let see if I can explain all the approach.
>
> `zha-device-handlers` implements quirks for _some_ Zigbee devices
>
> Simplifying it a lot, the quirk will be applied if the device signature matches with [...]
I have been very clear, thank you very much, I had that doubt, the quirks seem to me quite interesting, so I want to refine the ones I have at least of the devices I have at home to complete and improve them.

From what I have understood, given the edition of the files the only thing it does is to overwrite the attribute that is exposed from the quirk and then in both cover and select the "generic" class of cover is completed with that attribute and in the case of select show the menu if the device has exposed the attribute, right?

In case of not matching the signature the quirk is not applied and the attribute is not overwritten which will expose the default one that I believe is Window_Covering.

[image]

jasimancas, 2023-01-25:
I have uploaded (I think well) a PR with this change in question. I don't frequent git much, correct me if I did something wrong.

If it is OK I do PR on the HA one to update the two files.

MattWestb, 2023-01-25:
Do the PR in HA so can the ZHA maintainer looking on it and link the PR for the quirk updating in it (its normally no problem then javicalle have checking it and have getting all right).

javicalle, 2023-01-25:
> Is it possible making one handle in the quirk like "tuya_power_state" that ZHA can using for applying the handle for device that have one quirk that need some extra tagged function in ZHA ?

Yes, it can be done. 
The only limitation is that you cannot combine several handlers in the same cluster (ie: use the tuya_cover and tuya_backlight handlers at the same time, you would have to have a new tuya_cover_backlight handler with both)

> If I understand correctly, given the edition of the files the only thing it does is to overwrite the attribute that is exposed from the quirk and then in both cover and select the "generic" class of cover is completed with that attribute and in the case of select show the menu if the device has exposed the attribute, right?

There are 2 parts here:
1. the quirk part (the zha-device-handlers library) is in charge to modify or correct the Zigbee interaction of the device. Acts like a translator between the ZHA integration and the device. It can expose 'manufacturer' attributes or define new ones.
2. the ZHA integration that is is charge to expose in HA devices and entities. At first ZHA only 'speak' the Zigbee standar and knows nothing about 'manufacturer' attributes 

All the `ts130f.py` quirks are 'exposing' the 'manufacturer' attributes:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts130f.py#L42-L50

If we don't do it, the ZHA integration does not know how to interact with those attributes.

Now ZHA would know how to speak to these attributes.
And then, we have overwrited the `ep_attribute`. That way all the quirked `TS130F` devices will have a `tuya_window_covering` cluster

All the ZHA is based on some conventions. To present a cover in HA, ZHA would search internally for a cluster with `ep_attribute = "window_covering"`.
As you have found, that is the overwrited value in the ZHA `WindowCovering` cluster:
https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/closures.py#L596-L599

If we don't do nothin else, ZHA will not generate the cover entity for the `TS130F` devices (only the quirked ones) because the `ep_attribute` mismatch.

For this reason, in HA we do 2 things:
1. to extend the `ZhaCover` to match the `channel_names="tuya_window_covering"`. This way HA will render the cover entity 
2. to define  new `ZCLEnumSelectEntity` for **all** the entities with a `tuya_window_covering` cluster. That part will be done despite the cluster have or not that attribute. But we know that will have it because all the quirked have it and only the quirked ones have that `ep_attribute`

More or les what you have said, but I wanted to clarify some of the points expressed

> In case of not matching the signature the quirk is not applied and the attribute is not overwritten which will expose the default one that I believe is WindowCovering.

Correct, ZHA will treat the device like a standar zigbee cover (without any 'extra' attribute).

PS: I use the term 'expose' to define the implementation that made the attribute avaliable in the ZHA layer. That attribute can derivate in an HA entity or not, but you would be able to manage as a cluster attribute from the 'manage cluster' view.

jasimancas, 2023-01-26:
Great explanation, thank you very much, it has been very clear for future modifications.

The PR is done, I continue with the new attribute discovered when I test in production what it is.

closed 2023-08-24 by github-actions[bot]