zigpy/zha-device-handlers #2127: [Device Support Request]

alexvaltchev, 2023-01-23:
**Is your feature request related to a problem? Please describe.**

I purchased this item ZigBee Human Presence Sensor **"TS0601 by _TZE200_v6ossqfy"** and added to my Home Assistance with SkyConnect coordinator without issues. However, when I open it there is no information whatsoever. Does not have any sensor data. Check the image to see.

I use ZHA with SkyConnect coordinator. 

The link to AliExpress: [](https://www.aliexpress.com/item/1005004829286875.html)

I tried it with the normal Smart Life app, and it worked, but not under Home Assistant. Please help? I don't want to use Tuya integration; I want it local. Thank you.

**Describe the solution you'd like**
I want Home Assistant to recognize the Zigbee sensor automatically and to sense presence / occupancy as it works in Tuya and Smart Life app.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_v6ossqfy",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.1.7",
    "dev": false,
    "hassio": true,
[... 334 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.
[zha-b730587b825c0d4887e44af7df635197-_TZE200_v6ossqfy TS0601-790f4895d61d2335b9e2a7b1783801ec.json.txt](https://github.com/zigpy/zha-device-handlers/files/10480642/zha-b730587b825c0d4887e44af7df635197-_TZE200_v6ossqfy.TS0601-790f4895d61d2335b9e2a7b1783801ec.json.txt)

TheJulianJES, 2023-01-23:
This seems to be a duplicate of https://github.com/zigpy/zha-device-handlers/issues/2059.
Please check that issue (a custom quirk might even be available to solve this at the moment).

closed 2023-01-23 by TheJulianJES

TheJulianJES, 2023-01-23:
Please continue discussion in the other issue. There's some information in this comment: https://github.com/zigpy/zha-device-handlers/issues/2059#issuecomment-1400492734

alexvaltchev, 2023-01-23:
Okay, I will, thank you a lot!

baylanger, 2023-12-21:
#2059 was closed with the following comment
`im going to close this off, as I've since abandoned ZHA in favour of Zigbee2mqtt`

I understand someone posted code in 2059 back in Jan '23 but I prefer to wait for official supports from ZHA.

I'm going back to z2m... I'm just not sure if I should keep in place my old Zigbee stick or use Skyconnect w/ z2m. I'm just not sure if SkyConnect works great w/ z2m, any comment on this topic is appreciated.