zigpy/zha-device-handlers #2129: [Device Support Request] TZE200_fwoorn8y

antoweb, 2023-01-24:
**Is your feature request related to a problem? Please describe.**
Is possible to write a quirk for the device i subject? Attached are diagnostic information

[zha-c7d6588cfe70f37b240c21695e499f8b-_TZE200_fwoorn8y TS0601-365222b8b5d4cc9930d83addc54e88a7.json.txt](https://github.com/zigpy/zha-device-handlers/files/10490759/zha-c7d6588cfe70f37b240c21695e499f8b-_TZE200_fwoorn8y.TS0601-365222b8b5d4cc9930d83addc54e88a7.json.txt)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

antoweb, 2023-01-24:
I have used the same quirk of _TZE200_pay2byax changin just line 84 to:
        MODELS_INFO: [("_TZE200_fwoorn8y", "TS0601")],
Attach are the python quirk to rename in py and copy in \config\custom_zha_quirks
[ts0601_TZE200_fwoorn8y_magnet.txt](https://github.com/zigpy/zha-device-handlers/files/10490813/ts0601_TZE200_fwoorn8y_magnet.txt)

antoweb, 2023-01-24:
Yes work but withou illuminance and battery. Work just the open/close door (the most important) i dont know python if tehre is someone that want to edit this script for add illuminance and battery state is welcome :-)

antoweb, 2023-01-24:
Ok work also the battery sensor. Just illuminance is missing

javicalle, 2023-01-24:
The quirk here have the iluminance implemented:
https://github.com/zigpy/zha-device-handlers/issues/1502#issuecomment-1378743535

If not working, enable the debug logs and attach them.

antoweb, 2023-01-24:
Yes thanks but for this model must be modify the line 86

from:
MODELS_INFO: [("_TZE200_pay2byax", "TS0601")],

to:
MODELS_INFO: [("_TZE200_fwoorn8y", "TS0601")],

javicalle, 2023-01-24:
Well, the right way to do it would be to add your device like this:
```python
        MODELS_INFO: [
            ("_TZE200_fwoorn8y", "TS0601"),
            ("_TZE200_pay2byax", "TS0601"),
        ],
```

Your quirk does not implement DP=101 and the suggested one does.
Have you tried it?
Without the debug logs we can't know where is the problem and we can't help you.

antoweb, 2023-01-24:
Yes i have tried and works good..thanks for the suggestion to add the device

javicalle, 2023-01-24:
Great!
Do you mind to create the PR with the working code?

antoweb, 2023-01-24:
Wath os PR?

javicalle, 2023-02-05:
PR states for "Pull Request" the way to propose changes to the library mainstream.
Don't worry if you are not familiar.

Could you check that this version is working and fixes your device:
<details open>
<summary>ts0601_magnet.py</summary>

```python
"""Tuya contact sensor device."""

import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Ota, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_TYPE,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

IASZONE_ZONE_STATUS_ATTR = 0x0002

class ContactSwitchCluster(TuyaAttributesCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaPowerConfigurationCluster(
    TuyaLocalCluster, DoublingPowerConfigurationCluster
):
    """PowerConfiguration cluster for battery-operated tuya devices reporting percentage."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class ContactSwitchManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with ContactSwitch Sensor data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
        ),
        2: DPToAttributeMapping(
            DoublingPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",  # 0x0021
        ),
        101: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            converter=lambda x: 10000 * math.log10(x) + 1,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
    }

class TuyaContactSensor(CustomDevice):
    """Tuya contact sensor device."""

    signature = {
        # NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
        # device_version=1
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026, device_version=1,
        # input_clusters=[0, 1, 1280],
        # output_clusters=[10, 25])
        MODELS_INFO: [("_TZE200_pay2byax", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DoublingPowerConfigurationCluster.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    ContactSwitchCluster,
                    ContactSwitchManufCluster,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

</details>

patrickdass93, 2023-02-19:
How do I add this to HA? I can test to see if it works.

javicalle, 2023-02-19:
> How do I add this to HA? I can test to see if it works.

You would need enable the local quirk folder in your installation. There is a good guide to configure it:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Once configured, create in your `custom_zha_quirks` folder (i.e.) a `ts0601_magnet.py` file with the suggested code.
Save changes and repair your device. Check if the device signature has changed.

patrickdass93, 2023-02-22:
I got this error when restarting HA

`Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_magnet.py:54
First occurred: 12:36:10 PM (1 occurrences)
Last logged: 12:36:10 PM

Error setting up entry socket://192.168.0.219:6638 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 382, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_magnet.py", line 50, in <module>
    class ContactSwitchManufCluster(TuyaMCUCluster):
  File "/config/custom_zha_quirks/ts0601_magnet.py", line 54, in ContactSwitchManufCluster
    1: DPToAttributeMapping(
TypeError: DPToAttributeMapping.__init__() missing 1 required positional argument: 'dp_type'
`

javicalle, 2023-02-22:
Sorry, new version incoming...
I have added all the `dp_type=TuyaDPType.XXXX` in the `DPToAttributeMapping` definitions.

patrickdass93, 2023-02-24:
Hi @javicalle 

There's still an error.
```
Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_magnet.py:57
First occurred: 1:36:47 PM (1 occurrences)
Last logged: 1:36:47 PM

Error setting up entry socket://192.168.0.219:6638 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 382, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_magnet.py", line 50, in <module>
    class ContactSwitchManufCluster(TuyaMCUCluster):
  File "/config/custom_zha_quirks/ts0601_magnet.py", line 57, in ContactSwitchManufCluster
    dp_type=TuyaDPType.ENUM,
NameError: name 'TuyaDPType' is not defined
```

TheJulianJES, 2023-02-24:
Make sure you're importing `TuyaDPType` from zhaquirks.tuya.
For example:
`from zhaquirks.tuya import TuyaLocalCluster, TuyaDPType`

patrickdass93, 2023-02-24:
Thank you @TheJulianJES that worked.

@javicalle Unfortunately, I do not get the sensor to show data
[image]

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Asia/Kuala_Lumpur",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "sonoff": {
      "version": "3.4.0",
      "requirements": [
        "pycryptodome>=3.6.6"
      ]
    },
    "hacs": {
      "version": "1.30.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "wyzeapi": {
      "version": "0.1.17",
      "requirements": [
        "wyzeapy==0.5.16"
      ]
    },
    "ui_lovelace_minimalist": {
      "version": "v1.1.6-hotfix",
      "requirements": [
        "aiofiles==0.8.0",
        "aiogithubapi>=22.2.4"
      ]
    },
    "localtuya": {
      "version": "5.0.0",
      "requirements": []
    },
    "ble_monitor": {
      "version": "11.7.1",
      "requirements": [
        "pycryptodomex>=3.14.1",
        "janus>=1.0.0",
        "aioblescan>=0.2.14",
        "btsocket>=0.2.0",
        "pyric>=0.1.6.3"
      ]
    },
    "tuya_local": {
      "version": "2022.2.3",
      "requirements": [
        "pycryptodome~=3.17",
        "tinytuya==1.10.2"
[... 237 lines cut ...]
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

javicalle, 2023-02-24:
@patrickdass93 Your device signature doesn't match at all with the quirk. Please attach the product or vendor link.

Maybe that (quick and untested) quirk could work for you:
<details>
<summary>ts0601_magnet.py</summary>

```python
"""Tuya contact sensor device."""

import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_TYPE,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class ContactSwitchCluster(TuyaAttributesCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaPowerConfigurationCluster(
    TuyaLocalCluster, DoublingPowerConfigurationCluster
):
    """PowerConfiguration cluster for battery-operated tuya devices reporting percentage."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class ContactSwitchManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with ContactSwitch Sensor data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
        ),
        2: DPToAttributeMapping(
            DoublingPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",  # 0x0021
        ),
        101: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            converter=lambda x: 10000 * math.log10(x) + 1,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
    }

class TuyaContactSensor_var02(CustomDevice):
    """Tuya contact sensor device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_fwoorn8y", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    ContactSwitchManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaPowerConfigurationCluster,
                    ContactSwitchCluster,
                    ContactSwitchManufCluster,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

</details>

patrickdass93, 2023-02-24:
@javicalle is [this](https://rumah-pintar.id/product/arbit-smart-zigbee-door-window-sensor/) what you mean by product link?

javicalle, 2023-02-24:
> @javicalle is [this](https://rumah-pintar.id/product/arbit-smart-zigbee-door-window-sensor/) what you mean by product link?

Yes, that is.
The don't tell anything about device capabilities so no sure if it will report illuminace, battery or anithing else.
Probably (if the device gets the quirk) that new entities are not populated. If so, it will be better to create a new issue and work from it with your logs.

patrickdass93, 2023-02-24:
It should report contact(open/closed) and battery level.

javicalle, 2023-02-24:
Well, it's a matter of see if the device gets the quirk and what DPs is using...
Thes the proposed quirk and enable the debud logs if not working.

antoweb, 2023-03-02:
Hello, after last update HA to 2023.3 this quirks generate error and all Zigbee Integration cannot load. i have to remove from custom_zha_quirks any solution?

closed 2024-06-28 by github-actions[bot]