zigpy/zha-device-handlers #2145: The T1 pro Cube is missing new functions that are available in the device.

SirGoodenough, 2023-01-28:
The T1 pro cube is not functioning properly.  New functions such as the 'Throw' and the 'Hold' are not working. 'Side-up' is also not working. Drop is listed as a function that is no longer available.

See https://github.com/JJPro/CTP-R01-converter

For the Z2M Version of the cube driver for hints on how to make this work properly.
I would but do not have the knowledge to do so.

_Originally posted by @SirGoodenough in https://github.com/zigpy/zha-device-handlers/issues/1975#issuecomment-1407312731_
            
What I said there. 

Since that was a closed issue I didn't think it would be seen.

maxemoose, 2023-01-30:
See my comment on that thread.

SirGoodenough, 2023-02-21:
I don't know enough about this to know how to fix it, but I'm sure the people trying to use the new version of the cube would like to have all the functions available.
jjPro (link above) said they would be willing to provide what info they know, but doesn't know about the data the new cube sends out, so you have a head start.

SirGoodenough, 2023-06-29:
6 months, still looking for an interested party to take this on.

funkytwig, 2023-12-24:
Any news on this.

talormanda, 2023-12-27:
Commenting for exposure. I have this device as well and can't get it to work.

josefheidler, 2024-01-03:
> Commenting for exposure. I have this device as well and can't get it to work.

Same here!

talormanda, 2024-01-03:
> > Commenting for exposure. I have this device as well and can't get it to work.
> 
> Same here!

I found hitting the setup button 5 times in a row changes modes on it to handle the other functions, but nothing more than that...but at least you can flip, slide, twist, and knock now

Conundrum1911, 2024-01-09:
Looking into getting one of these....I use ZHA in Home Assistant, so wanted to confirm I could at least flip it to different sides to do different things, then rotate it to adjust values.

Essentially I'd want 1 side to be "lights" for a room and tap to turn on, tap to turn off, rotate to brighten/dim.

Then flip it to another side and be "blinds" and tap to open, tap to close, or rotate to open more/open less.

talormanda, 2024-01-09:
> Looking into getting one of these....I use ZHA in Home Assistant, so wanted to confirm I could at least flip it to different sides to do different things, then rotate it to adjust values. [...]

Add it to HA then hit the setup button 5 times to change modes to Action mode. After that, all the commands will work. I'm using ZHA with skyconnect and it functions.

Conundrum1911, 2024-01-09:
> Add it to HA then hit the setup button 5 times to change modes to Action mode. After that, all the commands will work. I'm using ZHA with skyconnect and it functions.

Checking before buying, but if the basic setup I want is possible right now, I'll pull the trigger and buy one.

Also what is the best blueprint to use?  The older magic cube ones or is there a better one?

talormanda, 2024-01-09:
> Checking before buying, but if the basic setup I want is possible right now, I'll pull the trigger and buy one.
>
> Also what is the best blueprint to use? The older magic cube ones or is there a [...]

i wasnt using any blueprint with mine. sorry to say i didnt even look into that. what do blueprints do? lol sorry

SirGoodenough, 2024-04-26:
Foe ZHA the blueprints for the original cube is the best you get, without a complete option set-up requested here.
There is a complete Z2M interview and I have blueprints for connecting to the new cube using that.
https://github.com/SirGoodenough/HA_Blueprints

skypyxis, 2024-07-16:
Is this correct? 
The original cube is fully supported but the new features on the T1 PRO are not yet supported by ZHA.

Does this mean there is no advantage to using the new cube with HA ZHA?

Barrerse, 2024-10-14:
> Commenting for exposure. I have this device as well and can't get it to work.

Seems like there is still no upate on this.

Raboo, 2024-10-14:
Didn't Aqara recently announce that they will start supporting home assistant on some of their products. Perhaps if we all send them an email asking them to add support for T1 Pro Cube in Home Assistant?

vladikoff, 2024-11-29:
For those subscribed to this issue, just want to confirm another case where updating to a latest firmware of the Zigbee USB stick (SONOFF in this case) fixed compatibility with the cube almost immediately