zigpy/zha-device-handlers #2148: [Device Support Request] DOUBLE_PRESS not supported for lumi.ctrl_ln1.aq1

maurobaluda, 2023-01-29:
**Is your feature request related to a problem? Please describe.**
DOUBLE_PRESS not supported for lumi.ctrl_ln1.aq1

**Describe the solution you'd like**
The feature works in in the 2-button variant lumi.ctrl_ln2.aq1

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0010"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0006",
        "0x0010"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0010",
        "0x0012"
      ],
      "out_clusters": []
    },
    "6": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0010",
        "0x0012"
      ],
      "out_clusters": []
    },
    "7": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0010",
        "0x0012"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.ctrl_ln2.aq1",
  "class": "zhaquirks.xiaomi.aqara.ctrl_ln.CtrlLn"
}
```

</details>

TheJulianJES, 2023-02-04:
It's possible that this signature just needs to be added in: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/ctrl_ln.py

mbaluda-ms, 2023-05-06:
Any progress on this issue?
I have some automation relying on the feature...
thanks

mbaluda-ms, 2023-07-24:
@TheJulianJES can you detail a bit more how the signature should be added?
`lumi.ctrl_ln2.aq1` is already listed

maurobaluda, 2024-06-21:
The problem is still there. The double click used to work but stopped being supported over a year ago.

closed 2024-12-25 by github-actions[bot]