zigpy/zha-device-handlers #2149: [Device Support Request] Brennstuhl Connect (TS0601) add attribute "local_temperatur_calibration" (0x0010)

DLieb, 2023-01-29:
**Is your feature request related to a problem? Please describe.**
The quirk is missing the attribute for local temperature calibration

**Describe the solution you'd like**
Add attribute to cluster and frontend

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_2cs6g9i7",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv.SiterwellGS361_Type2"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.1.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    },
    "better_thermostat": {
[... 377 lines cut ...]
                "attribute_name": "attr_reporting_status"
              }
            }
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2023-01-29:
Your (and my) device dont supporting local temperature calibration.
The quirk in ZHA is only basic support for the device but you can installing one extended function that is implanting all function we can getting from the device plus one function calibrating the heating / idle in ZHA GUI (is not 100% the same as local temperature calibration but fixing the heating offset).

https://github.com/jacekk015/zha_quirks

DLieb, 2023-01-31:
Tanks for the reply, I will check that out.
I thought the device is supporting the calibration because I tested iobroker before switching to homeassistant and there the device (next to a few more other attributes) had an attribute with that name, didn't test it though.