zigpy/zha-device-handlers #2151: [Device Support Request] TS011F _TZ3000_gjnozsaz

zsawyer, 2023-01-29:
**Is your feature request related to a problem? Please describe.**
I am using a RaspberryPi with dedicated HomeAssistant and connected Zigbee USB Dongle. 
I have connected 4 of these plugs TS011F _TZ3000_gjnozsaz (single power plug with switch and monitoring capability)

However I do get a lot of warnings for "Unknown cluster command" on the network (from all 4 devices). 
I am also seeing that some other devices are not working as expected (thermostats sometimes do not switch to the target temperature but I am not sure if that is related - I do not get these unknown command logs at that specific time). 

The plugs themselves work fine. The switch functionality is responsive and the current/power/voltage monitoring does not seem interrupted. The Back light mode switching also works. 

**Describe the solution you'd like**
Proper handling of the Unknown cluster commands. 

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_gjnozsaz",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.1.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.76-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "embedded",
[... 608 lines cut ...]
2023-01-23 00:52:14.613 WARNING (MainThread) [zigpy.zcl] [0xD400:1:0xe001] Unknown cluster command 225 b'\nt\x88`+\x8f\x04\x00\x00\x7f\x04\x00\x00p\x04\x00\x00W\x04\x00\x00?\x04\x00\x00%\x04\x00\x00\x13\x04\x00\x00\x02\x04\x00\x00\xf0\x03\x00\x00\xdb\x03\x00\x00\x01'
2023-01-23 00:52:14.814 WARNING (MainThread) [zigpy.zcl] [0xD400:1:0xe001] Unknown cluster command 225 b'\nt\x88`+\x8f\x04\x00\x00\x7f\x04\x00\x00p\x04\x00\x00W\x04\x00\x00?\x04\x00\x00%\x04\x00\x00\x13\x04\x00\x00\x02\x04\x00\x00\xf0\x03\x00\x00\xdb\x03\x00\x00\x01'
2023-01-23 00:52:15.014 WARNING (MainThread) [zigpy.zcl] [0xD400:1:0xe001] Unknown cluster command 225 b'\nt\x88`+\x8f\x04\x00\x00\x7f\x04\x00\x00p\x04\x00\x00W\x04\x00\x00?\x04\x00\x00%\x04\x00\x00\x13\x04\x00\x00\x02\x04\x00\x00\xf0\x03\x00\x00\xdb\x03\x00\x00\x01'
2023-01-23 00:52:15.215 WARNING (MainThread) [zigpy.zcl] [0xD400:1:0xe001] Unknown cluster command 225 b'\nt\x88`+\x8f\x04\x00\x00\x7f\x04\x00\x00p\x04\x00\x00W\x04\x00\x00?\x04\x00\x00%\x04\x00\x00\x13\x04\x00\x00\x02\x04\x00\x00\xf0\x03\x00\x00\xdb\x03\x00\x00\x01'
2023-01-23 00:52:15.416 WARNING (MainThread) [zigpy.zcl] [0xD400:1:0xe001] Unknown cluster command 225 b'\nt\x88`+\x8f\x04\x00\x00\x7f\x04\x00\x00p\x04\x00\x00W\x04\x00\x00?\x04\x00\x00%\x04\x00\x00\x13\x04\x00\x00\x02\x04\x00\x00\xf0\x03\x00\x00\xdb\x03\x00\x00\x01'
2023-01-23 00:52:15.615 WARNING (MainThread) [zigpy.zcl] [0xD400:1:0xe001] Unknown cluster command 225 b'\nt\x88`+\x8f\x04\x00\x00\x7f\x04\x00\x00p\x04\x00\x00W\x04\x00\x00?\x04\x00\x00%\x04\x00\x00\x13\x04\x00\x00\x02\x04\x00\x00\xf0\x03\x00\x00\xdb\x03\x00\x00\x01'
2023-01-23 00:52:15.815 WARNING (MainThread) [zigpy.zcl] [0xD400:1:0xe001] Unknown cluster command 225 b'\nt\x88`+\x8f\x04\x00\x00\x7f\x04\x00\x00p\x04\x00\x00W\x04\x00\x00?\x04\x00\x00%\x04\x00\x00\x13\x04\x00\x00\x02\x04\x00\x00\xf0\x03\x00\x00\xdb\x03\x00\x00\x01'
2023-01-23 00:52:16.018 WARNING (MainThread) [zigpy.zcl] [0xD400:1:0xe001] Unknown cluster command 225 b'\nt\x88`+\x8f\x04\x00\x00\x7f\x04\x00\x00p\x04\x00\x00W\x04\x00\x00?\x04\x00\x00%\x04\x00\x00\x13\x04\x00\x00\x02\x04\x00\x00\xf0\x03\x00\x00\xdb\x03\x00\x00\x01'
2023-01-23 00:52:16.218 WARNING (MainThread) [zigpy.zcl] [0xD400:1:0xe001] Unknown cluster command 225 b'\nt\x88`+\x8f\x04\x00\x00\x7f\x04\x00\x00p\x04\x00\x00W\x04\x00\x00?\x04\x00\x00%\x04\x00\x00\x13\x04\x00\x00\x02\x04\x00\x00\xf0\x03\x00\x00\xdb\x03\x00\x00\x01'
2023-01-23 05:15:21.244 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.stromschalter_wz_eulenlampe_active_power (<class 'homeassistant.components.zha.sensor.ElectricalMeasurement'>) took 0.545 seconds. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+zha%22
```

</details>

**Additional context**
Note that it seemed that the log spam happend at around the same time of day on two days in a row. Very late at night. 

Also note that "device type" 0x010a matches with https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py#L910 which seems wrong. 
```
class Plug_4AC_2USB_Metering(EnchantedDevice):
    """Tuya Tuya 4 outlet + 2 USB with power metering."""
```
However, these plugs are not a 4 outlet, they are single plugs with switch and power metering, no usbs but back light mode and last state support. I am not sure if this is of relevance. 

This issue is probably related: #1992

MattWestb, 2023-01-30:
Your device is one hybrid that is using normal Zigbee commands and looks working OK but its also have one tuya MCU cluster `0xe001` that is normally not used.
We is not knowing what commands the MCU is using and no user or devs have finding it and we is not knowing what the devices is doing.
Its reported for some other hybrid devices is sending MCU commands but its looks being harmful and is not making any bad more then spamming the log.

I think your device is supporting advance schedule that the MCU is doing and its also trying getting the time from the system but not with zigbee command its trying doing it with MCU commands (therefor all your device is doing at around the same time after you installed them).
For the moment we cant do so much but if getting information of the functions the device is using that is missing and deep digging its possible finding the commands and implanting them.

MattWestb, 2023-01-30:
Z2M was adding your device Dec 20, 2021 https://github.com/Koenkk/zigbee-herdsman-converters/commit/5cf941df210dea204ee190bc01a794f0f8fb9d46.
But they have restructure the tuya devices so i cant flowing where is today.

Z2M is having 3 types of TS011F plugs:
```
        zigbeeModel: ['TS011F'],
        model: 'TS011F_plug_1',
        description: 'Smart plug (with power monitoring)',
        vendor: 'TuYa',

        fingerprint: tuya.fingerprint('TS011F',
            ['_TZ3000_hyfvrar3', '_TZ3000_v1pdxuqq', '_TZ3000_8a833yls', '_TZ3000_bfn1w0mm', '_TZ3000_nzkqcvvs']),
        model: 'TS011F_plug_2',
        description: 'Smart plug (without power monitoring)',
        vendor: 'TuYa',

        fingerprint: [160, 100, 69, 68, 65, 64, 66].map((applicationVersion) => {
            return {modelID: 'TS011F', applicationVersion, priority: -1};
        }),
        model: 'TS011F_plug_3',
        description: 'Smart plug (with power monitoring by polling)',
        vendor: 'TuYa',

const TS011Fplugs = ['_TZ3000_5f43h46b', '_TZ3000_cphmq0q7', '_TZ3000_dpo1ysak', '_TZ3000_ew3ldmgx', '_TZ3000_gjnozsaz',
    '_TZ3000_jvzvulen', '_TZ3000_mraovvmm', '_TZ3000_nfnmi125', '_TZ3000_ps3dmato', '_TZ3000_w0qqde0g', '_TZ3000_u5u4cakc'];
```
Your is one type 1 normal with power monitoring or one type 3 that need being pulled for getting the readings from it.
Then looking what functions they is using is not signs of tuya MCU commands = not known to the comunity.

closed 2023-08-05 by github-actions[bot]