zigpy/zha-device-handlers #2153: [Device Support Request] TS0601 by _TZE200_whkgqxse Tuya Temp/Humidity sensor with LCD clock

Pukington, 2023-01-30:
Problem: This device connects to ZHA but doesn't supply temp/humidity/battery or anything etc.

<details>
<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_whkgqxse",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**

Here is the device in question: https://www.aliexpress.com/item/1005003980647546.html, 

it apparenlty works in Z2MQTT:

https://github.com/Koenkk/zigbee-herdsman-converters/commit/8b5b6d50bb436c3664c87b12a44c8075dbfb7586

Here's what HA looks like currently:

[image]

b2un0, 2023-01-30:
duplicate of #1771 1771

Pukington, 2023-02-01:
Thanks, the latest quirk worked to provide Temp/Humidity/Battery.  I'm not overly fussed by syncing time.  I did search but somehow didn't find that thread.  Thanks again.

closed 2023-02-01 by Pukington