zigpy/zha-device-handlers #2154: [Device Support Request] Neo Coolcam NAS-PD07 quirk seems to no longer work fully correctly.

PedanticAvenger, 2023-01-30:
**Is your feature request related to a problem? Please describe.**
Current install of HA on Yellow platform.  PAiring new NAS-PD07 devices results in discovery, entities for motion/temp/humidity, but battery and temp do not show up as entities nor do they show as values in any cluster.

**Describe the solution you'd like**
Quirk tweak, or guidance, to restore these missing entities.

<details>

<summary>Device signature</summary>

```{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0x0406",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_7hfcudw5",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_motion.NeoMotion"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.1.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Edmonton",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.30.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "alarmo": {
[... 334 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2023-02-05:
Device don't have a `PowerConfiguration` cluster so no battery info available.
Without any logs or adicional info I would suggest you to repair the device again keeping it awake pushing any button during all the pairing proccess.

PedanticAvenger, 2023-02-06:
I'll pair up another (I have one in box) and include the pairing log, as well as diagnostics, etc. this time.
Thanks!

PedanticAvenger, 2023-02-06:
The quirk has a NeoBatteryLevel defined (I think) for this device so there SHOULD be something.

<details>
<summary>Pairing Logs</summary>

```
[0x3246](SmokeSensor-EF-3.0): Device seen - marking the device available and resetting counter
[0x3246](SmokeSensor-EF-3.0): Update device availability -  device available: True - new availability: True - changed: False
[0x242C](TS0601): Device seen - marking the device available and resetting counter
[0x242C](TS0601): Update device availability -  device available: True - new availability: True - changed: False
New device 0x62e1 (b4:e3:f9:ff:fe:02:e3:0c) joined the network
[0x62e1] Scheduling initialization
Tries remaining: 3
[0x62e1] Requesting 'Node Descriptor'
Tries remaining: 2
[0x62e1] Extending timeout for 0x57 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x62E1), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=8, profile_id=0, cluster_id=19, data=Serialized[b'\x81\xe1b\x0c\xe3\x02\xfe\xff\xf9\xe3\xb4\x80'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=168, rssi=-58)
Device 0x62e1 (b4:e3:f9:ff:fe:02:e3:0c) joined the network
[0x62e1] Scheduling initialization
[0x62e1] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x62E1 ieee=b4:e3:f9:ff:fe:02:e3:0c is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x81\xe1b\x0c\xe3\x02\xfe\xff\xf9\xe3\xb4\x80'
[0x62e1:zdo] ZDO request ZDOCmd.Device_annce: [0x62E1, b4:e3:f9:ff:fe:02:e3:0c, 128]
Tries remaining: 3
[0x62e1] Requesting 'Node Descriptor'
Tries remaining: 2
[0x62e1] Extending timeout for 0x59 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x62E1), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=9, profile_id=0, cluster_id=32770, data=Serialized[b'W\x00\xe1b\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=172, rssi=-57)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x62E1 ieee=b4:e3:f9:ff:fe:02:e3:0c is_initialized=False> from ep 0 to ep 0, cluster 32770: b'W\x00\xe1b\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0x62e1:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0x62E1, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
[0x62e1:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_rsp([<Status.SUCCESS: 0>, 0x62E1, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)])
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x62E1), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=10, profile_id=0, cluster_id=32770, data=Serialized[b'Y\x00\xe1b\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=172, rssi=-57)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x62E1 ieee=b4:e3:f9:ff:fe:02:e3:0c is_initialized=False> from ep 0 to ep 0, cluster 32770: b'Y\x00\xe1b\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0x62e1] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x62e1] Discovering endpoints
Tries remaining: 3
[0x62e1] Extending timeout for 0x5b request
[... cut ...]
[... 519 lines cut ...]
              "0x0000": {
                "attribute_name": "measured_value"
              }
            }
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

```
</details>

closed 2023-08-12 by github-actions[bot]