zigpy/zha-device-handlers #2158: Neo Siren w/ Temperature and Humidity (model NAS-AB02B0) apparently not working in HA anymore.

PedanticAvenger, 2023-01-31:
**Is your feature request related to a problem? Please describe.**
Attempts to join NAS-AB02B0 siren to the network registers the device but it stays in flashing "pairing mode".  Entity "switch" present, does nothing. Diagnostic "Identify" present, does nothing.  No Temp/Humidity sensors discovered, unable to configure/control siren.

**Describe the solution you'd like**
Update to quirk and/or instructions on how to address.

<details>

<summary>Device signature</summary>

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_d0yu2xgi",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_siren.TuyaSiren2"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.1.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Edmonton",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.30.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
[... 281 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.
Feels related to #2154 but that is just a gut feeling.

javicalle, 2023-02-05:
The title says "not working in HA anymore". It was working before?
Please update to latest version (zha-quirks==0.0.92) and check again.
Remove the device from HA wait a couple of minutes and pair it again.

If not fixed, attach the debug logs from your device.

PedanticAvenger, 2023-02-06:
Not working anymore meaning "Several people in the forums say it works in their install, I setup my new yellow installation, discovered these sensors and they are not working properly/fully."  The sensors do not appear to be different PCB revisions, etc. but could certainly be running a different firmware.

I've been running the latest version of HA, current at posting, via HAOS on yellow updated to 2023.2.2 this morning.  I must assume that means the quirks are current as well given the bundled nature of this package.

Did all the setup again.

<details>
<summary>Pairing Log</summary>

```

Device 0x1b21 (b4:e3:f9:ff:fe:02:e0:2d) joined the network
[0x1b21] Scheduling initialization
Tries remaining: 3
[0x1b21] Requesting 'Node Descriptor'
Tries remaining: 2
[0x1b21] Extending timeout for 0x3c request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=62, profile_id=0, cluster_id=19, data=Serialized[b'\x81!\x1b-\xe0\x02\xfe\xff\xf9\xe3\xb4\x80'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=188, rssi=-53)
Device 0x1b21 (b4:e3:f9:ff:fe:02:e0:2d) joined the network
[0x1b21] Scheduling initialization
[0x1b21] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x1B21 ieee=b4:e3:f9:ff:fe:02:e0:2d is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x81!\x1b-\xe0\x02\xfe\xff\xf9\xe3\xb4\x80'
[0x1b21:zdo] ZDO request ZDOCmd.Device_annce: [0x1B21, b4:e3:f9:ff:fe:02:e0:2d, 128]
Tries remaining: 3
[0x1b21] Requesting 'Node Descriptor'
Tries remaining: 2
[0x1b21] Extending timeout for 0x3e request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=63, profile_id=0, cluster_id=32770, data=Serialized[b'<\x00!\x1b\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=200, rssi=-50)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x1B21 ieee=b4:e3:f9:ff:fe:02:e0:2d is_initialized=False> from ep 0 to ep 0, cluster 32770: b'<\x00!\x1b\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'
[0x1b21:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0x1B21, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
[0x1b21:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_rsp([<Status.SUCCESS: 0>, 0x1B21, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)])
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=64, profile_id=260, cluster_id=10, data=Serialized[b'\x00\x00\x00\x07\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=176, rssi=-56)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x1B21 ieee=b4:e3:f9:ff:fe:02:e0:2d is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00\x00\x00\x07\x00'
[0x1b21] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=0, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False), payload: b'\x07\x00'
[0x1b21] Uninitialized device command '0' params: Read_Attributes(attribute_ids=[7])
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=65, profile_id=0, cluster_id=32770, data=Serialized[b'>\x00!\x1b\x02@\x80\x02\x10RR\x00\x00,R\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radi [... cut ...]
              "0x4003": {
                "attribute_name": "start_up_on_off"
              }
            }
          }
        },
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```
</details>

javicalle, 2023-02-06:
There are some errors in the logs:
```
[0x1B21:1:0x0004] Received command 0x02 (TSN 70): get_membership_response(capacity=255, groups=[])
[0x1B21:1:0x0004] No explicit handler for cluster command 0x02: get_membership_response(capacity=255, groups=[])

[0x1b21:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x1b21:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_req([0x0000])

[0x1B21:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=5, command_id=36, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x1B21:1:0xef00] Unknown cluster command 36 b'\x00\x06'
[0x1B21:1:0xef00] Received command 0x24 (TSN 5): b'\x00\x06'
[0x1B21:1:0xef00] No explicit handler for cluster command 0x24: b'\x00\x06'

[0x1B21:1:0xef00] Unknown cluster command 17 b'\x00\x07D'
[0x1B21:1:0xef00] Received command 0x11 (TSN 6): b'\x00\x07D'
[0x1B21:1:0xef00] No explicit handler for cluster command 0x11: b'\x00\x07D'
```

Have you tried to keep the device awake during the pairing proccess? If not, just pair it again pressing any device button every second or so until the pairing proccess end.
If not working attach the new pairing logs.

PedanticAvenger, 2023-02-06:
On it.....
Note, that the HA new device screen never pops up as pairing done but the device appears in the paired device list.

<details>
<summary>Re-pair data with ongoing button presses.</summary>
```
Device 0x1b21 (b4:e3:f9:ff:fe:02:e0:2d) joined the network
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=150, profile_id=0, cluster_id=19, data=Serialized[b'\x82!\x1b-\xe0\x02\xfe\xff\xf9\xe3\xb4\x80'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=184, rssi=-54)
Device 0x1b21 (b4:e3:f9:ff:fe:02:e0:2d) joined the network
[0x1B21](TS0601): Update device availability -  device available: False - new availability: True - changed: True
[0x1B21](TS0601): Device availability changed and device became available, reinitializing channels
[0x1b21:zdo] ZDO request ZDOCmd.Device_annce: [0x1B21, b4:e3:f9:ff:fe:02:e0:2d, 128]
[0x1B21](TS0601): started initialization
[0x1B21:ZDO](TS0601): 'async_initialize' stage succeeded
[0x1B21:1:0x0402]: initializing channel: from_cache: False
[0x1B21:1:0x0402]: initializing uncached channel attributes: ['measured_value'] - from cache[False]
[0x1B21:1:0x0402]: Reading attributes in chunks: ['measured_value']
[0x1B21:1:0x0402]: finished channel initialization
[0x1B21:1:0x0003]: initializing channel: from_cache: False
[0x1B21:1:0x0003]: finished channel initialization
[0x1B21:1:0x0405]: initializing channel: from_cache: False
[0x1B21:1:0x0405]: initializing uncached channel attributes: ['measured_value'] - from cache[False]
[0x1B21:1:0x0405]: Reading attributes in chunks: ['measured_value']
[0x1B21:1:0x0405]: finished channel initialization
[0x1B21:1:0x0000]: initializing channel: from_cache: False
[0x1B21:1:0x0000]: finished channel initialization
[0x1B21:1:0xef00]: initializing channel: from_cache: False
[0x1B21:1:0xef00]: finished channel initialization
[0x1B21:1:0x0006]: initializing channel: from_cache: False
[0x1B21:1:0x0006]: initializing cached channel attributes: ['start_up_on_off']
[0x1B21:1:0x0006]: Reading attributes in chunks: ['start_up_on_off']
[0x1B21:1:0x0006]: initializing uncached channel attributes: ['on_off'] - from cache[False]
[0x1B21:1:0x0006]: Reading attributes in chunks: ['on_off']
[0x1B21:1:0x0006]: finished channel initialization
[0x1B21:1:0x0019]: initializing channel: from_cache: False
[0x1B21:1:0x0019]: finished channel initialization
[0x1B21:1:0x0402]: 'async_initialize' stage succeeded
[0x1B21:1:0x0003]: 'async_initialize' stage succeeded
[0x1B21:1:0x0405]: 'async_initialize' stage succeeded
[0x1B21:1:0x0000]: 'async_initialize' stage succeeded
[0x1B21:1:0xef00]: 'async_initialize' stage succeeded
[0x1B21:1:0x0006]: 'async_initialize' stage succeeded
[0x1B21:1:0x0019]: 'async_initialize' stage succeeded
[0x1B21](TS0601): power source: Battery or Unknown
[0x1B21](TS0601): completed initialization
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=151, profile_id=0, cluster_id=2, data=Serialized[b'\x83\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=192, rssi=-52)
[0x1b21:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x1b21:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_req([0x0000])
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=154, profile_id=260, cluster_id=0, data=Serialized[b'\x08\x01\n\x01\x00 U\xe2\xff \x1f\xe4\xff \x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=188, rssi=-53)
[0x1B21:1:0x0000] Received ZCL frame: b'\x08\x01\n\x01\x00 U\xe2\xff \x1f\xe4\xff \x01'
[0x1B21:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=1, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x1B21:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=85)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=31)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=1))])
[0x1B21:1:0x0000] Received command 0x0A (TSN 1): Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=85)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=31)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=1))])
[0x1B21:1:0x0000] Attribute report received: app_version=85, 0xFFE2=31, 0xFFE4=1
[0x1B21:1:0x0000] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=1, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_ [... cut ...]
[0x62E1](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0x1B21](TS0601): Device seen - marking the device available and resetting counter
[0x1B21](TS0601): Update device availability -  device available: True - new availability: True - changed: False
[0xCAAD](TS0601): Device seen - marking the device available and resetting counter
[0xCAAD](TS0601): Update device availability -  device available: True - new availability: True - changed: False
```

PedanticAvenger, 2023-02-06:
Right, trying this again.
Mashed the hell out of the button while pairing.  Still no temp/humi sensors appearing and no actions TO the device work.

<details>
<summary>Pair while button presses </summary>
```
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=15, profile_id=260, cluster_id=10, data=Serialized[b'\x00\x00\x00\x07\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=160, rssi=-60)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x1B21 ieee=b4:e3:f9:ff:fe:02:e0:2d is_initialized=False> from ep 1 to ep 1, cluster 10: b'\x00\x00\x00\x07\x00'
[0x1b21] Received ZCL while uninitialized on endpoint id 1, cluster 0x000a id, hdr: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=0, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False), payload: b'\x07\x00'
[0x1b21] Uninitialized device command '0' params: Read_Attributes(attribute_ids=[7])
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=16, profile_id=0, cluster_id=32773, data=Serialized[b'c\x00!\x1b\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=164, rssi=-59)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x1B21 ieee=b4:e3:f9:ff:fe:02:e0:2d is_initialized=False> from ep 0 to ep 0, cluster 32773: b'c\x00!\x1b\x01\x01'
[0x1b21] Discovered endpoints: [1]
[0x1b21] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x1b21:1] Discovering endpoint information
Tries remaining: 3
[0x1b21] Extending timeout for 0x67 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=17, profile_id=0, cluster_id=32772, data=Serialized[b'g\x00!\x1b\x14\x01\x04\x01Q\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xef\x02\x19\x00\n\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=168, rssi=-58)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x1B21 ieee=b4:e3:f9:ff:fe:02:e0:2d is_initialized=False> from ep 0 to ep 0, cluster 32772: b'g\x00!\x1b\x14\x01\x04\x01Q\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xef\x02\x19\x00\n\x00'
[0x1b21:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 0xEF00
[0x1B21:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=105, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0x1B21:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0x1b21] Extending timeout for 0x69 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1B21), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=18, profile_id=260, cluster_id=0, data=Serialized[b'\x18i\x01\x04\x00\x00B\x10_TZE200_d0yu2xgi\x05\x00\x00B\x06TS0601'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=164, rssi=-59)
[0x1B21:1:0x0000] Received ZCL frame: b'\x18i\x01\x04\x00\x00B\x10_TZE200_d0yu2xgi\x05\x00\x00B\x06TS0601'
[0x1B21:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=105, command_id=1, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x1B21:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE200_d0yu2xgi')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
[0x1b21] Read model 'TS0601' and manufacturer '_TZE200_d0yu2xgi' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x1b21] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_d0yu2xgi' nwk=0x1B21 ieee=b4:e3:f9:ff:fe:02:e0:2d is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_d0yu2xgi' nwk=0x1B21 ieee=b4:e3:f9:ff:fe:02:e0:2d is_initialized=True>
Checking quirks for _TZE200_d0yu2xgi TS0601 (b4:e3:f9:ff:fe:02:e0:2d)
Considering <class 'zhaquirks.tuya.ts0601_siren.TuyaSiren2'>
[... cut ...]
[0x1B21:1:0xef00] Decoded ZCL frame: TuyaManufClusterSiren:mcu_version_rsp(param=MCUVersionRsp(tsn=1792, version=68))
[0x1B21:1:0xef00] Received command 0x11 (TSN 4): mcu_version_rsp(param=MCUVersionRsp(tsn=1792, version=68))
[0x1B21:1:0xef00] No explicit handler for cluster command 0x11: mcu_version_rsp(param=MCUVersionRsp(tsn=1792, version=68))
[0xCAAD](TS0601): Device seen - marking the device available and resetting counter
[0xCAAD](TS0601): Update device availability -  device available: True - new availability: True - changed: False
```
</details>

closed 2023-08-13 by github-actions[bot]