zigpy/zha-device-handlers #2159: [Device Support Request]LoraTap 2Button Remote (TS0042 _TZ3000_fkvaniuu)

DenZyHa, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
TS0042 _TZ3000_fkvaniuu
The button press event cannot be caught by ZHA.
The 4-button type remote control of the same product is working.(TS0044)
HA is built from a Docker image. 

**Describe the solution you'd like**
Device support or guidance, to restore these missing entities.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0xe000"
      ],
      "out_clusters": [
        "0x0006",
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [],
      "out_clusters": [
        "0x0006"
      ]
    }
  },
  "manufacturer": "_TZ3000_fkvaniuu",
  "model": "TS0042",
  "class": "zhaquirks.tuya.ts0042.TuyaSmartRemote0042TOPlusA"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.1.7",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Asia/Tokyo",
    "os_name": "Linux",
    "os_version": "5.13.x",
    "run_as_root": true
  },
  "custom_components": {
    "localtuya": {
[... 635 lines cut ...]
[0x1B38:1:0x0006]: finished channel initialization
[0x1B38:1:0x0019]: initializing channel: from_cache: False
[0x1B38:1:0x0019]: finished channel initialization
[0x1B38:2:0x0006]: initializing channel: from_cache: False
[0x1B38:2:0x0006]: finished channel initialization
[0x1B38:2:0x0006]: 'async_initialize' stage succeeded
[0x1B38:1:0x0001]: failed to get attributes '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>
[0x1B38:1:0x0001]: async_initialize: retryable request #1 failed: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>. Retrying in 0.9s
[0x1B38:1:0x0001]: initializing channel: from_cache: False
[0x1B38:1:0x0001]: initializing uncached channel attributes: ['battery_voltage', 'battery_percentage_remaining'] - from cache[False]
[0x1B38:1:0x0001]: Reading attributes in chunks: ['battery_voltage', 'battery_percentage_remaining']
[0x1B38:1:0x0001] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=55, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0x1B38:1:0x0001] Sending request: Read_Attributes(attribute_ids=[32, 33])
[0x1b38] Extending timeout for 0x37 request
```

</details>

**Additional context**
Purchased from the link below

[https://www.aliexpress.com/item/1005003432430621.html](https://www.aliexpress.com/item/1005003432430621.html)
Color: 2 Button Remote

[image]

MattWestb, 2023-02-01:
Debug logs please then pressing button one and two.
Its looks like the system is loading the quirks OK and its working for other versions of this device but its tuya . . .

DenZyHa, 2023-02-01:
@MattWestb
Thank you for your reply.
I added the following to "configuration.yaml", rebooted, and pressed buttons 1 and 2, 
but the button presses were not logged in "home-assistant.log" .
(Other debug logs are output.)

<details>
<summary>configuration.yaml</summary>

```yaml
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```

</details>

DenZyHa, 2023-02-01:
I also attach the full log from HA startup just in case.
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/10556754/home-assistant.log)

MattWestb, 2023-02-01:
Looks like it dont like have the coordinator as its parent of some unknown reason.
```
2023-02-01 20:41:07.604 INFO (MainThread) [zigpy.application] Device 0xd3e1 (a4:c1:38:44:b8:20:be:1d) left the network
2023-02-01 20:41:07.604 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD3E1](TS0042): Update device availability -  device available: True - new availability: False - changed: True
2023-02-01 20:41:07.604 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD3E1](TS0042): Device availability changed and device became unavailable
```
And after that is not in the network and cant sending your button pressing.
I think you need one good router device so it can having as its parent and not getting problem then the coordinator is doing some other things (its always better have end device connected to routers for not getting problems).

Also what hardware is your coordinator ?
I can see its running `EmberZNet version: 6.7.8.0 build 373` that is normally the stable EZSP. But if its one Sonoff its needs 6.7.9.0 for fixing wrong tuned radio that can getting problems with sleeping end devices.

DenZyHa, 2023-02-02:
@MattWestb 
The coordinator hardware is TYGWZ-01.
https://zigbee.blakadder.com/Tuya_TYGWZ-01.html

You pointed out that there may be a problem with the coordinator, 
so I deleted the ZHA integration and recreated it. 
The 2ButtonRemote(TS0042 _TZ3000_fkvaniuu) is now working!!!

Thank you very much!!!
That was a really big help.

MattWestb, 2023-02-02:
Great that you was getting it working.
Have you updated the firmware in the ZBGW  zigbee module ?
The original its little buggy and shall running the recommended EmberZNet version: 6.7.8.0.
Also then i was updating my ZBGW i was having similar problem and was being fixed by making one new network (i think old info in the storage off the Zigbee module).

DenZyHa, 2023-02-02:
Yes.
FW of ZBGW has been updated to EmberZNet 6.7.8.0.