zigpy/zha-device-handlers #2160: [Device Support Request] Xiaomi WS-USC04 LUMI lumi.switch.b2naus01

DigitalWarrior, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
Device state (On/Off) is not updated when the switches are physically pressed. HA is able to turn the switches on and off. 
https://www.amazon.com/dp/B081ZRBPLC
https://www.zigbee2mqtt.io/devices/WS-USC04.html

**Describe the solution you'd like**
Have Home Assistant know the status of switch regardless of how it is flipped. 

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0012",
        "0x0702",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": []
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "31": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
[... 688 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
I am able to get the correct state of the switch via the Mange Zigbee Device. 
[image]
Once I do that the state is updated correctly in Home Assistant.

TheJulianJES, 2023-02-04:
Can you press the "Reconfigure device" button in the UI and provide a screenshot of what it looks like?
Also, what coordinator are you using? If it's a Conbee, on what firmware version is it?

DigitalWarrior, 2023-02-05:
Here is the screen shot. 
[image]
Full table. 

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/richard/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/richard/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Cluster | Binding | Reporting |  
-- | -- | -- | --
  |   | Attribute | min/max/change |  
DeviceTemperatureCluster | X | current_temperature: | 30/900/50 |  
Identify |   |   |   |  
Groups |   |   |   |  
Scenes |   |   |   |  
OnOffCluster | X | on_off: | 0/900/1 |  
Alarms |   |   |   |  
MultistateInputCluster |   |   |   |  
XiaomiMeteringCluster | x | instantaneous_demand: | 5/900/1 |  
  |   | current_summ_delivered | 30/900/1 |  
  |   | current_tier1_summ_delivered: | 30/900/1 |  
  |   | current_tier2_summ_delivered: | 30/900/1 |  
  |   | current_tier3_summ_delivered: | 30/900/1 |  
  |   | current_tier4_summ_delivered: | 30/900/1 |  
  |   | current_tier5_summ_delivered: | 30/900/1 |  
  |   | current_tier6_summ_delivered: | 30/900/1 |  
  |   | status: | 1/900/1 |  
OppleSwitchCluster | x |   |   |  
Time |   |   |   |  
Ota |   |   |   |  
AnalogInput |   |   |   |  
GreenPowerProxy |   |   |   |  

</body>

</html>

I used to have the Cronbee II but migrated over to the Skyconnect

[image]

TheJulianJES, 2023-02-05:
I assume the status still doesn't update properly (even after the reconfiguration)?

DigitalWarrior, 2023-02-05:
correct

closed 2023-08-11 by github-actions[bot]

realbuz, 2024-02-26:
I'm still experiencing the same issue with everything up to date as off today.