zigpy/zha-device-handlers #2161: [BUG] Current position percentage on TS130F Tuya cover not working properly

guillesanzs, 2023-02-01:
Hello,
Hope somebody could help me, or perhaps this issue could help others....

**Bug/problem description:** 
   "class": "zhaquirks.tuya.ts130f.TuyaTS130GP"

I can read/write the attribute via ZHA:
    TuyaCoveringCluster (Endpoint id: 1, Id: 0x0102, Type: in)
        current_position_lift_percentage (id: 0x0008)

but I always read the same value regardless of whether I activate the cover or not. This leads me to problems with automations that triggers based on the current position of the cover.

**To Reproduce:**
Raspberry-Pi-4b, HAOS
Coordinator: CONBEE-II
Covers: 
  "manufacturer": "_TZ3000_zirycpws",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130GP"

Setting cover position, opening, closing… (to 40% at the image)
 
[image]

Then, even waiting for minutes, when I read the current position attribute, I always get zero:
 
[image]

Other point: immediately and only after reading the attribute, if we look at the position of the cover entity, It has been changed to zero

[image]

but the real position of the physical cover is still 40%.

**Additional information:**
-	If I write a different value to the cluster attribute, it will last forever when I read it.
-	I think I remember that a few months ago it worked well
-	The calibration_time (id: 0xf003) attribute on the same cluster works fine

**Solution I'd like?** 

- Know if a simple solution exists
- If it is a bug (on the zhaquirk ??), to be fixed
- Perhaps, in the meantime, the community to be advised to avoid automations triggering based on the current position of the cover as the behavior of the covers, especially when HA restarts, would be somehow unpredictable.

javicalle, 2023-02-02:
Could you enable the debug logs and attach the HA interaction with the device?

guillesanzs, 2023-02-04:
Thank you for your help. 

I’m attaching a filtered log file (home-assistant_2023-02-04T16-49-38.061Z_a.log) and other one (index_ieee_qry.txt) with just the records containing the ieee of the device and an index to the other file. 
I looked for the ieee: a4:c1:38:0f:63:45:ee:b9 keeping related records before and after.
Hope this is what you asked me, otherwise please let me know.
[home-assistant_2023-02-04T16-49-38.061Z_a.log](https://github.com/zigpy/zha-device-handlers/files/10609361/home-assistant_2023-02-04T16-49-38.061Z_a.log)
[index_ieee_qry.txt](https://github.com/zigpy/zha-device-handlers/files/10609362/index_ieee_qry.txt)

javicalle, 2023-02-04:
Could you also attach the device diagnostics?

And a question: after reading the cluster attribute if you set a cover position from HA (or from the device), the value is updated and keeped in time?

guillesanzs, 2023-02-05:
I'm attaching the device diagnostics.

[zha-e9a3ea159f895001306b4586af9e3e5f-_TZ3000_zirycpws TS130F-60356910b2e6c9b13c975a29ba1e6fe4.json (3).txt](https://github.com/zigpy/zha-device-handlers/files/10610832/zha-e9a3ea159f895001306b4586af9e3e5f-_TZ3000_zirycpws.TS130F-60356910b2e6c9b13c975a29ba1e6fe4.json.3.txt)

I'm not sure if I understand your question. Let me follow the example: after reading the cluster, the position indicator shows 0% but physically it is at 40%. Then, if I open the cover (up arrow), it starts to rise from 40%, not from 0%. Does it answer your question?

javicalle, 2023-02-05:
Yes, it does.

From what I can see, the device reports is updating the device data cache (as the normal behavior does), but when you read the device attribute (which does not take into account the cache), device reports that 0 value (which seems wrong) and stores it in the cache (which is the normal behavior).

From this point I don't see an easy solution. Maybe we can force the quirk to handle all the info from cache but seems too much taking into account that others attributes are getted from device without any problem.

I would suggest to fix the automations tringy to get the cover position from the entity status, and not from the Zigbee attribute read but not sure if that approach will feet your requirements.

I'm also curious about what happen when HA polls the device status. If I'm not wrong, every hour or so HA polls the device status and I have doubts if the cover position will be updated to 0 or not.

guillesanzs, 2023-02-06:
But I have problems with a trigger like this:
```
trigger:
  - platform: numeric_state
    entity_id: cover.room_1
    attribute: current_position
    below: 2
```

because when HA refresh the position (for instance when HA restarts), the automation triggers even if the cover position is above 2.

Does this trigger take into account the cache?. Or what should I do to make automations getting the cover position from the entity status?

javicalle, 2023-02-06:
> Does this trigger take into account the cache?

I can't say. 

> Or what should I do to make automations getting the cover position from the entity status?

Your automation seems good to me. And I don't know why is triggering on refresh.
I would try with something like:
```yaml
trigger:
  - platform: state
    entity_id:
      - cover.room_1
    from: open
    to: closed
```

But I don't know if would have the same problem on refresh. 🤷🏻‍♂️

guillesanzs, 2023-02-07:
Better, I explain the original problem and perhaps you could help with some ideas.
The original problem is covers very often physically stop when reaching top or bottom positions but, when the cover is almost completely closed it stays in ‘closing’ state at position 99% and, when the cover is almost completely open it stays in ‘opening’ state at position 1%. This behavior leads to problems in lovelaces showing the covers states. I could see that setting the position of the cover at 100% or 0% respectively, the cover moves the remining 1%. So I tried the following automation:

```
trigger:
  - platform: numeric_state
    entity_id: cover.room1
    for:
      hours: 0
      minutes: 0
      seconds: 2
    attribute: current_position
    below: 2
condition: []
action:
  - service: cover.set_cover_position
    data:
      position: 0
    target:
      entity_id: cover.pers_multi_este_cover
```

reaching the problems I previosuly described...

guillesanzs, 2023-02-07:
... and taking into account my design, which I devised for greater availability purposes: 

[image]

guillesanzs, 2023-02-07:
I think I worked out to a reasonable workaround by this automation:

```
trigger:
  - platform: state
    entity_id:
      - cover.room_1
    to: closing
    for:
      hours: 0
      minutes: 0
      seconds: 30
    id: "1"
  - platform: state
    entity_id:
      - cover.room_1
    to: opening
    for:
      hours: 0
      minutes: 0
      seconds: 30
    id: "2"
condition: []
action:
  - service: light.turn_off
    data: {}
    target:
      entity_id:
        - light.int_4c_room1_light_5
        - light.int_4c_room1_light_7
  - if:
      - condition: trigger
        id: "1"
    then:
      - service: cover.set_cover_position
        data:
          position: 0
        target:
          entity_id: cover.room_1
    else:
      - service: cover.set_cover_position
        data:
          position: 100
        target:
          entity_id: cover.room_1
mode: single
```
where 30 seconds = 10 seconds + callibration time

closed 2023-08-13 by github-actions[bot]