zigpy/zha-device-handlers #2162: Add generic tuya trv device

parrjsam, 2023-02-01:
New tiya trv device _TZE200_xby0s3ta needs adding to ZHA. I've currently fudged it using a current quirk and included this device name in the list in the current tuya trv quirk. But the functionality isn't 100%. Firstly the device type isn't recognised and so I have manually changed the zha device type in the HA config, to be "climate". This didn't ultimately fix the device either, the ability to increase the temperature, like a usual climate device is missing and so is the battery level. 

Funnily enough with this device, I got the same device bit a few months ago, and ZHA is reading this new one differently hence the workaround above. The previous device is covered by a quirk already, this device shows as _TZE200_cwnjrr72. 

Having _TZE200_xby0s3ta implemented into ZHA would be a great help, with all the standard trv functionality. Thanka

TheJulianJES, 2023-02-04:
Please fill out the issue template. Provide the device signature and the diagnostic information for the device.

MattWestb, 2023-02-04:
Add it on one new line after this https://github.com/zigpy/zha-device-handlers/blob/84d02be7abde55a6cee80fa155f0cbbc20347c40/zhaquirks/tuya/ts0601_trv.py#L1532
If the quirk i loaded OK it shall not being needed changing the type then it getting the TRV / climate cluster from the quirk.
If its not looking OK try restarting HA.
Also put debug logging and post the log here.

parrjsam, 2023-02-06:
**Is your feature request related to a problem? Please describe.**
As above in first post

**Describe the solution you'd like**
for this device, generic tuya TRV to be recognised by ZHA

<details>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_xby0s3ta",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.2.0",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.10.28-Unraid",
    "run_as_root": true
  },
  "custom_components": {
    "fontawesome": {
      "version": "2.1.5",
      "requirements": []
    },
    "anniversaries": {
      "version": "4.5.0",
      "requirements": [
        "python-dateutil>=2.8.1",
        "integrationhelper>=0.2.2",
        "voluptuous>=0.12.1"
      ]
    },
    "govee": {
      "version": "0.2.2",
      "requirements": [
        "govee-api-laggat==0.2.2",
[... 373 lines cut ...]
        }
      }
    }
  }
}

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
I have managed to add others, by doing as @MattWestb  has suggested with adding the device into the existing quirk, and also by changing the device class to climate. But this is all manual work around which took me days to figure out, and others may not be able to, or know how to, do the work around, so it would be good to just include this new device into the ZHA device handler. Thanks

parrjsam, 2023-02-06:
> Add it on one new line after this
>
> https://github.com/zigpy/zha-device-handlers/blob/84d02be7abde55a6cee80fa155f0cbbc20347c40/zhaquirks/tuya/ts0601_trv.py#L1532
>
>
> If the quirk i loaded OK it shall [...]

Thanks @MattWestb  this did work, as I just stated in my comment I have just previously posted. I thought it would only be fair to reply directly to your comment also. Thanks

parrjsam, 2023-02-06:
> Please fill out the issue template. Provide the device signature and the diagnostic information for the device.

As above, in separate comment. Thanks

greem, 2023-09-30:
I've just bought one of these and would be delighted to see it fully supported in ZHA.

PR #2856 (2023-12-23): Add Tuya TRV "Sandy Beach HY367" `_TZE200_xby0s3ta`
## Proposed change

Add another Tuya variant
https://www.amazon.co.uk/gp/product/B093B4SCBT

## Additional information

Fixes #2162

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2023-12-23 by TheJulianJES