zigpy/zha-device-handlers #2163: [Device Support Request] Motion Sensor with Light Sensor and Scene Switch

kaesebrot1199, 2023-02-02:
**Is your feature request related to a problem? Please describe.**
I paired the motion sensor. IAS_ZONE works but i have no entities for light sensor and scene switch. I need help for create a quirks for this device.
https://www.aliexpress.com/item/1005004767798199.html?spm=a2g0o.order_list.order_list_main.16.3c7d5c5fNzEbT2&gatewayAdapt=glo2deu

TS0202 _TZ3210_cwamkvua

Sorry for formating the request, it is my first Github request.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0500",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_cwamkvua",
  "model": "TS0202",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.0b4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.80",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "browser_mod": {
      "version": "2.2.0",
      "requirements": []
    },
    "localtuya": {
      "version": "5.0.0",
[... 327 lines cut ...]
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
![image](https://user-images.githubusercontent.com/58260827/216245005-0f3f6373-d613-4847-a15a-4d931e36b5f7.png)
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2023-02-02:
I have that quirk in my local environment:
<details open>
<summary>ts0601_pirmotion.py</summary>

```python
"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Ota, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaPowerConfigurationCluster,
)

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class SensitivityLevel(t.enum8):
    """Sensitivity level enum."""

    LOW = 0x00
    MEDIUM = 0x01
    HIGH = 0x02

class OnTimeValues(t.enum8):
    """Sensitivity level enum."""

    _10_SEC = 0x00
    _30_SEC = 0x01
    _60_SEC = 0x02
    _120_SEC = 0x03

class PirMotionManufCluster(TuyaMCUCluster):
    """Tuya manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update({0xEF09: ("sensitivity_level", SensitivityLevel)})
    attributes.update({0xEF0A: ("keep_time", OnTimeValues)})

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "occupancy",
            dp_type=TuyaDPType.BOOL,
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
        9: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "sensitivity_level",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: SensitivityLevel(x),
        ),
        10: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "keep_time",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: OnTimeValues(x),
        ),
        12: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: (10000 * math.log10(x)) + 1,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        9: "_dp_2_attr_update",
        10: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class PirMotion(CustomDevice):
    """Tuya PIR motion sensor."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_3towulqd", "TS0601"),
            ("_TZ3210_cwamkvua", "TS0601"),
        ],
        ENDPOINTS: {
            # endpoints=1 profile=260 device_type=0x0402
            # in_clusters=[0x0000, 0x0001, 0x0500],
            # out_clusters=[0x000a, 0x0019]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    PirMotionManufCluster,
                    TuyaOccupancySensing,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```

</details>

Here you have a guide to enable the local quirk:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

kaesebrot1199, 2023-02-03:
Thanks for your response.
I added your suggested custom quirks but load fails with "input cluster mismatch".

2023-02-03 09:34:33.221 INFO (MainThread) [zigpy.device] [0x0755] Read model 'TS0202' and manufacturer '_TZ3210_cwamkvua' from <Endpoint id=1 in=[power:0x0001, ias_zone:0x0500, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
2023-02-03 09:34:33.222 INFO (MainThread) [zigpy.device] [0x0755] Discovered basic device information for <Device model='TS0202' manuf='_TZ3210_cwamkvua' nwk=0x0755 ieee=a4:c1:38:10:a9:14:28:d7 is_initialized=True>
2023-02-03 09:34:33.222 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0202' manuf='_TZ3210_cwamkvua' nwk=0x0755 ieee=a4:c1:38:10:a9:14:28:d7 is_initialized=True>
2023-02-03 09:34:33.223 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZ3210_cwamkvua TS0202 (a4:c1:38:10:a9:14:28:d7)
2023-02-03 09:34:33.223 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_pirmotion.PirMotion'>
2023-02-03 09:34:33.223 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint

javicalle, 2023-02-03:
Probably not working version but can you update the signature part like this:
```python 

    signature = {
        MODELS_INFO: [
            ("_TZE200_3towulqd", "TS0601"),
            ("_TZ3210_cwamkvua", "TS0601"),
        ],
        ENDPOINTS: {
            # endpoints=1 profile=260 device_type=0x0402
            # in_clusters=[0x0000, 0x0001, 0x0500],
            # out_clusters=[0x000a, 0x0019]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster.cluster_id,
                    IasZone.cluster_id,
                    PirMotionManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

```

(the new `PirMotionManufCluster` part)

kaesebrot1199, 2023-02-03:
I changed the signature code in ts0601_pirmotion.py and rebind the device.
Device signature in HA remains at "class": "zigpy.device.Device"

[Uploading home-assistant1233.log…]()

javicalle, 2023-02-03:
Ouuuch
Replace the `MODELS_INFO` part:
```python
        MODELS_INFO: [
            ("_TZE200_3towulqd", "TS0601"),
            ("_TZ3210_cwamkvua", "TS0202"),
        ],
```

kaesebrot1199, 2023-02-03:
considering works now: Considering <class 'ts0601_pirmotion.PirMotion'>

Now we have to configure the right cluster bindings to get the entities in HA?
When i connected the motion sensor to Tuya i found the following features of the sensor:

Motion: detected / clear
Scene Switch: single click / double click / long click
Environmental light: Bright/Dark
Battery: %

What we have to do for customs quirks now to have this properties as HA entities?
Thanks.
[home-assistant1500.log](https://github.com/zigpy/zha-device-handlers/files/10579963/home-assistant1500.log)

<details>
<summary>Additional logs</summary>
"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Ota, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaPowerConfigurationCluster,
)

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class SensitivityLevel(t.enum8):
    """Sensitivity level enum."""

    LOW = 0x00
    MEDIUM = 0x01
    HIGH = 0x02

class OnTimeValues(t.enum8):
    """Sensitivity level enum."""

    _10_SEC = 0x00
    _30_SEC = 0x01
    _60_SEC = 0x02
    _120_SEC = 0x03

class PirMotionManufCluster(TuyaMCUCluster):
    """Tuya manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update({0xEF09: ("sensitivity_level", SensitivityLevel)})
    attributes.update({0xEF0A: ("keep_time", OnTimeValues)})

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "occupancy",
            dp_type=TuyaDPType.BOOL,
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
        9: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "sensitivity_level",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: SensitivityLevel(x),
        ),
        10: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "keep_time",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: OnTimeValues(x),
        ),
        12: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: (10000 * math.log10(x)) + 1,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        9: "_dp_2_attr_update",
        10: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class PirMotion(CustomDevice):
    """Tuya PIR motion sensor."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_3towulqd", "TS0601"),
            ("_TZ3210_cwamkvua", "TS0202"),
        ],
        ENDPOINTS: {
            # Endpoint id=1 in=[power:0x0001, ias_zone:0x0500, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A]
            # endpoint=1, profile=260, device_type=1026, device_version=1, input_clusters=[1, 1280, 61184, 0], output_clusters=[25, 10])
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster.cluster_id,
                    IasZone.cluster_id,
                    PirMotionManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    PirMotionManufCluster,
                    TuyaOccupancySensing,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
</details>

javicalle, 2023-02-03:
We need to map the Tuya DPs to clusters.
Decons is not too friendly to get the required info, but (as always) Z2M had done a good job.
DPs mappings are:
* occupancy: DP3
* battery: DP25 (battery_voltage?)
* tap_action: DP101
* light/brightness: DP102

My quick and dirty proposal would be:

<details open>

```python
"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Ota, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaPowerConfigurationCluster,
)

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TapAction(t.enum8):
    """Sensitivity level enum."""

    SINGLE = 0x00
    HOLD = 0x01
    DOUBLE = 0x02

class EnvironmentalLight(t.enum8):
    """Sensitivity level enum."""

    BRIGHT = 0x00
    DARK = 0x01

class PirMotionManufCluster(TuyaMCUCluster):
    """Tuya manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update({0xEF65: ("scene_switch", TapAction)})
    attributes.update({0xEF66: ("environmental_light", EnvironmentalLight)})

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        3: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "occupancy",
            dp_type=TuyaDPType.BOOL,
        ),
        25: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "scene_switch",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: TapAction(x),
        ),
        102: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "environmental_light",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: EnvironmentalLight(x),
        ),
    }

    data_point_handlers = {
        3: "_dp_2_attr_update",
        25: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
    }

class PirMotion(CustomDevice):
    """Tuya PIR motion sensor."""

    signature = {
        MODELS_INFO: [("_TZ3210_cwamkvua", "TS0202")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster.cluster_id,
                    IasZone.cluster_id,
                    PirMotionManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    PirMotionManufCluster,
                    TuyaOccupancySensing,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```

</details>

kaesebrot1199, 2023-02-07:
[home-assistant1756.log](https://github.com/zigpy/zha-device-handlers/files/10677578/home-assistant1756.log)
[image]

I updated ts0601_pirmotion.py, reboot, and repaired the device.
Only  "occupancy" available.

jposouza, 2023-05-24:
Hello. Thanks for the post.

Any solution for this?

GeneralLouis, 2023-07-26:
@javicalle you still around and willing to help?

I have been following this thread with the same device trying to get it (and the rest of the devices that came with my Fingerbot Sense Kickstarter from Adaprox) and I think I have made progress.

Originally I would just get the sensor for PIR but it would just always be "open", now that I have enabled quirks and dropped your file in it now reacts to motion but that is it. Also, what is weird, is the ZigBee info tab doesn't show a quirk active like it is on the 2-button TuYa switch I got working.

Current PIR and scene button (TS0202)
[image]

Adaprox (TuYa guts) TS0042 with Quirk loading
[image]