zigpy/zha-device-handlers #2164: [Device Support Request] Eglo Zigbee PIR sensor IP44

simlun, 2023-02-03:
**Is your feature request related to a problem? Please describe.**

Suport for Eglo Connect-Z Zigbee PIR sensor IP44

- https://www.eglo.com/se/tillbehoer-connect-z-sensor-99106.html
- https://www.eglo.com/de/zubehoer-connect-sensor-97475.html

I own a couple of art nr 99106 sold in Sweden.

I'm using Home Assistant and ZHA in my home setup and it doesn't pick up my device at all when I scan for new devices.

I'm a software developer and I bought a ConBee II to being able to debug and add this Eglo PIR sensor to ZHA, but I don't know how to get started adding quirks and whatnot so I can use my Eglo PIR sensors at home. I don't know what to install to get started, nor the workflow to add new devices.

Please help me getting started and contribute to ZHA.

Best regards
Simon Lundmark

TheJulianJES, 2023-02-04:
Is there anything in the logs when you try to pair the devices? Quirks generally don't affect pairing at all.

DaseineWesen, 2023-02-25:
I have the same problem as well, the sensor won´t show up, nether in the logs nor in the dashboard or the device list.

closed 2023-08-31 by github-actions[bot]