zigpy/zha-device-handlers #2165: [BUG] Smartenergy cluster: Instantaneous demand and Summation delivered error in log

CSRLabs, 2023-02-03:
**Describe the bug**
Home Assistant core 2023.2.1 - Python 3.10.9
Using a DIY device based on PTVO firmware with a custom quirk I get the following error in the log:

```
2023-02-03 22:02:51.579 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.csrlabs_pzem004t_instantaneous_demand (<class 'homeassistant.components.zha.sensor.SmartEnergyMetering'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('power') it is using; expected one of ['W', 'kW']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+zha%22
2023-02-03 22:02:51.587 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.csrlabs_pzem004t_summation_delivered (<class 'homeassistant.components.zha.sensor.SmartEnergySummation'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'kWh', 'MWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+zha%22
```

And the correspondent entities are missing the unit of measurement

[image]

This is the relevant code in the quirk

```
class SmartEnergyCluster(LocalDataCluster, Metering):

    cluster_id = Metering.cluster_id
    CURRENT_SUMM_DELIVERED_VALUE_ID = 0x0000
    INSTANTANEOUS_DEMAND_VALUE_ID = 0x0400

    POWER_WATT = 0x0000
    
    """Setting unit of measurement."""
    _CONSTANT_ATTRIBUTES = {0x0300: POWER_WATT}
    
    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.energy_bus.add_listener(self)

    def energy_reported(self, value):
        """Energy reported."""
        self._update_attribute(self.CURRENT_SUMM_DELIVERED_VALUE_ID, value)
        
    def instantaneous_demand(self, value):
        """Instantaneous power demand reported."""
        self._update_attribute(self.INSTANTANEOUS_DEMAND_VALUE_ID, value)
```

The device was working fine before the upgrade.

Any idea on how to fix this or adjust the quirk code is welcome

javicalle, 2023-02-04:
Not sure if can be related to the `summation_formatting`/`demand_formatting` or the `multiplier`/`divisor` attributes:
https://github.com/home-assistant/core/blob/fc38b4327f0ce153ac757492959eba9aefbcbc1a/homeassistant/components/zha/sensor.py#L483-L490

I would try to set values to check.

CSRLabs, 2023-02-04:
I inserted this code in the quirk to set multiplier and divisor

```
_CONSTANT_ATTRIBUTES = {
        0x0300: 0,  # unit_of_measure: kWh
        0x0301: 1,  # multiplier
        0x0302: 1000,  # divisor
        0x0303: 0b0_0100_011,  # summation_formatting (read from plug)
        0x0306: 0,  # metering_device_type: electri
}
```

I deleted and re-paired the device, but it seems it doesn't help as I'm now getting these errors:

```
2023-02-04 10:46:39.569 ERROR (MainThread) [homeassistant.util.logging] Exception in async_set_state when dispatching '00:12:4b:00:25:9c:36:80:1:0x0702_attribute_updated': (0, 'current_summ_delivered', 0.001)
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/zha/sensor.py", line 177, in async_set_state
    self.async_write_ha_state()
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/helpers/entity.py", line 566, in async_write_ha_state
    self._async_write_ha_state()
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/helpers/entity.py", line 607, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/helpers/entity.py", line 572, in _stringify_state
    if (state := self.state) is None:
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/sensor/__init__.py", line 466, in state
    value = self.native_value
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/zha/sensor.py", line 172, in native_value
    return self.formatter(raw_state)
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/zha/sensor.py", line 486, in formatter
    return self._channel.summa_formatter(value)
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/zha/core/channels/smartenergy.py", line 243, in _formatter_function
    assert self._summa_format
AssertionError

2023-02-04 10:46:39.625 ERROR (MainThread) [homeassistant.util.logging] Exception in async_set_state when dispatching '00:12:4b:00:25:9c:36:80:1:0x0702_attribute_updated': (1024, 'instantaneous_demand', 0.0)
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/zha/sensor.py", line 177, in async_set_state
    self.async_write_ha_state()
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/helpers/entity.py", line 566, in async_write_ha_state
    self._async_write_ha_state()
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/helpers/entity.py", line 607, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/helpers/entity.py", line 572, in _stringify_state
    if (state := self.state) is None:
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/sensor/__init__.py", line 466, in state
    value = self.native_value
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/zha/sensor.py", line 172, in native_value
    return self.formatter(raw_state)
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/zha/sensor.py", line 437, in formatter
    return self._channel.demand_formatter(value)
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/zha/core/channels/smartenergy.py", line 245, in _formatter_function
    assert self._format_spec
AssertionError

```

javicalle, 2023-02-04:
Could you test just removing the `summation_formatting`?
I will try to check latter?

CSRLabs, 2023-02-04:
@javicalle done, deleted and re-paired device.
Now the quirk looks like this:

```
_CONSTANT_ATTRIBUTES = {
        0x0300: 0,  # unit_of_measure: kWh
        0x0301: 1,  # multiplier
        0x0302: 1000,  # divisor
        0x0306: 0,  # metering_device_type: electric
}
```

but I still get the same errors and no unit of measurement.

javicalle, 2023-02-04:
Ummmm, have you deleted the `__pycache__` folder inside the custom quirks folder?
You shouldn't get any errors because of the `summation_formatting`.
Be sure to delete it, restart HA and repair.

CSRLabs, 2023-02-04:
Deleted `__pycache__` restarted HA deleted device, re-paired, same error as above.

TheJulianJES, 2023-02-04:
Can you provide the diagnostic information for the device (after a re-pair with the custom quirks)?

CSRLabs, 2023-02-04:
This is the log with device pairing:

```
2023-02-04 18:23:50.184 INFO (MainThread) [zigpy_znp.zigbee.application] Permitting joins for 254 seconds
2023-02-04 18:23:50.187 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=6, profile_id=0, cluster_id=<ZDOCmd.Mgmt_Permit_Joining_req: 0x0036>, data=Serialized[b'\x06\xfe\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-04 18:23:50.194 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.MgmtPermitJoinReq.Req(AddrMode=<AddrMode.Broadcast: 15>, Dst=0xFFFC, Duration=254, TCSignificance=0)
2023-02-04 18:23:50.205 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MgmtPermitJoinReq.Rsp(Status=<Status.SUCCESS: 0>)
2023-02-04 18:23:50.209 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MgmtPermitJoinRsp.Callback(Src=0x0000, Status=<Status.SUCCESS: 0>)
2023-02-04 18:23:50.213 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.Broadcast: 15>, address=0xFFFC), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=54, TSN=6, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x06\xFE\x00')
2023-02-04 18:23:50.220 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32822, SecurityUse=0, TSN=3, MacDst=0x0000, Data=b'\x00')
2023-02-04 18:23:50.224 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=3, profile_id=0, cluster_id=32822, data=Serialized[b'\x03\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-04 18:23:50.226 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
2023-02-04 18:23:50.227 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
2023-02-04 18:23:50.239 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-02-04 18:23:57.685 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0x3FBB, SrcIEEE=00:12:4b:00:25:9c:36:80, ParentNwk=0x0000)
2023-02-04 18:23:57.688 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0x3FBB, SrcIEEE=00:12:4b:00:25:9c:36:80, ParentNwk=0x0000)
2023-02-04 18:23:57.691 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0x3FBB, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2023-02-04 18:23:57.699 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
2023-02-04 18:24:01.241 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2023-02-04 18:24:01.249 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2023-02-04 18:24:02.690 INFO (MainThread) [zigpy.application] New device 0x3fbb (00:12:4b:00:25:9c:36:80) joined the network
2023-02-04 18:24:02.692 DEBUG (MainThread) [zigpy.device] [0x3fbb] Scheduling initialization
2023-02-04 18:24:02.696 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2023-02-04 18:24:02.697 INFO (MainThread) [zigpy.device] [0x3fbb] Requesting 'Node Descriptor'
2023-02-04 18:24:02.698 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2023-02-04 18:24:02.699 DEBUG (MainThread) [zigpy.device] [0x3fbb] Extending timeout for 0x07 request
2023-02-04 18:24:02.699 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3FBB), dst_ep=0, source_route=None, extended_timeout=True, tsn=7, profile_id=0, cluster_id=<ZDOCmd.Node_Desc_req: 0x0002>, data=Serialized[b'\x07\xbb?'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-04 18:24:02.702 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteChk.Req(Dst=0x3FBB, RtStatus=<RouteStatus.ACTIVE: 1>, Options=<RouteOptions.NO_ROUTE_CACHE|MTO_ROUTE: 3>)
2023-02-04 18:24:02.715 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteChk.Rsp(Status=<RoutingStatus.FAIL: 1>)
2023-02-04 18:24:02.720 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0x3FBB, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2023-02-04 18:24:02.733 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
2023-02-04 18:24:04.037 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x3FBB), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=2, TSN=7, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x07\xBB\x3F')
[... cut ...]
    return self._channel.demand_formatter(value)
  File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/components/zha/core/channels/smartenergy.py", line 245, in _formatter_function
    assert self._format_spec
AssertionError
```
The first time the device is seen by HA the two entities with errors appear as unknown as you can see below:

[image]

Following HA restart the two entities are OK but with no unit of measurement and in the log I see two entries complaining for unit of measurement 'None'

CSRLabs, 2023-02-04:
Here is the quick I'm currently testing

[PZEM-004T](https://github.com/CSRLabs/PZEM-004T/blob/main/pzem004t.py)

TheJulianJES, 2023-02-04:
Can you provide the diagnostic information txt file? It can be downloaded on the device page under the three dots -> download diagnostic informations

CSRLabs, 2023-02-04:
Here is the file

[zha-536e57d62c4c1ef339dbddd9d9b8ce02-CSRLabs pzem004t-062b94e387c1e11ee615e8d55bc5421e.json.txt](https://github.com/zigpy/zha-device-handlers/files/10609303/zha-536e57d62c4c1ef339dbddd9d9b8ce02-CSRLabs.pzem004t-062b94e387c1e11ee615e8d55bc5421e.json.txt)

TheJulianJES, 2023-02-04:
Can you add back ``0x0303: 0b0_0100_011,  # summation_formatting``, then restart HA after you've changed the quirk, remove the device and wait a minute or so before adding it back (re-pairing alone is not enough).
Then send the logs here if it still errors out (and a new diagnostic file). Also, try to read that `summation_formatting` attribute through the UI. It's under the three dots -> Manage Zigbee device -> Clusters. Then post a screenshot of what it returns (and try to restart HA again after having read the attribute (which caches it)).

---

Also, make sure you check the type of attributers you're setting in the ``LocalDataCluster``. For example, ``rms_voltage`` needs to be an integer. So you need to ``round()`` it first. See https://github.com/zigpy/zha-device-handlers/pull/2041 for reference.

---

Btw., I don't think you need to use ``Bus`` anymore. You should be able to just do something like this:
```python
self.endpoint.smartenergy_metering.update_attribute(Metering.attributes_by_name["current_summ_delivered"], value)
```
(You'd still need the ``LocalDataCluster``s, but nothing with bus stuff)

You can find all names here: https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/measurement.py, https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/smartenergy.py, https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/homeautomation.py (and maybe in other files in the clusters folder)

CSRLabs, 2023-02-04:
@TheJulianJES thanks for taking care of this.
Adding back `0x0303: 0b0_0100_011,  # summation_formatting` did not change anything, still the same behavior.
I will take some time to study your third line advice and possibly change my quirk accordingly.
BTW, it would be of great help if you could point me to an example quirk (a similar energy monitor device or so) that proved to work properly with the latest version of HA

CSRLabs, 2023-02-05:
@TheJulianJES reading the values from "Manage Zigbee Device" seemed to fix the unit of measurement issue, see the screenshot below:

[image]

[image]

[image]

So that means that the user always needs to do this manual action to make the device working?
Is there any way to complete the device pairing from inside the quirk without needing any extra action from the UI?

TheJulianJES, 2023-02-05:
These values are read automatically on pairing. How did you remove the device from ZHA before re-pairing it?

CSRLabs, 2023-02-05:
I went on devices and then clicked on the red bin "Remove"

[image]

Then restarted HA, cliked Add Device and powered on my device, and it paired.

CSRLabs, 2023-02-05:
I eventually found a way to fix this, I'm not sure whether this is actually the right way, but it works for me.
In the quirk I added this code to force attribute update for constant attributes, when the MeteringCluster class gets initialized

```
def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.consumption_bus.add_listener(self)

        # initialize constant attributes
        self._update_attribute(self.UNIT_OF_MEASUREMENT, 0)
        self._update_attribute(self.MULTIPLIER, 1)
        self._update_attribute(self.DIVISOR, 1000)
        self._update_attribute(self.SUMMATION_FORMATTING, 0b0_0100_011)
        self._update_attribute(self.METERING_DEVICE_TYPE, 0)
```

To simulate a "fresh pairing" and get rid of all device traces I followed these steps

deleted .homeassistant profile
started HA
reinstalled ZigBee integration choosing to erase network settings and create a new network
re-paired device

The device pairs nicely, and all entities are available from the beginning without any additional manual action. No errors in the log.

[image]

I see also that it is still not possible to choose the precision (decimal digits) for voltage and current (at least I did not find a way) so my quirk still has a class patching to achieve this. I know it's a dirty way to do it but it works for now.

```
from homeassistant.components.zha.sensor import ElectricalMeasurement as EM

def myformatter(self, value: int) -> int | float:
    """Return 'normalized' value."""
    multiplier = getattr(self._channel, f"{self._div_mul_prefix}_multiplier")
    divisor = getattr(self._channel, f"{self._div_mul_prefix}_divisor")
    value = float(value * multiplier) / divisor
    return round(value, self._decimals)

EM.formatter = myformatter
```

TheJulianJES, 2023-02-05:
> I eventually found a way to fix this, I'm not sure whether this is actually the right way

It's not. Like mentioned, the attributes are read when initially pairing the device: https://github.com/home-assistant/core/blob/04e05af44cbea2156d904d0d34e4e0d41c9725c1/homeassistant/components/zha/core/channels/smartenergy.py#L91-L98
It's likely that the device wasn't correctly removed (from ``zigbee.db``) yet. Wait a minute or so after removing the device from ZHA before re-pairing it. The quirk needs to be present during pairing though.

> To simulate a "fresh pairing" and get rid of all device traces I followed these steps

The attribute readings are saved in the ``zigbee.db`` file (which intentionally isn't deleted when deleting the integration).

> I see also that it is still not possible to choose the precision (decimal digits) for voltage and current

ZCL has limitations, but ``ac_current_divisor `` and ``ac_voltage_divisor`` on the EM cluster should work if I'm correct.

CSRLabs, 2023-02-05:
@TheJulianJES thanks for following up.
I'm building a fresh HA system. I will try to remove the `_update_attribute` additional code from the quirk and pair the device one more time.
Not sure if it is worth mentioning, the coordinator I'm using is based on CC2652P with latest 20221226 Koenkk FW.

closed 2023-08-11 by github-actions[bot]