zigpy/zha-device-handlers #2166: [Device Support Request]

grego1981, 2023-02-03:
**Is your feature request related to a problem? Please describe.**
I have some hysen HY-10 zigbee trvs that are partially detected from tuya ts601 module as Moes type2. ZHA is intervieweing them but cannot add them. I have the following error in the logs and I believe that these have the attirbute present in the moes type1... 

Following the logs-->

```
WARNING (MainThread) [homeassistant.components.zha.core.gateway] Couldn't start EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis coordinator (attempt 3 of 3)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 174, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 163, in new
    await app._load_db()
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 80, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.10/site-packages/zigpy/appdb.py", line 577, in load
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 34, in get_device
    return _DEVICE_REGISTRY.get_device(device)
  File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/registry.py", line 148, in get_device
    device = candidate(device._application, device.ieee, device.nwk, device)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 874, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 85, in __init__
    self.add_endpoint(endpoint_id, replace_device=replaces)
  File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 100, in add_endpoint
    ep = custom_ep_type(self, endpoint_id, replacement_data, replace_device)
  File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 124, in __init__
    cluster = c(self, is_server=True)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/ts0601_trv.py", line 785, in __init__
    self.endpoint.device.window_detection_bus.add_listener(self)
AttributeError: 'MoesHY368_Type2' object has no attribute 'window_detection_bus'
```

**Describe the solution you'd like**
Maybe the block of code from Moes type1 may be the solution but in HASSos I cannot (actually do not know) how to access and change the file zhaquirks/tuya/ts0601_trv.py
```
    def __init__(self, *args, **kwargs):
        """Init device."""
        self.window_detection_bus = Bus()
        super().__init__(*args, **kwargs)
```

TheJulianJES, 2023-02-03:
Please fill out the issue template. Device signature and diagnostic information from the device are still missing.

grego1981, 2023-02-03:
Hi and thanks for your answer, is there a way to get these information? Is there a guide to help me retrieve the missing data? Thank you and sorry for any inconvenience caused!

grego1981, 2023-02-03:
I have tried following the guidlines from HA ZHA document but the part:
Both the “Zigbee Device Signature” and “Diagnostics” information can be found by clicking Settings -> Devices & Services -> Zigbee Home Automation (click Configure) -> Devices (pick your device) -> Click “Zigbee Device Signature” and “Download Diagnostics” respectively.

Is not usable as the device is not added and not showing in ZHA addon,,,

Thia is the log when adding the device as new

```
New device 0x9957 (ec:1b:bd:ff:fe:95:40:50) joined the network
[0x9957] Scheduling initialization
Tries remaining: 3
[0x9957] Requesting 'Node Descriptor'
Tries remaining: 2
[0x9957] Extending timeout for 0x2b request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9957), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=252, profile_id=0, cluster_id=19, data=Serialized[b'\x85W\x99P@\x95\xfe\xff\xbd\x1b\xec\x80'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=232, rssi=-42)
Device 0x9957 (ec:1b:bd:ff:fe:95:40:50) joined the network
[0x9957] Scheduling initialization
[0x9957] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9957 ieee=ec:1b:bd:ff:fe:95:40:50 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x85W\x99P@\x95\xfe\xff\xbd\x1b\xec\x80'
[0x9957:zdo] ZDO request ZDOCmd.Device_annce: [0x9957, ec:1b:bd:ff:fe:95:40:50, 128]
Tries remaining: 3
[0x9957] Requesting 'Node Descriptor'
Tries remaining: 2
[0x9957] Extending timeout for 0x2d request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9957), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=253, profile_id=260, cluster_id=61184, data=Serialized[b'\t\x17$\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=216, rssi=-46)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9957 ieee=ec:1b:bd:ff:fe:95:40:50 is_initialized=False> from ep 1 to ep 1, cluster 61184: b'\t\x17$\x00\x00'
[0x9957] Received ZCL while uninitialized on endpoint id 1, cluster 0xef00 id, hdr: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=23, command_id=36, *direction=<Direction.Client_to_Server: 1>, *is_reply=True), payload: b'\x00\x00'
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9957), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=254, profile_id=0, cluster_id=32770, data=Serialized[b'-\x00W\x99\x02@\x80\x00\x00R\xff\x00\x00,\xff\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=216, rssi=-46)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9957 ieee=ec:1b:bd:ff:fe:95:40:50 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'-\x00W\x99\x02@\x80\x00\x00R\xff\x00\x00,\xff\x00\x00'
[0x9957] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x9957] Discovering endpoints
Tries remaining: 3
[0x9957] Extending timeout for 0x2f request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9957), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=255, profile_id=0, cluster_id=32770, data=Serialized[b'-\x00W\x99\x02@\x80\x00\x00R\xff\x00\x00,\xff\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=216, rssi=-46)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9957 ieee=ec:1b:bd:ff:fe:95:40:50 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'-\x00W\x99\x02@\x80\x00\x00R\xff\x00\x00,\xff\x00\x00'
[0x9957:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0x9957, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_ma [... cut ...]
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9957), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=78, profile_id=260, cluster_id=61184, data=Serialized[b'\t\x16\x01\x00\x17u\x02\x00\x04\x00\x00\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=220, rssi=-45)
Ignoring message (b'09160100177502000400000001') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (ec:1b:bd:ff:fe:95:40:50, 1)'

```

TheJulianJES, 2023-02-03:
Ah, missed that part initially. What HA version are you running?

grego1981, 2023-02-03:
I'm using HAOS with 2023.2.0 on x86 machine. I'm using the SONOFF Zigbee 3.0 USB Dongle Plus-E.

TheJulianJES, 2023-02-04:
Might be a duplicate of: https://github.com/zigpy/zha-device-handlers/issues/2114
FYI @xonestonex, as you did the initial implementation

It seems like just adding the bus to the ``MoesHY368_Type2`` constructor should be enough to fix this (and Tuya TRVs really need to be cleaned up in the future)
```python
self.window_detection_bus = Bus()
```

TheJulianJES, 2023-02-04:
@grego1981 Please setup custom quirks like this: https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

So add a ``custom_zha_quirks`` folder in your HA ``/config`` directory. Then, add the following to your ``configuration.yaml``:
```yaml
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

[ts0601_trv.py.zip](https://github.com/zigpy/zha-device-handlers/files/10609089/ts0601_trv.py.zip)

In that ``custom_zha_quirks`` folder, extract the above file and put the extracted ``.py`` file in that folder. Then restart HA. Then see if you can pair the thermostat. If it works, let me know and we'll fix it for the next HA release (by which point you should remove the custom quirk).

If HA doesn't start or it still doesn't pair, remove the ``.py`` file and send me the logs.

grego1981, 2023-02-04:
@TheJulianJES I have followed your instructions and the devices are detected and added now! I see though there is a difference in how they work.. The thermostat card has only the option of heat (in Zigbee2MQTT had the option to turn off) although I see a switch entity, I have to test it a bit and see. In all I'll inform you of the stability as well, due to the fact that with Z2M the trvs went offline after 3 days. and needed to cold boot them!

grego1981, 2023-02-05:
The switch seems to control the window detection, when enabling the switch the window icon shows up on the TRV's screen

closed 2023-02-06 by dmulcahey

grego1981, 2023-02-12:
Since my last message everything works rock solid! I think that everything works as it should!

PR #2171 (2023-02-06): Add missing window_detection_bus to Tuya TRV MoesHY368_Type2
Fixes #2166
Fixes #2114

Adding the ``window_detection_bus`` was apparently missed when adding a new signature.
(Tuya TRVs should be cleaned up at some point too)

~~Waiting for confirmation that this actually fixes the issue in https://github.com/zigpy/zha-device-handlers/issues/2166~~ ✅
commit 959a3b70 (2023-02-06): Add missing window_detection_bus to Tuya TRV MoesHY368_Type2 (#2171)

Fixes #2166