zigpy/zha-device-handlers #2174: [Device Support Request] MIR-TE100-TY temp and humidity sensor no entities in ZHA

pbvdven, 2023-02-04:
**Is your feature request related to a problem? Please describe.**
The device shows no entities
[image]

**Describe the solution you'd like**
Would be nice that this device would work with zha. i thought it would because it says here its supported https://zigbee.blakadder.com/Tuya_MIR-TE100.html but maybe its a newer version?

<details>
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.2.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.10.0-21-amd64",
    "supervisor": "2023.01.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.23",
    "chassis": "desktop",
    "run_as_root": true
  },
  "custom_components": {
    "helium": {
      "version": "0.3.8",
      "requirements": [
        "beautifulsoup4>=4.9.0",
        "jsonpath>=0.82",
        "xmltodict>=0.12.0",
        "httpx>=0.16.1"
      ]
    },
    "watchman": {
      "version": "0.5.1",
      "requirements": [
        "prettytable==3.0.0"
      ]
    },
    "webrtc": {
      "version": "v3.0.2",
      "requirements": []
    },
    "sonoff": {
      "version": "3.3.1",
      "requirements": [
        "pycryptodome>=3.6.6"
      ]
    },
    "auth_header": {
      "version": "1.4",
      "requirements": []
    },
    "smartir": {
      "version": "1.17.6",
      "requirements": [
        "aiofiles==0.6.0"
      ]
    },
    "ezviz_cloud": {
      "version": "0.1.0.6",
      "requirements": [
        "pyezviz==0.2.0.12"
      ]
    },
    "hacs": {
[... 300 lines cut ...]
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

javicalle, 2023-02-05:
I would suggest to add your device in the `TuyaTempHumiditySensor` quirk:
https://github.com/zigpy/zha-device-handlers/blob/1bfe93211342b370aa0c3ab37f384c35fceb4c32/zhaquirks/tuya/ts0601_sensor.py#L78-L103

Here are a good guide to config your local quirk folder.
Put the current `ts0601_sensor.py` inside:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py
Edit the `TuyaTempHumiditySensor` class adding your device like:
```python
class TuyaTempHumiditySensor(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""

    # RelativeHumidity multiplier
    RH_MULTIPLIER = 10

    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [
            ("_TZE200_bjawzodf", "TS0601"),
            ("_TZE200_zl1kmjqx", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
    .../...
```
Save changes, restart HA and repair your device.

pbvdven, 2023-02-05:
Hi, @javicalle 

When i add the change zha stops working after i reboot all zha entities are unavailable without the change but with the original quirk zha still works after reboot .

"""Tuya temp and humidity sensors."""

from typing import Any, Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import (
    RelativeHumidity,
    SoilMoisture,
    TemperatureMeasurement,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaSoilMoisture(SoilMoisture, TuyaLocalCluster):
    """Tuya local SoilMoisture cluster with a device RH_MULTIPLIER factor if required."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster with a device RH_MULTIPLIER factor."""

    def update_attribute(self, attr_name: str, value: Any) -> None:
        """Apply a correction factor to value."""

        if attr_name == "measured_value":
            value = value * (
                self.endpoint.device.RH_MULTIPLIER
                if hasattr(self.endpoint.device, "RH_MULTIPLIER")
                else 100
            )
        return super().update_attribute(attr_name, value)

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
        2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            # converter=lambda x: x * 10,  --> move conversion to TuyaRelativeHumidity cluster
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
            converter=lambda x: x * 2,  # double reported percentage
        ),
    }

[... 192 lines cut ...]
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    SoilManufCluster,
                    TuyaTemperatureMeasurement,
                    TuyaSoilMoisture,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

pbvdven, 2023-02-05:
@javicalle i only put the quirk in the folder and restarted ha not changed anything but i get this error in home assistant after reboot and zha entities are not available any idea what im doing wrong?

Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts0601_sensor.py:54
First occurred: 22:51:46 (1 occurrences)
Last logged: 22:51:46

Error setting up entry ConBee II, s/n: DE2423162 - dresden elektronik ingenieurtechnik GmbH for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 382, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_sensor.py", line 50, in <module>
    class TemperatureHumidityManufCluster(TuyaMCUCluster):
  File "/config/custom_zha_quirks/ts0601_sensor.py", line 54, in TemperatureHumidityManufCluster
    1: DPToAttributeMapping(
TypeError: DPToAttributeMapping.__init__() missing 1 required positional argument: 'dp_type'

pbvdven, 2023-02-05:
@javicalle thanks for helping someone changed the file https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py and that's not working for me but i took a previous version and edit that one and now the device is working.

javicalle, 2023-02-06:
> i took a previous version and edit that one and now the device is working.

👍🏻 I didn't realize that this change was pending.

Can you confirm that RH values are fine too?

pbvdven, 2023-02-06:
> > i took a previous version and edit that one and now the device is working.
> 
> 
> 
> 👍🏻 I didn't realize that this change was pending.
> 
> 
> 
> Can you confirm that RH values are fine too?
> 
> 
> 
> 

Yes the values are correct. Thanks

[image]
[image]

tof92130, 2023-03-01:
> @javicalle thanks for helping someone changed the file https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py and that's not working for me but i took a previous version and edit that one and now the device is working.

I am facing same problem.
Which previous version did you take ?

javicalle, 2023-03-02:
That would be fixed in the 2022.03 HA release.

PR #2179 (2023-02-06): Add MIR-TE100-TY sensor
Add the device signature for MIR-TE100-TY temp and humidity sensor:
* _TZE200_zl1kmjqx

Added to `TuyaTempHumiditySensor` quirk.

Fixes: #2174