zigpy/zha-device-handlers #2177: [Device Support Request] IKEA Åskväder - Support on/off status notification for upper layers

janhf, 2023-02-05:
**Is your feature request related to a problem? Please describe.**
The IKEA Åskväder is a modular multiple socket outlet. There is one zigbee module which allows to turn on/off all following devices. This device does not report its current status over zigbee. It is technically visible like a light-bulb to zigpy. You can pair other remotes (buttons, switches) to this device, but you cannot pair another light to it.

It is really annoying, that you can't use the button on this device to control other actions in home assistant with it. (For example switching on another light or device).

**Describe the solution you'd like**
The plug should report its on/off status to the upper layer regulary (perhaps after a configurable time).
This device is powered from mains. Energy draw for polling should not be a problem here.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000",
        "0xfc57"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "ASKVADER on/off switch",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
[... 654 lines cut ...]
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

janhf, 2023-02-05:
I may be able to produce the needed code by myself, but i need a little advise for examples. Maybe some device already does/uses something like this already?

MattWestb, 2023-02-06:
You can from the device card clicking  `+`  right from automatons and using the device is turned on / off as trigger and do automatons for turning other light on and off.
It can being that you is getting little delay in the reporting from the device.

Then the device is `ON_OFF_PLUG_IN_UNIT = 0x010A` i think you can also changing the type in the setting from light to Outlet and you is getting one better icon for your device.

janhf, 2023-02-06:
That's the current setup:
[image]
It does not work. The event "turned on" is never triggered automatically. 

[image]
If i read the attribute "on_off" from cluster "OnOff" manually, the value is updated and the automation is triggered.

TheJulianJES, 2023-02-06:
Can you reconfigure the device and send a screenshot of the reconfiguration dialog?

janhf, 2023-02-06:
Here is is the screenshot:
[image]

Ok. That's the trick:
Reconfiguring the device seems to have done something with the bindings.
The event is sent right now.

closed 2023-02-06 by janhf