zigpy/zha-device-handlers #2178: [Device Support Request] TS0601_TZE200_kafooqvr

Paddy-van-CH, 2023-02-06:
**Is your feature request related to a problem? Please describe.**
Device attributes not showing up in HA. 

**Describe the solution you'd like**
The device does look very similar to the Moss BHT. So I have tried to add the device to the ts0601_electric_heating quirk, but no luck. 

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_kafooqvr",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.1.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Zurich",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.6",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.90",
      "zigpy-deconz==0.19.2",
      "zigpy==0.53.0",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.2"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "0403",
        "pid": "6015",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate+"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      },
      {
        "type": "_zigate-zigbee-gateway._tcp.local.",
        "name": "*zigate*"
      },
      {
        "type": "_zigstar_gw._tcp.local.",
        "name": "*zigstar*"
      },
      {
        "type": "_slzb-06._tcp.local.",
        "name": "slzb-06*"
      }
    ],
    "dependencies": [
      "file_upload"
    ],
    "after_dependencies": [
      "onboarding",
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 24641,
    "manufacturer": "_TZE200_kafooqvr",
    "model": "TS0601",
    "name": "_TZE200_kafooqvr TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4098,
    "power_source": "Mains",
    "lqi": 84,
    "rssi": null,
    "last_seen": "2023-01-29T15:01:05",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0051",
          "in_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "acc7239cb0f7a63ac0de8ce9a9c0f712",
    "area_id": "technik",
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "SMART_PLUG",
          "id": 81
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0001": {
                "attribute_name": "app_version",
                "value": 72
              },
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE200_kafooqvr"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/10659906/home-assistant.log)

```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2023-02-08:
Please add a link to manufacturer or vendor site to understand the device capabilties.
If you add the device manufacturer the device loads the `MoesBHT` quirk?
If so, then enable the debud logs and attach the relevant info.

Paddy-van-CH, 2023-02-09:
Here's the link to the vendor site: [https://www.brelag.com/shop/wtfha025-knockautx-wandthermostat-fur-bodenheizungs-stellantrieb-4696#attr= ](url) and the user manuals: [https://drive.google.com/drive/folders/1Dbje4-V5GnaDrfeANIxYWRDWmJMYgrew](url)
Unfortunately no English versions available. I'll try my hands on translating it if needed.

I've added the device to the MoseBHT quirk and it did load it. (I.e. coping the quirk code into a file in "custom_quirks" folder, adding the folder in configuration.yaml)
Debug log attached. 
[ts0601_electric_heating.log](https://github.com/zigpy/zha-device-handlers/files/10693588/ts0601_electric_heating.log)

Paddy-van-CH, 2023-02-09:
Here's a quick translation.
[Gebrauchsanleitung KnockautX Wandthermostat für Bodenheizungs-Stellantrieb WTFHA025 V3.0_en.docx](https://github.com/zigpy/zha-device-handlers/files/10693763/Gebrauchsanleitung.KnockautX.Wandthermostat.fur.Bodenheizungs-Stellantrieb.WTFHA025.V3.0_en.docx)

javicalle, 2023-02-19:
Your logs show that the quirk is not loading. Have you added your device in the local quirk?

It should be something like:
```python
class MoesBHT(TuyaThermostat):
    """Tuya thermostat for devices like the Moes BHT-002GCLZB valve and BHT-003GBLZB Electric floor heating."""

    signature = {
        #  endpoint=1 profile=260 device_type=81 device_version=1 input_clusters=[0, 4, 5, 61184],
        #  output_clusters=[10, 25]
        MODELS_INFO: [
            ("_TZE200_aoclfnxz", "TS0601"),
            ("_TZE200_2ekuz3dz", "TS0601"),
            ("_TZE200_ye5jkfsb", "TS0601"),
            ("_TZE200_u9bfwha0", "TS0601"),
            ("_TZE200_kafooqvr", "TS0601"),
        ],
        .../...
```

Save changes, restart HA and repair the device.

Paddy-van-CH, 2023-02-20:
Yes, I have. Copied it into a custom quirk and added _TZE200_kafooqvr.
Check the ts601_electric_heating.log attached in my earlier post. The quirk does load, but device attributes are not showing up.

javicalle, 2023-02-20:
Sorry, you are right. I haven't seen it in my previous revision:
```
2023-01-29 14:09:38.162 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_kafooqvr TS0601 (2c:11:65:ff:fe:99:e4:9f)
2023-01-29 14:09:38.163 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_electric_heating.MoesBHT'>
2023-01-29 14:09:38.163 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 2c:11:65:ff:fe:99:e4:9f: <class 'ts0601_electric_heating.MoesBHT'>
```

Can you attach the device signature with the quirk applied?

I'm not familiar with the TRV devices, but I see that HA creates new entities:
```
2023-01-29 14:09:38.296 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.tze200_kafooqvr_ts0601_rssi
2023-01-29 14:09:38.296 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event entity_registry_updated[L]: action=create, entity_id=sensor.tze200_kafooqvr_ts0601_rssi>
2023-01-29 14:09:38.297 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.tze200_kafooqvr_ts0601_lqi
2023-01-29 14:09:38.297 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event entity_registry_updated[L]: action=create, entity_id=sensor.tze200_kafooqvr_ts0601_lqi>
2023-01-29 14:09:38.298 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.tze200_kafooqvr_ts0601_hvac_action
2023-01-29 14:09:38.298 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event entity_registry_updated[L]: action=create, entity_id=sensor.tze200_kafooqvr_ts0601_hvac_action>
2023-01-29 14:09:38.299 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new climate.zha entity: climate.tze200_kafooqvr_ts0601_thermostat
2023-01-29 14:09:38.299 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event entity_registry_updated[L]: action=create, entity_id=climate.tze200_kafooqvr_ts0601_thermostat>
```

@MattWestb sorry to ping you directly, do you know what entities must be expected here? And what can be the problem?

MattWestb, 2023-02-20:
If i remember its only basic support = one Climate with only heat and no presets, and one perhaps not working action sensor that can working as On / Off or heat / idle then Thermostats is supporting off as working mode.

Paddy-van-CH, 2023-02-23:
Here we go:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_kafooqvr",
  "model": "TS0601",
  "class": "ts0601_electric_heating.MoesBHT"
}

javicalle, 2023-02-23:
Ummmm, I bet that manufacturer should be added also in HA:
https://github.com/home-assistant/core/blob/e0601530a01b26021f5478694a6b0fd88f01b1a2/homeassistant/components/zha/climate.py#L576-L596

But that 'UserInterface' entity should be created in HA:
https://github.com/home-assistant/core/blob/e0601530a01b26021f5478694a6b0fd88f01b1a2/homeassistant/components/zha/core/channels/hvac.py#L317-L319

Not sure which is the best way to test it in HA.
@Paddy-van-CH can you edit the HA classes? If not, I suppose that you would need to enable the ZHA integration as a `custom_component`. Are you familiar to that?

Paddy-van-CH, 2023-02-25:
> Not sure which is the best way to test it in HA. @Paddy-van-CH can you edit the HA classes? If not, I suppose that you would need to enable the ZHA integration as a `custom_component`. Are you familiar to that?

Nope. Not something I've done before. What's the best place to start? Happy to try and dig in.

closed 2023-08-31 by github-actions[bot]