zigpy/zha-device-handlers #2181: [Device Support Request] consumer_connected binary_sensor for lumi.plug.mmeu01

myhomeiot, 2023-02-06:
Please add ```consumer_connected``` binary_sensor for lumi.plug.mmeu01.
Thanks in advance.

I add it as attribute into OppleCluster class (plug_eu.py) and test it using Manage Zigbee Device from Home Assistant - it's report correctly if something physically connected into the socket (even without power consumption):
```
    attributes = {
        0x0009: ("mode", types.uint8_t, True),
        0x0201: ("power_outage_memory", types.Bool, True),
        0x0207: ("consumer_connected", types.Bool, True),
    }
```

javicalle, 2023-02-08:
You can do it yourself by editing the file from the browser:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/plug_eu.py

Save the changes and follow the prompts to generate the PR.

That way other users can benefit from the change thanks to you.

myhomeiot, 2023-02-09:
> Save the changes and follow the prompts to generate the PR.

Unfortunately it's requires also the changes in Home Assistant Core, so PRs must be coordinated.

javicalle, 2023-02-09:
Well, it can be added in the quirk at any time.
In HA it would be needed that the `zha-quirk` version is released to be able to create the new entities.

closed 2023-02-16 by TheJulianJES