zigpy/zha-device-handlers #2182: [Device Support Request] TS0601_TZE200_m9skfctm

An4cl3t0, 2023-02-07:
**Type of device**
Smoke sensor from tuya

https://aliexpress.com/item/1005003970944017.html?algo_exp_id=e2fad572-652e-4690-814b-2bf040b0d445-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000027602351038%22%7D&pdp_npi=3%40dis%21EUR%2115.14%218.0%21%21%21%21%21%402100bc5c16757939001111922d06d8%2112000027602351038%21sea%21EN%210&curPageLogUid=hw6ak9mCzp9z

**Is your feature request related to a problem? Please describe.**
Device attributes not showing up in HA.

**Describe the solution you'd like**
Proper integration with smoke alarm and battery indicators (co2 detection? it's reported here https://github.com/Koenkk/zigbee2mqtt/issues/15349)

**Device signature**
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_m9skfctm",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

**Diagnostic info**
`{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.1.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Rome",
    "os_name": "Linux",
    "os_version": "5.10.0-20-amd64",
    "supervisor": "2023.01.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "23.0.0",
    "chassis": "laptop",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.6",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.90",
      "zigpy-deconz==0.19.2",
      "zigpy==0.53.0",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.2"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "0403",
        "pid": "6015",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate+"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      },
      {
        "type": "_zigate-zigbee-gateway._tcp.local.",
        "name": "*zigate*"
      },
      {
        "type": "_zigstar_gw._tcp.local.",
        "name": "*zigstar*"
      },
      {
        "type": "_slzb-06._tcp.local.",
        "name": "slzb-06*"
      }
    ],
    "dependencies": [
      "file_upload"
    ],
    "after_dependencies": [
      "onboarding",
      "usb",
      "zeroconf"
    ],
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 35696,
    "manufacturer": "_TZE200_m9skfctm",
    "model": "TS0601",
    "name": "_TZE200_m9skfctm TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4417,
    "power_source": "Battery or Unknown",
    "lqi": 94,
    "rssi": null,
    "last_seen": "2023-02-07T16:08:28",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0051",
          "in_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": "Sensore Antincendio Niki",
    "device_reg_id": "f52ca593cf6a90852eb96af3ad7dbb12",
    "area_id": "camera_da_letto",
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "SMART_PLUG",
          "id": 81
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0001": {
                "attribute_name": "app_version",
                "value": 72
              },
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE200_m9skfctm"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}`

Thanks for your time.

javicalle, 2023-02-08:
I would suggest to add your device to the `TuyaCOSensor` quirk:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_co.py

If don't fully works, enable the debug logs and attach it here.

An4cl3t0, 2023-02-10:
Hi, I tried adding the custom quirk but i have another problem, I edited configuration.yaml as here

```
# Loads default set of integrations. Do not remove.
default_config:

# Text to speech
tts:
  - platform: google_translate

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

# Amazon Echo example configuration.yaml entry
emulated_hue:
  listen_port: 80
  # Amazon Echo/Alexa stopped working on different ports. Search for "Philips Hue Bridge V1 (round)" in the Alexa App to discover devices.

sensor:
  - platform: linux_battery
    battery: 0

zha:
  enable_quirks: true
  custom_quirks_path: /usr/share/hassio/homeassistant/config/custom_zha_quirks/
  zigpy_config:
    ota:
      ikea_provider: true                        # Auto update Trådfri devices
      ledvance_provider: true                    # Auto update LEDVANCE/OSRAM devices
      salus_provider: true                       # Auto update SALUS/Computime devices
      inovelli_provider: true
```

and placed the modified file, but the verification of configuration.yaml fails with this message:
`                    Invalid config for [zha]: not a directory for dictionary value @ data['zha']['custom_quirks_path']. Got '/usr/share/hassio/homeassistant/config/custom_zha_quirks/'. (See /config/configuration.yaml, line 22). `

I checked the folder and it's correct, what am I doing wrong?

An4cl3t0, 2023-02-10:
Hi, I tried adding the custom quirk but i have another problem, I edited configuration.yaml as here

# Loads default set of integrations. Do not remove.
default_config:

# Text to speech
tts:
  - platform: google_translate

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

# Amazon Echo example configuration.yaml entry
emulated_hue:
  listen_port: 80
  # Amazon Echo/Alexa stopped working on different ports. Search for "Philips Hue Bridge V1 (round)" in the Alexa App to discover devices.

sensor:
  - platform: linux_battery
    battery: 0

zha:
  enable_quirks: true
  custom_quirks_path: /usr/share/hassio/homeassistant/config/custom_zha_quirks/
  zigpy_config:
    ota:
      ikea_provider: true                        # Auto update Trådfri devices
      ledvance_provider: true                    # Auto update LEDVANCE/OSRAM devices
      salus_provider: true                       # Auto update SALUS/Computime devices
      inovelli_provider: true

and placed the modified file, but the verification of configuration.yaml fails with this message:
                   Invalid config for [zha]: not a directory for dictionary value @ data['zha']['custom_quirks_path']. Got '/usr/share/hassio/homeassistant/config/custom_zha_quirks/'. (See /config/configuration.yaml, line 22).

I checked the folder and it's correct, what am I doing wrong?

An4cl3t0, 2023-02-10:
Wrongly clicked "closed".

javicalle, 2023-02-10:
Configure your `custom_quirks_path` like:
```python
zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks/
```

An4cl3t0, 2023-02-10:
Same, 
`Invalid config for [zha]: not a directory for dictionary value @ data['zha']['custom_quirks_path']. Got '/config/custom_zha_quirks/'. (See /config/configuration.yaml, line 22). `
Folder is there.

remoteassist, 2023-02-11:
@An4cl3t0 did you create a folder name /custom_zha_quirks ? Look like it's complaining that it's can't see the folder.

javicalle, 2023-02-18:
I can't see any problem in your config.
Try leaving just the minimal configuration and double check the indentation.

TheJulianJES, 2023-02-18:
Can you try just:
```yaml
zha:
  custom_quirks_path: custom_zha_quirks
```
(note: `enable_quirks` is true by default, so no need to include it)

An4cl3t0, 2023-03-13:
Sadly the same : 
`Invalid config for [zha]: not a directory for dictionary value @ data['zha']['custom_quirks_path']. Got 'custom_zha_quirks'. (See /config/configuration.yaml, line 22). `

Path of the custom quirk:
`/usr/share/hassio/homeassistant/config/custom_zha_quirks/ts0601_co.py`

Configuration.yaml:
`# Loads default set of integrations. Do not remove.
default_config:

# Text to speech
tts:
  - platform: google_translate

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

# Amazon Echo example configuration.yaml entry
emulated_hue:
  listen_port: 80
  # Amazon Echo/Alexa stopped working on different ports. Search for "Philips Hue Bridge V1 (round)" in the Alexa App to discover devices.

sensor:
  - platform: linux_battery
    battery: 0

zha:
  custom_quirks_path: custom_zha_quirks
  zigpy_config:
    ota:
      ikea_provider: true                        # Auto update Trådfri devices
      ledvance_provider: true                    # Auto update LEDVANCE/OSRAM devices
      salus_provider: true                       # Auto update SALUS/Computime devices
      inovelli_provider: true`

TheJulianJES, 2023-03-13:
Did you also try specifying the full path?
```yaml
zha:
  custom_quirks_path: /usr/share/hassio/homeassistant/config/custom_zha_quirks
```

An4cl3t0, 2023-03-13:
Yes I tried also the full path.
I'm on Supervised: 
```
## System Information

version | core-2023.3.3
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 5.10.0-21-amd64
arch | x86_64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2023.03.1
agent_version | 1.2.2
docker_version | 23.0.1
disk_total | 87.4 GB
disk_used | 10.9 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.5.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 10 febbraio 2023 alle ore 21:47
-- | --
current_recorder_run | 13 marzo 2023 alle ore 01:09
estimated_db_size | 87.74 MiB
database_engine | sqlite
database_version | 3.38.5

</details>
```

TheJulianJES, 2023-03-13:
But if you're running supervised `/config/custom_zha_quirks` should be the full path, as for the HA Core container, the config directory is just mounted at `/config` I thinl

An4cl3t0, 2023-03-13:
Tried /config/custom_zha_quirks but same error, then entered the container with:
` docker exec -it $(docker ps -f name=homeassistant -q) bash`

the custom_zha_quirks path is:
` /config/config/custom_zha_quirks`

"I'm stupid" moment ;) 

Fixed the doubled config folder problem and now works, I will report the device integration tomorrow. Thanks

An4cl3t0, 2023-03-13:
Looks that there is no need for the custom quirk as the device was already added here:
https://github.com/zigpy/zha-device-handlers/pull/2201

It still misses the low battery info but there is this issue open for that:
https://github.com/zigpy/zha-device-handlers/issues/1838

So I'm closing this. Thanks again.

closed 2023-03-13 by An4cl3t0

01fr, 2023-05-06:
Greetings

Is there any way to expose the siren of this device (TS0601_TZE200_m9skfctm) as an entity, so it can be used in other automations?

doreks, 2023-10-23:
> Greetings
> 
> Is there any way to expose the siren of this device (TS0601_TZE200_m9skfctm) as an entity, so it can be used in other automations?

What model do you have and could you please share converter file used in case it's working?

Thanks

liouma, 2023-11-07:
> So I'm closing this. Thanks again.

hello, did you really get it working ?
I have a similar device but sadly it doesn't work. I tried with and without the quirk. The smoke sensor is recognized but smoke sensor does not trigger in HA when I click the test button.