zigpy/zha-device-handlers #2183: Ikea E1744 (SYMFONISK Sound Controller) Not Working

Akuo95, 2023-02-08:
Hello to everyone
I have one problem with the ikea remote E1744, if i try to pair the divice won't work....
Sometimes ZHA could add the device and see the green screen but won't work (no command received to HA), sometimes still blue screen.

Here is all the log of the pairing process:
```
2023-02-08 00:52:20.318 INFO (MainThread) [zigpy.application] New device 0xd186 (cc:cc:cc:ff:fe:e4:cf:ce) joined the network
2023-02-08 00:52:20.320 DEBUG (MainThread) [zigpy.device] [0xd186] Scheduling initialization
2023-02-08 00:52:20.322 INFO (MainThread) [zigpy.device] [0xd186] Requesting 'Node Descriptor'
2023-02-08 00:52:20.324 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x32 request
2023-02-08 00:52:36.944 INFO (MainThread) [zigpy.application] Device 0xd186 (cc:cc:cc:ff:fe:e4:cf:ce) joined the network
2023-02-08 00:52:36.945 DEBUG (MainThread) [zigpy.device] [0xd186] Scheduling initialization
2023-02-08 00:52:36.945 DEBUG (MainThread) [zigpy.device] [0xd186] Canceling old initialize call
2023-02-08 00:52:36.948 INFO (MainThread) [zigpy.device] [0xd186] Requesting 'Node Descriptor'
2023-02-08 00:52:36.949 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x37 request
2023-02-08 00:52:37.057 INFO (MainThread) [zigpy.application] Device 0xd186 (cc:cc:cc:ff:fe:e4:cf:ce) joined the network
2023-02-08 00:52:37.058 DEBUG (MainThread) [zigpy.device] [0xd186] Scheduling initialization
2023-02-08 00:52:37.058 DEBUG (MainThread) [zigpy.device] [0xd186] Canceling old initialize call
2023-02-08 00:52:37.062 INFO (MainThread) [zigpy.device] [0xd186] Requesting 'Node Descriptor'
2023-02-08 00:52:37.063 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x38 request
2023-02-08 00:52:39.164 INFO (MainThread) [zigpy.application] Device 0xd186 (cc:cc:cc:ff:fe:e4:cf:ce) joined the network
2023-02-08 00:52:39.165 DEBUG (MainThread) [zigpy.device] [0xd186] Scheduling initialization
2023-02-08 00:52:39.166 DEBUG (MainThread) [zigpy.device] [0xd186] Canceling old initialize call
2023-02-08 00:52:39.168 INFO (MainThread) [zigpy.device] [0xd186] Requesting 'Node Descriptor'
2023-02-08 00:52:39.168 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x39 request
2023-02-08 00:52:39.269 INFO (MainThread) [zigpy.application] Device 0xd186 (cc:cc:cc:ff:fe:e4:cf:ce) joined the network
2023-02-08 00:52:39.270 DEBUG (MainThread) [zigpy.device] [0xd186] Scheduling initialization
2023-02-08 00:52:39.270 DEBUG (MainThread) [zigpy.device] [0xd186] Canceling old initialize call
2023-02-08 00:52:39.273 INFO (MainThread) [zigpy.device] [0xd186] Requesting 'Node Descriptor'
2023-02-08 00:52:39.274 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x3a request
2023-02-08 00:52:39.481 INFO (MainThread) [zigpy.application] Device 0xd186 (cc:cc:cc:ff:fe:e4:cf:ce) joined the network
2023-02-08 00:52:39.482 DEBUG (MainThread) [zigpy.device] [0xd186] Scheduling initialization
2023-02-08 00:52:39.482 DEBUG (MainThread) [zigpy.device] [0xd186] Canceling old initialize call
2023-02-08 00:52:39.483 DEBUG (MainThread) [zigpy.zdo] [0xd186:zdo] ZDO request ZDOCmd.Device_annce: [0xD186, cc:cc:cc:ff:fe:e4:cf:ce, 128]
2023-02-08 00:52:39.485 INFO (MainThread) [zigpy.device] [0xd186] Requesting 'Node Descriptor'
2023-02-08 00:52:39.485 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x3b request
2023-02-08 00:52:40.414 DEBUG (MainThread) [zigpy.zdo] [0xd186:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0xD186, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
2023-02-08 00:52:40.415 DEBUG (MainThread) [zigpy.zdo] [0xd186:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_rsp([<Status.SUCCESS: 0>, 0xD186, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)])
2023-02-08 00:52:40.430 DEBUG (MainThread) [zigpy.zdo] [0xd186:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0xD186, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
2023-02-08 00:52:40.431 DEBUG (MainThread) [zigpy.zdo] [0xd186:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_rsp([<Status.SUCCESS: 0>, 0xD186, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)])
2023-02-08 00:52:40.438 INFO (MainThread) [zigpy.device] [0xd186] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
2023-02-08 00:52:40.439 INFO (MainThread) [zigpy.device] [0xd186] Discovering endpoints
2023-02-08 00:52:40.440 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x3c request
2023-02-08 00:52:41.420 INFO (MainThread) [zigpy.device] [0xd186] Discovered endpoints: [1]
2023-02-08 00:52:41.421 INFO (MainThread) [zigpy.device] [0xd186] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
2023-02-08 00:52:41.421 INFO (MainThread) [zigpy.endpoint] [0xd186:1] Discovering endpoint information
2023-02-08 00:52:41.422 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x3d request
2023-02-08 00:52:42.594 INFO (MainThread) [zigpy.endpoint] [0xd186:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=6, device_version=1, input_clusters=[0, 1, 3, 32, 4096], output_clusters=[3, 4, 6, 8, 25, 4096])
2023-02-08 00:52:42.599 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x3e request
[... cut ...]
2023-02-08 00:52:59.086 DEBUG (MainThread) [zigpy.device] [0xd186] Extending timeout for 0x48 request
```
I have also found this in the log:
```
Logger: zigpy.device
Source: runner.py:128
First occurred: 00:53:43 (1 occurrences)
Last logged: 00:53:43

Failed to parse message (b'08110a0af04800') on cluster 768, because Data is too short to contain 1 bytes
```

Other zigbee devices works well, i have also tried different quirks without success...

Thanks a lot in advance for all your help.
Bye
Lorenzo

Akuo95, 2023-02-08:
This is also the LOG when i try to add withour removing the device.
```
2023-02-08 01:03:51.265 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0x5242, SrcIEEE=cc:cc:cc:ff:fe:e4:cf:ce, ParentNwk=0xE360)
2023-02-08 01:03:51.266 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0x5242, SrcIEEE=cc:cc:cc:ff:fe:e4:cf:ce, ParentNwk=0xE360)
2023-02-08 01:03:51.267 INFO (MainThread) [zigpy.application] Device 0x5242 (cc:cc:cc:ff:fe:e4:cf:ce) joined the network
2023-02-08 01:03:51.268 DEBUG (MainThread) [zigpy.application] Device cc:cc:cc:ff:fe:e4:cf:ce changed id (0xd186 => 0x5242)
2023-02-08 01:03:51.269 DEBUG (MainThread) [zigpy.device] [0x5242] Skipping initialization, device is fully initialized
2023-02-08 01:03:51.269 DEBUG (MainThread) [zigpy.application] Device is initialized <IkeaSYMFONISK model='SYMFONISK Sound Controller' manuf='IKEA of Sweden' nwk=0x5242 ieee=cc:cc:cc:ff:fe:e4:cf:ce is_initialized=True>
2023-02-08 01:03:51.279 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x5242:cc:cc:cc:ff:fe:e4:cf:ce entering async_device_initialized - is_new_join: False
2023-02-08 01:03:51.280 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x5242:cc:cc:cc:ff:fe:e4:cf:ce has been reset and re-added or its nwk address changed
2023-02-08 01:03:51.280 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] skipping discovery for previously discovered device - 0x5242:cc:cc:cc:ff:fe:e4:cf:ce
2023-02-08 01:03:51.281 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x5242](SYMFONISK Sound Controller): started configuration
2023-02-08 01:03:51.281 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5242:ZDO](SYMFONISK Sound Controller): 'async_configure' stage succeeded
2023-02-08 01:03:51.286 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5242:1:0x0000]: Configuring cluster attribute reporting
2023-02-08 01:03:51.287 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5242:1:0x0000]: finished channel configuration
2023-02-08 01:03:51.287 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5242:1:0x0001]: Performing cluster binding
2023-02-08 01:03:51.289 DEBUG (MainThread) [zigpy.device] [0x5242] Extending timeout for 0xb0 request
2023-02-08 01:03:51.289 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x5242), dst_ep=0, source_route=None, extended_timeout=True, tsn=176, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b'\xb0\xce\xcf\xe4\xfe\xff\xcc\xcc\xcc\x01\x01\x00\x03DG\xdd\x1c\x00K\x12\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-08 01:03:51.290 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteChk.Req(Dst=0x5242, RtStatus=<RouteStatus.ACTIVE: 1>, Options=<RouteOptions.NO_ROUTE_CACHE|MTO_ROUTE: 3>)
2023-02-08 01:03:51.292 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5242:1:0x0003]: Configuring cluster attribute reporting
2023-02-08 01:03:51.293 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5242:1:0x0003]: finished channel configuration
2023-02-08 01:03:51.294 DEBUG (MainThread) [zigpy.zcl] [0x5242:1:0x1000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=177, command_id=65, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2023-02-08 01:03:51.295 DEBUG (MainThread) [zigpy.zcl] [0x5242:1:0x1000] Sending request: get_group_identifiers(start_index=0)
2023-02-08 01:03:51.296 DEBUG (MainThread) [zigpy.device] [0x5242] Extending timeout for 0xb1 request
2023-02-08 01:03:51.296 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x5242), dst_ep=1, source_route=None, extended_timeout=True, tsn=177, profile_id=260, cluster_id=4096, data=Serialized[b'\x01\xb1A\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-08 01:03:51.298 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5242:1:0x0020]: Performing cluster binding
2023-02-08 01:03:51.299 DEBUG (MainThread) [zigpy.device] [0x5242] Extending timeout for 0xb2 request
2023-02-08 01:03:51.299 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x5242), dst_ep=0, source_route=None, extended_timeout=True, tsn=178, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b'\xb2\xce\xcf\xe4\xfe\xff\xcc\xcc\xcc\x01 \x00\x03DG\xdd\x1c\x00K\x12\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-08 01:03:51.300 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5242:1:0x0008]: Performing cluster binding
[... cut ...]
2023-02-08 01:06:13.549 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x5242](SYMFONISK Sound Controller): Update device availability -  device available: True - new availability: True - changed: False
2023-02-08 01:07:43.549 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x5242](SYMFONISK Sound Controller): Device seen - marking the device available and resetting counter
2023-02-08 01:07:43.549 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x5242](SYMFONISK Sound Controller): Update device availability -  device available: True - new availability: True - changed: False
```

MattWestb, 2023-02-08:
Looks like the device is having problem communicating with some routers and leaving and trying on new one but in the end its have getting the network setting but was timing out before all setting was transferred.
If its still contaminating do one reconfigure from device card and wake up the device so it can receiving the commands.

If its not working delete the device and wait one minute and resetting it and re-adding it.
You can also testing moving it so its trying one other router as parent.
Then you is using one TI coordinator is very likely it cant being pared direct to it (looks they is having some problems with direct children).

Also then its pared and not working one restart of HA can fixing some locks in the device initialization. 

Then its OK pared and configuration you shall getting all events in the log and also one checkin from it every 50 minutes.

wes1993, 2023-02-10:
Hello to everyone,
I have the same problem, did someone have found a solution?
Tomorrow i will post my log here, my coordinator is Sonoff USB Dongle (CC2652P).

Bye
Stefano

closed 2023-08-16 by github-actions[bot]